teckel12/LuaTelemetry #8: Script panic not enough memory

oleost, 2017-09-25:
I have Taranis x9d+.

Just updated OpenTX to test your LuaTelemetry.

Treid both `opentx-x9d+-multimodule-lua-en-2.2.0.bin` and `opentx-x9d+-multimodule-lua-luac-en-2.2.0.bin`

But when selecting iNav as Scripts in display menu, and then holding page to get to it, I get this error message. 

```
Script Panic
not enough memory
```

And script is places in folder
[image]

vavisimo, 2017-09-25:
Hi, did you try changing radio setting of metric to imperial?

oleost, 2017-09-25:
Yes, tried both imperial and metric.

vavisimo, 2017-09-25:
ill try x9d+ and report back ( did not try the letest edit )

teckel12, 2017-09-25:
This script is only tested on the Q X7 (as that's the only Taranis transmitter I own).  I've only used the simulator to work with the additional screen size of the x9d/x9d+.

I didn't realize I was that close to being out of memory with this script.  Is there a way of finding out the memory allocation and lua script memory usage?  Also, what about using the .luac file?  It's written somewhere that the compiled .luac script will use less memory, but I can't figure out how to get the transmitter to work with a .luac file.

Oh, and you could remove the following section from the bottom of the lua script that may free up some memory:

```
  -- *** Altitude graph for X9D ***
  if (not QX7) then
    lcd.drawRectangle(199, 9, 13, 48, SOLID)
    height = math.max(math.min(math.ceil(data.altitude / 400 * 46), 46), 0)
    lcd.drawFilledRectangle(200, 56 - height, 11, height, INVERS)
    local max = 56 - math.max(math.min(math.ceil(data.altitudeMax / 400 * 46), 46), 0)
    lcd.drawLine(200, max, 210, max, DOTTED, FORCE)
    lcd.drawText(199, 58, "Alt", SMLSIZE)
  end
```

Anyway, insight would be appreciated.

teckel12, 2017-09-25:
Also, maybe you have multiple Lua script screens activated which causes the low memory?

oleost, 2017-09-26:
Can confirm `iNavOd.zip` loads up telemetry screen.

Tried removing suggested code above, did not make any difference

I had another small Lua script running, tried removing it and power cycling, did not make any difference.

oleost, 2017-09-26:
Another error appered when using `InavOd.zip`

[image]

It happened after acquiring position lock. Before acquiring lock, the telemetry screen worked.

teckel12, 2017-09-26:
`InavOd.lua` didn't do some table checking which could result in that `Script syntax error`.

Getting back to the latest release, it's possible that it's running out of memory because you have other Lua scripts on other screens.  These all add up and can cause out of memory errors.  If you could remove all other screens and only load the latest Lua Telemetry screen that should work.  It works just fine on my Q X7.

vavisimo, 2017-09-26:
I setup only one Lua script (latest), all other are disabled by none under Screen 2, 3 ... and so on, still script panic - not enough memory.

teckel12, 2017-09-26:
Try this:
https://raw.githubusercontent.com/iNavFlight/LuaTelemetry/development/iNav.lua

Same code, less whitespace. The problem I have in debugging this is that the Q X7 doesn't currently show the free memory for me to see what changes reduce memory and by how much. Trying to get an OpenTX nightly build that fixes this but the download server is offline for OpenTX right now :/

vavisimo, 2017-09-26:
Reporting back, still panic without whitespace

teckel12, 2017-09-26:
:( what does your transmitter report for free memory? From home screen, long press enter, statistics, + key

oleost, 2017-09-26:
44032b Free mem on x9d+

vavisimo, 2017-09-26:
Lua Scripts Free memory
11264b here x9d+

teckel12, 2017-09-26:
Interesting there's that much memory free yet it says it's out of memory. I read that it shouldn't go lower than 4000b to avoid problems.

teckel12, 2017-09-27:
I'm now running the nightly build firmware on my transmitter so I can monitor the free memory. I'll do some tests to see the best way to free up more memory.

Since you're running the `lua-luac` firmware, is the `iNav.luac` file being generated? In other words, is it running out of memory trying to compile the script or actually running the compiled script.

I'll be updating the development branch in a bit that you can try out and see if it runs.

vavisimo, 2017-09-27:
im not running the luac FW, just the lua... if you upload ill try , np

teckel12, 2017-09-27:
@vavisimo @oleost Try this development build:

https://raw.githubusercontent.com/iNavFlight/LuaTelemetry/development/iNav.lua

vavisimo, 2017-09-27:
works, loaded  :) sweet, but could not test it with Tx... tomorrow i will

teckel12, 2017-09-27:
At least I'm going in the right direction with reducing memory.

I also removed the altitude gauge for the x9d/x9d+.  If you want to try to add it back in, it's the commented out lines towards the bottom of the file (lines that start with `--`).  It's possible that puts you over the edge of free memory.

teckel12, 2017-09-27:
You may want to download the latest development  build as I found an error where the distance would show as 3 times too far:  https://raw.githubusercontent.com/iNavFlight/LuaTelemetry/development/iNav.lua

teckel12, 2017-09-28:
If anyone running the Taranis x9d/x9d+ could confirm that the [latest development branch script](https://raw.githubusercontent.com/iNavFlight/LuaTelemetry/development/iNav.lua) works I'll merge it to master and close this issue.

vavisimo, 2017-09-28:
latest development branch script  works too, dont see any changes, but yes they do :)

teckel12, 2017-09-28:
@vavisimo Thanks!  I'll merge to master and close this issue.  Let me know if there's any issues.

teckel12, 2017-09-28:
Development branch merged to master, found a display bug on the X9D / X9D+ so using the master branch is advised.

closed 2017-09-28 by teckel12