teckel12/LuaTelemetry #10: Script Syntax Error

oleost, 2017-09-29:
Tried latest master branch.

[image]

Radio is set to imperial, got the firmware with both lua and luac. 
Other info is that its an airplane model.

oleost, 2017-09-29:
Forgot to mention it loads up the telemetry, but after a while it gives this error message. I believe it was at the same time acquired gps lock.

teckel12, 2017-09-29:
@oleost Interesting...  This means that it knows the GPS information is a table (array in other languages), but you don't have the "lat" key in that table.  Once GPS is locked, you should always have two keys "lat" and "lon" in the GPS table.  It appears that's not the case for you for whatever reason.

I've added an additional check to make sure there's also a GPS fix before showing GPS coordinates.  Hopefully this resolves the problem for you.  If not, I'll have to devise a method that will work if the key doesn't exist that's still fast and doesn't use much memory.  Here's the development build:

https://raw.githubusercontent.com/iNavFlight/LuaTelemetry/development/iNav.lua

teckel12, 2017-09-29:
If this doesn't work, if you could turn on telemetry logs (to save every 0.1 seconds) and get me the log file that would be greatly appreciated.  I could throw it into the simulator and see if I can duplicate the problem, as I can't now.  It could be different GPS or FC hardware, but the logs _should_ help isolate this.

oleost, 2017-09-30:
Seems like it doesnt give syntax error. But even with 8 satelites it still states "no gps fix."

[Fly-2017-10-01.zip](https://github.com/iNavFlight/LuaTelemetry/files/1346679/Fly-2017-10-01.zip)

teckel12, 2017-09-30:
The only telemetry your logs show is `RSSI`.  Either you don't have the log switch turned on for all telemetry sensors (which is default) or your transmitter is not receiving telemetry.  If you look at your logs, there's only a heading for the `RSSI` telemetry.  With mine, I get the following telemetry sensors:

> Tmp1(@C),Tmp2(@C),A4(V),VFAS(V),Curr(A),Alt(ft),A2(V),RSSI(dB),RxBt(V),Fuel(%),VSpd(f/s),Hdg(@),AccX(g),AccY(g),AccZ(g),Dist(ft),GAlt(ft),GSpd(mph),GPS

And this is in addition to all the data in your log file.

Also, be sure to discover telemetry sensors **AFTER** you get a GPS fix.  If you don't do it **AFTER** a GPS fix, it won't find all the GPS sensors.  You'll get `Tmp2` without doing this, but you won't get `0420/Dist`, `GAlt`, `GSpd`, `GPS`.

oleost, 2017-10-05:
Hmm. I did have all the sensors except 0420/Dist I believe.

Anyway, did a new rediscovery.

https://youtu.be/IccFwtiBrZ8

No Syntax error this time, but it says `NO GPS Fix`  in Lua screen.

teckel12, 2017-10-05:
@oleost Your GPS telemetry data doesn't seem right (at least compared to mine).  At the start of the video where you show your raw telemetry data, the GPS line doesn't show GPS coordinates.  I can't really tell what it's showing, but it doesn't look right.  Also, your other telemetry numbers screen is showing something strange for GPS too.  When I add GPS to my telemetry screen, I get actual latitude and longitude values.

Basically, the iNav telemetry sketch is detecting if you have GPS coordinates if it has the lat and lon in a table.  If there's no table, there's no GPS coordinates.  This is how every Lua script I've seen works, how the OpenTX Lua GPS example works, and how I'm getting GPS data from my X4R and XSR receivers.

Maybe you could try to install the example OpenTX Lua GPS example here:

https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/handling_gps_sensor_data.html

In any case, it doesn't seem right at all.  I see very meaningful GPS coordinate information and you seem to be getting some kind of garbage (which is why the iNav script would not detect GPS coordinates).

teckel12, 2017-10-05:
I just thought of something else.  It could be that you changed the GPS telemetry sensor.  If you delete the sensor and rediscover, it will add it back with the default values.  It's possible that you changed it at some point and therefore it's no longer putting the lat an lon in a table like it does by default.

oleost, 2017-10-06:
Spot on.

I deleted all sensors and then readded them and it worked.
Unsure why it has changed, I have never done that on purpose.

So for now it looks like this is working on Taranis plus so cloosing this issue.

Thanks for taking your time to help me figure it out.

closed 2017-10-06 by oleost

teckel12, 2017-10-06:
No problem at all!  I needed a few testers as I was only testing in my limited environment with just a couple models. This was exactly what the project needed.  This exercise also allowed me to figure out how to minimize the memory used which there's basically zero documentation on.  Finally, confirmation that it works on an actual X9D+, as I've only tested it in the simulator.  So that you as well.