teckel12/LuaTelemetry #36: LUA telemetry for Crossfire

advancetom, 2017-10-22:
Hey there i found this treat but cant working with my crossfire micro V2.
I saw my sensors but the lua says no telemetry!!!
It is only working with SPort not Crossfire yet ?

teckel12, 2017-10-22:
I thought I was very clear that this script is for SmartPort telemetry. It's not that it's not yet Crossfire compatible, it's that Crossfire isn't SmartPort telemetry and therefore not compatible.

Someone with Crossfire would need to rewrite the crossfire telemetry to be compatible with SmartPort telemetry and then it would be compatible by default with this script. But, that would also probably break any existing Crossfire telemetry scripts and transmitter model setup.

My guess is that this would never be possible, but I don't have Crossfire to know. Closing this, but let me know if you feel Crossfire is SmartPort and it should work. Also, let me know what confused you to think tb at this script was Crossfire compatible. I give requirements which I believe exclude Crossfire.

teckel12, 2017-10-22:
I just read this issue for OpenTX:

https://github.com/opentx/opentx/issues/4922

It seems that there may be issues with Crossfire telemetry if using the Taranis Q X7.  No telemetry would be more at the transmitter level than the lua script level.  Do normal telemetry screens work for you?

advancetom, 2017-10-22:
Yes my normal Telemetry Screen Works Fine

teckel12, 2017-10-22:
@advancetom Isn't Crossfire a totally different telemetry from SmartPort?  You obviously use a different transmitter module and receiver.

- What telemetry sensors do you get on your transmitter?  
- What telemetry values do you get for Tmp1 and Tmp2?

advancetom, 2017-10-25:
Downlink
TRSS | Signalstärke Downlink
TQly | Empfangsqualität Sender
TSNR | Downlink SNR
GPS | GPS Koordinaten
Alt | Höhe über NN
Sats | Anzahl Satelliten
RxBt | Akkuspannung
Curr | Aktueller Strom
Capa | Akkukapazität
Ptch | Pitch Winkel
Roll | Roll Winkel
Yaw | Yaw Winkel

Uplink

Uplink | zum Empfänger
1RSS | Signalstärke Antenne 1
2RSS | Signalstärke Antenne 2 (nur bei Diversity Empfänger)
RQly | Empfangsqualität Empfänger
RSNR | Uplink SNR
RFMD | 0= 4Hz; 1= 50Hz, 2= 150Hz Aktualisierungsrate
TPWR | Sendeleistung

This are the values from crsf protokoll (Crossfire)
Tmp1 and Tmp2 i dont have it

teckel12, 2017-10-25:
Yeah, those are totally different and there's some critical ones missing. Crossfire telemetry in INAV would need to be updated to work with the Lua Telemetry script.

I'll look at the INAV code and see if additional sensors are even available.

teckel12, 2017-10-25:
@advancetom I just looked at the Crossfire telemetry code. It's sending Crossfire-specific telemetry packages that are then deconstructed by the Crossfire module in your transmitter to show as telemetry values. Basically, it's not possible without new firmware in the transmitter module to accommodate additional flight information from INAV. Not impossible, but highly unlikely.

KnuckleUpFPV, 2018-10-03:
https://github.com/tgreer/inav-telemscript

This is a crossfire inav script if anyone here is interested. Qx7 version is coming.

teckel12, 2018-10-03:
@KnuckleUpFPV Actually, looking at that source code, it seems that maybe Crossfire support could be forced into Lua Telemetry.  As I don't have Crossfire I've never seen what the actual transmitter-side telemetry values you get with Crossfire.  This source code gives a lot of insight.

The main potential problem is the flight mode.  It's sent as a string, and I'm not sure of all the possibilities and if they match with the logic in Lua Telemetry enough to make it work.  But, if I could get a telemetry log and Taranis config from someone who's flying Crossfire I could at least setup a simulation and then see if shoehorning Crossfire into Lua Telemetry is possible.

Below are the instructions for exporting radio and model settings that are needed for a simulation:

1. Connect transmitter to PC using standard method (trim in, power, connect USB) see: https://youtu.be/GAETmZsKqwY?t=2m13s
1. Run OpenTX Companion
1. From 'Read/Write' menu, select 'Read Models and Settings From Radio'
1. From 'File' menu, select 'Save As...' and save radio models and settings to OTX file
1. ZIP OTX file and attach it
1. Make sure you specify the model number/name from your transmitter

I'll also need telemetry log files.  If you're not currently creating telemetry log files, each of your telemetry sensors need to have the `Logs` box checked (I believe that's the default).  Then, you just need to setup a special function to record SD Logs.  I set this to when my arm switch is in the arm position `SF&darr;` (use the arm switch for your model), then set the function to `SD Logs` and the time to `0.1s` or `0.2s`.  As the function is set to record when the arm switch is enabled, it automatically saves log files to your SD card's `LOG` folder for each flight.

Sending a few of those along with the ZIPped OTX file created above will allow me to simulate your entire flight in OpenTX Companion.

teckel12, 2018-10-03:
### Resources
https://github.com/tgreer/inav-telemscript/blob/master/incrsf.lua
https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/crsf.c

It looks like the flight modes are VERY limited:

```
ACRO
ANGL
HOR
!FS
```

There's been some interest for Crossfire telemetry flight modes to match the OSD flight mode strings.  That seems logical but I've yet to see any work done on this.

I could add support in INAV for enhanced flight modes (and matching the OSD flight modes).  I'd just need someone with Crossfire to test my changes before submitting a PR to INAV.  I'd be doing this with INAV 2.0 so the only requirement is that you're currently running INAV 2.0.  Just let me know what target you use and I can build an enhanced flight mode HEX file for you to test.

I'll look at what essential flight modes would be required for Lua Telemetry to work.  It doesn't seem right now that Crossfire even gives an indication when you're armed, so that at least would be a must.  

These enhancements should help everyone with Crossfire, not just those running Lua Telemetry.

teckel12, 2018-10-03:
@tgreer Interested in helping out?  Seems you could even help with the coding too, which would be great as I've had only one pull request for Lua Telemetry that wasn't from me.

KnuckleUpFPV, 2018-10-03:
I have months worth of telemetry logs with gps and barometer info from wings and quads. If i can figure out how to attach the file here im happy to do so. If not i can email it to you.

teckel12, 2018-10-03:
@KnuckleUpFPV You should be able to follow these instructions to save your transmitter's OTX file with all the model/config information:

1. Connect transmitter to PC using standard method (trim in, power, connect USB) see: https://youtu.be/GAETmZsKqwY?t=2m13s
1. Run OpenTX Companion
1. From 'Read/Write' menu, select 'Read Models and Settings From Radio'
1. From 'File' menu, select 'Save As...' and save radio models and settings to OTX file

This will be a single OTX file.  Then, you can just copy over the files from your `LOG` folder from your transmitter's SD card to your computer.

1. From a folder on your computer with all the log files and OTX file
1. Select all the log files and the OTX file
1. Right click 
1. Select `Send to`
1. Select `Compressed (zipped) folder`

Then just drag/drop the created ZIP file to a reply here and it will auto-upload and attach it.

KnuckleUpFPV, 2018-10-04:
I know how to get the files. Getting it here was the issue. I can take care of that tomorrow after work. Didnt know you could drag and drop. Im only ever on github on my phone. Ill get you what you need tomorrow.

tgreer, 2018-10-04:
I wrote my own instead.

teckel12, 2018-10-04:
@KnuckleUpFPV Sounds good! What target is your FC so I can get you a custom build with additional flight modes?

teckel12, 2018-10-04:
@tgreer looks nice. It looks familiar, is it a fork or another project modified to Crossfire telemetry sensors or did you combine a couple different projects?

KnuckleUpFPV, 2018-10-04:
Omnibus f4 pro is the target i have the most fcs of. We have two rssi readings, and a link quality. Link quality is far more accurate than rssi. I will be home in a little bit to get the info to you. Another hour and im off of work.

teckel12, 2018-10-04:
@KnuckleUpFPV Oh, are you running INAV 2.0 final?

KnuckleUpFPV, 2018-10-04:
Im running 1.8, and 1.9.1. I wont have any flights on 2.0 till this weekend most likely. Just put my flite test goblin and my s800 on 2.0. Ill let you know what firmware version i send telemetry for tonight.

teckel12, 2018-10-04:
@KnuckleUpFPV No hurry, I'm still sorting out the code and what Lua Telemetry needs at a minimum.

teckel12, 2018-10-05:
Justification for enhancement to crsf telemetry: https://github.com/iNavFlight/inav/issues/3781

KnuckleUpFPV, 2018-10-05:
[inav lua crossfire.zip](https://github.com/iNavFlight/LuaTelemetry/files/2448511/inav.lua.crossfire.zip)

KnuckleUpFPV, 2018-10-05:
the s800 is an omnibus f4 pro v2
the sonic model ar is an matekf405 aio

if you need anything else let me know.

tgreer, 2018-10-05:
Combination of others tweaked to work with inav and crossfire.

KnuckleUpFPV, 2018-10-05:
I messed up.  I included two sonic model ar files. So its only matek f405 aio. Noisy gyro version. Ill attach an s800 tonight.

KnuckleUpFPV, 2018-10-05:
So do i just flash this to my fc and then try your lua?

teckel12, 2018-10-05:
@KnuckleUpFPV Flash your FC and when you fly monitor your flight mode telemetry sensor. Previously it would just show ACRO, HOR, ANGL. Now it will show RDY (ready), NRDY (not ready), and many different flight modes, for example HOLD, RTH, etc. 

Lua Telemetry won't *yet* work, as I'll need some telemetry logs with the  flight modes from this firmware to shoehorn in the Crossfire telemetry information.

The telemetry for Crossfire is quit different than SBUS. Lua Telemetry depends on this telemetry info to know if (for example) your ready to arm or not. Previously, Crossfire didn't even report if the craft was ready to arm, not ready, or armed. That's key information that Lua Telemetry requires to function. With this firmware, Lua Telemetry will be modified to look at the Crossfire telemetry and know the flight status and mode so it can operate.

With a few telemetry logs from your transmitter with this firmware, the process of making Lua Telemetry compatible with Crossfire is fairly trivial.

First, let me know if the firmware show additional telemetry flight modes, then supply some additional telemetry logs from your transmitter. I can then submit a pull request to INAV to include these changes in the next release. When I get the changes done in Lua Telemetry, I'll give instructions on how to install and test.

Let me know if there's any issues.

KnuckleUpFPV, 2018-10-05:
Will do. Ill be flying this wing tomorrow so i will get it flashed and ready to go tonight. Should have the logs for you tomorrow evening.

teckel12, 2018-10-05:
BTW, I'm not happy with the `RDY` and `NRDY` flight statuses.  I just put them in there as a temporary placeholders.  Now that I've thought about it a bit, ready to arm should probably be `OK`?  Still not sure about the `NRDY` (not ready) flight status.  I was thinking maybe `WAIT`?

How it should work is when you first power up your model it should be in the not ready flight status, so `WAIT` kind of makes sense.  Once it's ready to arm it would show `OK`, which also makes sense.  Once armed, it would show your flight status.  So the progression would be something like `WAIT`, `OK`, `ANGL` or whatever.

The only rub here is that `WAIT` could be forever (for example, if something isn't calibrated, the GPS isn't working or whatever the fault is).  So is `WAIT` still a good note or would some other 4 character note be better?

None of this really matters to Lua Telemetry, as it will take this flight information and then do it's own thing, displaying what it normally does and playing the correct voice file.  So this makes no difference to Lua Telemetry, but it could for people using Crossfire without Lua Telemetry.

KnuckleUpFPV, 2018-10-05:
I think RTA for ready to arm. And wait would be fine.

KnuckleUpFPV, 2018-10-05:
When betaflight isnt ready it tells you why. Like if you dont have enough sats it says gps.

teckel12, 2018-10-05:
@KnuckleUpFPV Use this one, I discovered there was an issue with the previous build (don't have Crossfire so I couldn't test it first, sorry):

[inav_2.0.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/2450974/inav_2.0.0_MATEKF405SE.hex.zip)

teckel12, 2018-10-05:
@KnuckleUpFPV I'll look at the Betaflight source to see what it uses.  Best to use the same as much as possible.

KnuckleUpFPV, 2018-10-05:
Ok

KnuckleUpFPV, 2018-10-05:
Im positive it says gps if you are under 5 satellites and try to arm. It pops up right on the osd.  Crossfire only gives coordinates and number of satellites. I dont know what you mean by betaflight only adds air as flight mode. Its has stab horz and acro as flight modes. I have a logic setup on my transmitter. If Im waiting for sats with goggles off my transmitter says gps lock after it sees 5 or more satellites. The betaflight lua doesn't show anything but settings. I havent ever seen a betaflight lua like the inav and ardu lua.

teckel12, 2018-10-05:
@KnuckleUpFPV The OSD has nothing to do with telemetry. And it's also not specific to Crossfire (or anything else). INAV will also show this information on OSD, and will give thr reason why it can't arm. 

Betaflight currently only has one ADDITIONAL flight mode over INAV, which is AIR. But with the firmware I posted, AIR is added along with many others.

I was under the impression that you were saying that Betaflight was giving you some information over Crossfire that INAV wasn't. But, that's not the case (other than AIR, which won't be the case after my pull request). After my pull request, INAV will give more information than betaflight for Crossfire telemetry.

KnuckleUpFPV, 2018-10-06:
Yeah man sorry about that. Wasnt clear enough.

teckel12, 2018-10-06:
@KnuckleUpFPV I've kinda went back and forth about this... But I think it makes sense to report if there's no GPS lock but no other reasons for not being able to arm.

The reasoning is that there are SO many reasons why you can't arm, it's just too difficult to do with only 4 characters and not very useful as almost everyone runs FPV so the OSD will tell you the reason in a whole line of text.

But for almost every flight, you first wait for a GPS lock

KnuckleUpFPV, 2018-10-07:
I think that sounds great. Had some issues with the wing today. Drove an hour to not get it in the air. Was able to get my other wing with the omnibus in the air. If you have a hex for the omnibus f4 pro. It might be the better option right now. I can just go to the school tomorrow and sling it in the air.

ZinhoRacing, 2018-10-07:
Just another point of view :)
The Flight mode is not limited to 4 characters with crossfire protocol.
Here is what i did (removing the details of unable to arm reasons)

`  // use same logic as OSD, so telemetry displays same flight text as OSD when armed
    const char *flightMode = "Ready to arm";
    if (ARMING_FLAG(ARMED)) {
				if (FLIGHT_MODE(FAILSAFE_MODE)) {
					flightMode = "!FAILSAFE!";
				} else if (FLIGHT_MODE(MANUAL_MODE)) {
					flightMode = "Manual";
				} else if (FLIGHT_MODE(NAV_RTH_MODE)) {
					flightMode = "Return Home";
				} else if (FLIGHT_MODE(NAV_POSHOLD_MODE)) {
					if (FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
						// 3D HOLD
						flightMode = "3D Hold";
					} else {
						flightMode = "Pos Hold";
					}
				} else if (FLIGHT_MODE(NAV_CRUISE_MODE) && FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
					flightMode = "3D Cruise";
				} else if (FLIGHT_MODE(NAV_CRUISE_MODE)) {
					flightMode = "Cruise";
				} else if (FLIGHT_MODE(NAV_ALTHOLD_MODE) && navigationRequiresAngleMode()) {
					// If navigationRequiresAngleMode() returns false when ALTHOLD is active,
					// it means it can be combined with ANGLE, HORIZON, ACRO, etc...
					// and its display is handled by OSD_MESSAGES rather than OSD_FLYMODE.
					flightMode = "AltitudeHold";
				} else if (FLIGHT_MODE(NAV_WP_MODE)) {
					flightMode = "Nav WayPoint";
				} else if (FLIGHT_MODE(ANGLE_MODE)) {
					flightMode = "ANGLE";
				} else if (FLIGHT_MODE(HORIZON_MODE)) {
					flightMode = "HORIZON";
				}else if (isAirmodeActive()) {
					flightMode = "AIR";
				}else {
					flightMode = "ACRO";
				}		
	} else
        {flightMode = "Unable to Arm"; }
`

I liked your idea associated to WAIT (for GPS fix) and !ERR for Error and not able to Arm.

teckel12, 2018-10-07:
@ZinhoRacing While Crossfire may not be limited to 4 characters, that's what the OSD is limited to for flight modes. Also, the more characters, the slower your telemetry will be. So sending many characters will make all of your telemetry slow down for everything else, which may not be ideal. Finally, reasons for not arming are not flight modes even in OSD, but are error messages instead. 

Honestly, I'm not sure if reasons for not arming are really that useful or should be sent as telemetry. No other telemetry sends this information. Also, the info is already available on the OSD. Finally, they hardly ever come up, some really never. 

Basically, it seems like an unnessasary duplication of code from OSD. If it's included at all, it probably should be combined code with osd.c instead of being duplicated in both.

So, focusing on just the flight modes, I'd like to get that working. It will not only assist all Crossfire pilots, but also allow for lua scripts like Lua Telemetry to function (hence my involvement).

I think there should be a conversation about how to combine the error messages on the OSD, Crossfire, and maybe many other places (like OLED display for example). That would be a different pull request.

ZinhoRacing, 2018-10-07:
Totaly agree with you @teckel12, I think we have the same view on that point.
Limiting to 4 characters will make telem speed stay in good health. That was my first idea when i thought of codes for flight modes (issu posted on Inav).
We only need to agree to a standardized list of flight modes to be able to make some scripts work (Lua Telem of course and I'm using one that I modifyed from yaapu for Horus).
About having some error messages available on telem is for me a nice to have, because some times I go to fly without googles or ground station Monitor. So if I'm able to see on my radio screen over a script a reason why I can't arm it would be very helpful. And again, i Agree with you it should be a different pull request.

teckel12, 2018-10-08:
@ZinhoRacing @KnuckleUpFPV Here's the latest build:

[inav_2.0.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/2455876/inav_2.0.0_MATEKF405SE.hex.zip)
[inav_2.0.0_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/2455877/inav_2.0.0_OMNIBUSF4PRO.hex.zip)

And here's the changes to `crsf.c`:
```
    const char *flightMode = "OK";
    if (ARMING_FLAG(ARMED)) {
        if (isAirmodeActive()) {
            flightMode = "AIR";
        } else {
            flightMode = "ACRO";
        }
        if (FLIGHT_MODE(FAILSAFE_MODE)) {
            flightMode = "!FS!";
        } else if (FLIGHT_MODE(MANUAL_MODE)) {
            flightMode = "MANU";
        } else if (FLIGHT_MODE(NAV_RTH_MODE)) {
            flightMode = "RTH";
        } else if (FLIGHT_MODE(NAV_POSHOLD_MODE)) {
            flightMode = "HOLD";
        } else if (FLIGHT_MODE(NAV_CRUISE_MODE) && FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
            flightMode = "3CRS";
        } else if (FLIGHT_MODE(NAV_CRUISE_MODE)) {
            flightMode = "CRS";
        } else if (FLIGHT_MODE(NAV_ALTHOLD_MODE) && navigationRequiresAngleMode()) {
            flightMode = "AH";
        } else if (FLIGHT_MODE(NAV_WP_MODE)) {
            flightMode = "WP";
        } else if (FLIGHT_MODE(ANGLE_MODE)) {
            flightMode = "ANGL";
        } else if (FLIGHT_MODE(HORIZON_MODE)) {
            flightMode = "HOR";
        }
#ifdef USE_GPS
    } else if (feature(FEATURE_GPS) && (!STATE(GPS_FIX) || !STATE(GPS_FIX_HOME))) {
        flightMode = "WAIT"; // Waiting for GPS lock
#endif
    } else if (isArmingDisabled()) {
        flightMode = "!ERR";
    }
```

teckel12, 2018-10-08:
To do:
* [ ] Make changes to INAV so arming and GPS status are sent along with expanded flight modes
* [ ] Create detection and remapping of Crossfire telemetry to standard SmartPort telemetry
* [ ] Testing

ZinhoRacing, 2018-10-08:
What is you idea on first point ? On my side for Horus script I used MSP over crossfire CRSF protocol from code that i merged from betaflight to have the arming status, GPS status (fix type, alt, speed) and distance / direction to home available. I thought at beginning to add it to flight mode too, but had no real idea how to do it. I wish I could help.

teckel12, 2018-10-08:
I fear getting Crossfire to work with Lua Telemetry will be a lot more challenging that I had thought.  Doing any type of development in the OpenTX Companion will be very difficult to impossible as the sensor names and values are all different or totally missing in Crossfire making it impossible to setup a development environment to write code against.  And writing code in a vacuum is a real challenge.

Looking at Crossfire telemetry logs, the following seem true and I need confirmation:
* RxBt = battery voltage?  This is different than SBUS, which VBAT is the battery voltage and RxBt is the voltage going to the receiver (more like 4.5 - 5 volts).
* GSpd and Hdg missing?  I'm not getting these telemetry sensors in the logs, do they work?
* Yaw is the same as Hdg?
* No distance from home info?
* No GPS altitude info?
* No vertical speed?

teckel12, 2018-10-08:
@KnuckleUpFPV I don't follow how you're getting the arming status, GPS status, fix type, alt speed, distance, etc.  I don't see any of those telemetry sensors on the radio.  Not sure what your doing to get that data, or how you would display it.  Maybe it's something Horus specific which I don't have the faintest idea how that would work as I don't have that transmitter nor does Lua Telemetry work with it because it's totally different.

It sounds like you're saying that in addition to the normal Crossfire telemetry you're also sending other information over the Crossfire telemetry?  Would this be something that all Crossfire users do?  Is it something that works with INAV and Taranis transmitters?  If it's something specific to Betaflight or Horus or something where you need to do something custom, it may require a developer (with this specific hardware) to write the code, which wouldn't be me as I don't understand a lick of how you're getting this data on your transmitter.

KnuckleUpFPV, 2018-10-09:
He is using mavlink over bluetooth. The crossfire sends all that information over the bluetooth link as it receives it from the gps. Gspd and heading are reported to the taranis in the telemetry. Rxbat is battery voltage at the model. Sorry haven't had a chance to get your firmware in the air yet. Weekend ended up getting busy.

ZinhoRacing, 2018-10-09:
@teckel12 @KnuckleUpFPV 
Ok, here is the full explanation. (not using bluethoot or Mavlink)
I came across the same missing values when i started to think of using a widget for Horus (I wanted to use LuaTelem but Horus is not supported yet, hope we will do it :) ).
My configuration is Inav2.0.0, using crossfire (crsf protocol) as radio link. On the radio, i'm using as well crossfire (OpenTx2.2.2). So I have a full Crsf protocol radio link from Inav to OpenTx. As you spotted, OpenTx or Inav are not sending/decoding all the values needed for LuaTelem (as for the widget i worked on, same values are missing). So i tryed to find a solution because changing OpenTx is very difficult. I find out that in Betaflight, They have been able to get MSP protocol to passthrough CRSF radio link (MSP protocol is used for Configurator and some light scripts to change calibration like PID). I then merged all the necessary code in Inav (custom made). It do work. I'll post part of the Lua code (telem function) to explain how i did tonight (I'm at work and don't have the source here with me.)

About crsf native telem here is what I understood analysing code in Inav and OpenTx and testing.

- **RxBt =** battery voltage, this is the pack voltage as read by the FC (source Inav code and tested OK)
- **GSpd and Hdg missing?** not working, I had to use MSP ( message MSP  106, MSP_RAW_GPS, getting GPS status(fix), GPS hspeed that is horizontal speed, GPS Hdop, GPS Alt)
- **Yaw is the same as Hdg?** yes, it is the heading (source from Inav code and tested OK) it is in radian 
Here is part of my Lua code for Attitude
``` status.roll = math.deg(getValue("Roll"))
	status.pitch = -math.deg(getValue("Ptch"))
	status.yaw = math.deg(getValue("Yaw"))
	if (status.yaw < 0 ) then
		status.yaw = status.yaw + 360
	end
```
- **No distance from home info?** I get it over MSP (message MSP 107, MSP_COMP_GPS, getting Home Dist and Home Angle)
- **No GPS altitude info?** I get it over MSP 106, Note that Home Altitude is Alt sensor in native CRSF telem.
- **No vertical speed?** I get it over MSP (MSP 109 MSP_ALTITUDE, this is the value used for OSD in Inav).
- **About Armed status**, I had to use another MSP message MSP 150 MSP_STATUS_EX, getting INAV ArmingFlags where bit 2 is the bit to know if it is Armed (in my code statusArmed). 
``` elseif (cmd == 150) then
     if #(rx_buf) > 0 then
         status.packSensorStatus = readrxbufferMSP (rx_buf, 5, 6, 1)
         status.armingFlags = readrxbufferMSP (rx_buf, 14, 15, 1)
         status.statusArmed = bit32.extract (status.armingFlags, 2, 1) 
     end
 end
```
So I have all the informations needed to make it work. Of course it still need improvement as I didn't look after rates and so, but It is a start.

When you will see the Lua code of the my Horus Widget, it will all be clear I hope :)

Summary :  To make it work, I changed Inav code (adding MSP capability, code from betaflight) +  coded CRSF + MSP telem decoding in the Horus Widget (Yaapu Widget).

jorgermp, 2018-11-23:
> https://github.com/tgreer/inav-telemscript
> 
> This is a crossfire inav script if anyone here is interested. Qx7 version is coming.
@KnuckleUpFPV I have a qx7, if you send me the file I can test it.

teckel12, 2018-11-24:
@jorgermp After INAV merges my changes, Lua Telemetry will work with Crossfire on the X9D, X7 and Horus transmitters.

The issue was never with Lua Telemetry, it was that Crossfire telemetry provided rudimentary flight information which doesn't give enough info for Lua Telemetry to function. Like actually knowing when the craft is armed as an example. This changes once INAV is updated in the next release.

KnuckleUpFPV, 2018-11-25:
@teckel12  if I can get a matekf405 hex I can test this out. The matekf405se board I have was destroyed. Havent had much time to fly. Actually got out the other day and lost a wing for a whole week. Found it pierced on a tree limb.

KnuckleUpFPV, 2018-11-25:
@teckel12  the script looks amazing by the way. Took me a second to figure out there were more screens to go through. Absolutely amazing job.

KnuckleUpFPV, 2018-11-25:
i tested the omnibus hex posted here and it doesnt work unfortunately.

teckel12, 2018-11-25:
@KnuckleUpFPV The HEX posed here is for testing the additional flight data information added to INAV. Basically, testing that the new flight modes are transmitted and displayed on your Taranis transmitter. That needs to be incorporated into INAV before a new release of Lua Telemetry that would understand what the new flight modes translate to. There's other things missing as well from Crossfire, like craft distance. Lua Telemetry calculates this based on launch GPS location.

Basically, the HEX is step 1, which we're not fully through as it's not yet been incorporated into INAV (should be in 2.1).

Whe you say it doesn't work, I assume you mean Lua Telemetry, which would be partly true. Look for the flight modes from telemetry sensors, that's what should be tested.

KnuckleUpFPV, 2018-11-25:
It doesnt show anything in the lua. There is no. Flight mode or anything

teckel12, 2018-11-25:
@KnuckleUpFPV Correct, that's not what the HEX is for.  The HEX is to test the telemetry sensors, not having anything to do with Lua at all.  The important telemetry sensor (and the only one changed in the HEX) is the flight mode.

Maybe you should read this entire issue as it's explained above.  Lua Telemetry won't work till what is reported via telemetry with this HEX can be verified.  Keep in mind that I don't have Crossfire, there's no possible way I can test anything without someone providing me with test results so I can confirm what it does.  What would be idea would be to also get telemetry logs from your transmitter, which will show the flight mode so I can verify what it's doing.

Basically, Lua Telemetry is ready to work with Crossfire, but with a bunch of grand assumptions.  I need community support to verify the HEX reports flight modes, get telemetry logs from the transmitter, so I can test things on my end and then release a version of Lua Telemetry that fills in the holes that Crossfire doesn't provide currently.  For example, not even knowing the armed status, not knowing distance, etc.

So forget about Lua Telemetry, that's never been the issue.  The issue is that Crossfire doesn't output information needed for Lua Telemetry to function (basic core things like armed status).  I'm trying to get telemetry logs and confirmation on what telemetry data is coming from INAV.  Then, and only then, can I fold this into Lua Telemetry.

I hope that makes sense, but I think you should read this entire issue as it will probably make more sense.

teckel12, 2018-11-25:
> It doesnt show anything in the lua. There is no. Flight mode or anything

To me, this means you get a totally blank screen.  "doesnt show anything" means 100% blank screen.  Maybe a screen shot?  Even if you have nothing setup, it should show *something*, or an error, or something.

KnuckleUpFPV, 2018-11-25:
Ok i guess i misunderstood what you were trying to say. I dont know what to look for because flight modes were already reported to crossfire. I loaded your omnibus hex and see no change in whats reported in flight modes when switches are changed. It shows the same thing. So before I updated to your hex the fm sensor still reports the same thing after changing to your hex.

[60 more replies]

PR #243 (2019-02-26): V1.6.0
* Adds Crossfire support (requires INAV v2.1.0-RC3 or later)
* Horus: Enhanced display speed (up to 40% faster)
* Horus: Script variables initialize on flight/telemetry reset
* Compatible with Betaflight using FrSky X or R9 series receivers (with reduced functionality) and TBS Crossfire support with Betaflight 4.0+

Fixes #36
Fixes #235 
Fixes #236 
Fixes #237 
Fixes #239
Fixes #240 
Fixes #241
Fixes #242
Fixes #244
Fixes #245
Fixes #246 
Fixes #248 
Fixes #250
closed 2019-02-26 by teckel12