teckel12/LuaTelemetry #39: Requirement of "smartport telemetry receiver" unnecessarily restrictive?

Dronek, 2017-10-31:
I noted on the main page of this project that a "SmartPort telemetry compatible receiver" is required for this script to work. However, as far as I see (without going through the whole code) this script gets it's telemetry values from the radio (e.g. X9D) regardless of the actual telemetry protocol, no?

Using e.g. crossfire, the telemetry values show up in the same place on the radio. The set of telemetry variables might be slightly different and some of their names too but the lua script might be easily changed to account for this. Same applies for mavlink telemetry and so on.

I'm going to play around with the script in order to adapt it to crossfire/iNav telemetry and let you know how it went. Maybe the script can be generalized by having a config header or something to make it easily configurable for use with different telemetry variable sets...

putimir, 2017-10-31:
Hi, while you are at it, can you take a look at the https://github.com/codecae/betaflight-tx-lua-revised/issues/32 and see if you can help.
It would be really great to have iNav/VTX configuration ability through Crossfire link...

teckel12, 2017-10-31:
@Dronek The problem is that each telemetry protocol sends totally different information with different sensor names. You could, for example. Rename your Crossfire sensors to match SmartPort and **some** of the telemetry information would make it to this script. But, I've looked into the other telemetry protocols (including Crossfire) and they simply don't offer the telemetry information that this script needs to function at even a base level.  A good example would be, flight modes and flight modifiers.which would probably require a Crossfire firmware update to work. 

Basically, it's not that the Lua Telemetry script is limiting, it's that the other telemetry protocols are limiting. A severally crippled version of Lua Telemetry could be created that would work with Crossfire, but it wouldn't be nearly as useful.

I suggest you give it a go! Go through your Crossfire telemetry sensors and rename them to what SmartPort uses. And then you'll see what I mean. 

Also, I only have SmartPort telemetry. I also figured as it's by far the most popular it would be the most widely useful. While I have no problem with expanding Lua Telemetry to work with Crossfire, it would be impossible for me to test, so not really realistic for me to be the one to do it.  Someone would need to write it, test it, and submit a pull request for it to be implemented. At some point I think someone will do that. But unfortunately, that person can't be me as I use only SmartPort. I also have no reason to upgrade to Crossfire, as I can fly further than I can see with FrSky. I also have a Mavic if I want to fly 3 miles away for some reason.

teckel12, 2017-10-31:
@putimir Totally unrelated to this issue an unrelated to the Lua Telemetry script in general. But, the Betaflight Lua script already basically works with INAV (if you're using SmartPort). I re-wrote the SmartPort telemetry code in INAV for 1.8.0 and now it supports PID and VTX control over SmartPort.

If you're r trying to do this over Crossfire, I don't think tjat will work nor can I help you as I don't have Crossfire.

But in any case, this is totally out of scope for this project and this issue.

Dronek, 2017-10-31:
@teckel12 this was not meant to make YOU implement all protocols. I'll see what I can do with it. I have been working with telemetry logs based on iNav and crossfire for most part of this year and there's actually quite a lot of information that can be calculated from a small set of variables. For example: groundspeed, distance, odometer, direction of movement, LOS vertical angle (glidescope) can be calculated purely from a stream of time, GPS coordinates and altitude. mAh consumption can be calculated from time and current draw and so on. No problem with crossfire. SmartPort telemetry is transmitting some additional values that are calculated anyway and form therefore redundant data in the telemetry stream. The quality of some calculated values might be even increased when calculated on the radio based on a minimal set of telemetry variables.

Since github is about collaboration, generalizing this script might attract more people who are willing to help with different tasks (visualization, number crunching, etc.). It's just an idea :)

I will let you know as soon as I checked it with crossfire.

putimir, 2017-10-31:
>  Totally unrelated to this issue an unrelated to the Lua Telemetry script in general.

Totally agree, I was only pointing out a general gap in"this area" (inav, telemetry/msp over crsf, additionaly horus ports etc...) to someone who has the means of developing some solutions.

Sorry to clutter, I just hope i'm not alone in this situation and I also have to express gratitude for your input and expertise in this area!

teckel12, 2017-10-31:
@Dronek Crossfire is far more limited than you think. The core of Lua Telemetry is flight modes and flight mode modifiers. These are lacking in Crossfire. There's also other data that are missing or in a different format.

Doing a lot of calculations on the transmitter is really out of the question because of the limited available memory to the microprocessor. I'm basically at the memory limit with Lua Telemetry right now. In other words, even if a limited amount of Crossfire compatibility could be programmed, it probably wouldn't fit anyway.

Again, I've looked at the Crossfire telemetry protocol and it doesn't look worth the effort. It really seems like a Crossfire-specific Lua script would be a better idea because it could never function like the SmartPort Lua Telemetry script.

teckel12, 2017-10-31:
@putimir There's no gap at all. As stated, I've rewritten the SmartPort telemetry in 1.8.0 and it's now functional with the betaflight Lua PID/VTX script over MSP.

If you're saying there's a lack with crossfire I wouldn't know or care. But for sure this isn't the place to discuss Crossfire limitations as this script has nothing to do with Crossfire. Please see the **Requirements** section.

closed 2017-10-31 by teckel12

Dronek, 2017-10-31:
Understood, your project, your rules. No offense.

teckel12, 2017-11-01:
Not really my rules, I just don't think it fits here.