teckel12/LuaTelemetry #43: Script Syntax Error

MRC3742, 2017-11-02:
Telemetry screen will display properly with versions up to and including v1.1.1
After that version I am getting a Syntax error immediately upon switching to the telemetry screen.

With current development version 1.1.6, error is in line 375 and states in debug as follows:
Script     iNav error: /SCRIPTS/TELEMETRY/iNav.lua:375: bad argument #3 to 'drawText' (string expected, got nil)
The problem seems to be in the "ext" value. This seems to be the descriptor that follows the telemetry values listed ( ft, mph, A, %, V, db)
If I change "ext" to a numeric value, such as "3" the script will run with 3 shown as the descriptor in all telemetry fields.

Possibly a problem with my setup as no one else seems to be having an issue?

Taranis X9D+  OpenTX v2.2.0
SPRacing EVO with INav v1.8.0 and Ublox GPS
No Current Sensor

MRC3742, 2017-11-02:
Problem solved and yes it was in my setup.

Telemetry distance sensor 0420 was renamed to Dist as per README.md instructions but it had a value descriptor of "Raw(-)" 
I manually changed this value to "ft" and all is working again with v1.1.6.

This is how OpenTX discovered that sensor and deleting all sensors and rediscovering did not cure the issue. Possibly an addition to the renaming instructions to also check its descriptor value and set to either "ft" or "m"

This issue can be closed as this description above was the fix.

teckel12, 2017-11-02:
@MRC3742 Looking into this now.  I added the auto metric/imperial calibration and indication and maybe didn't test this with the default sensors.

teckel12, 2017-11-02:
@MRC3742 Good catch!  I didn't account for a zero value for the sensor unit (Lua uses a base of 1 for arrays).  Changing the unit of the sensor to `m` or `ft` is the correct solution.  But, I've released a v1.1.7 which will avoid this error and defaults the senor to meters unless you change it to feet.

No reason for you to upgrade to v1.1.7 as there's no other changes and you've already set the `Dist` sensor's unit.

shaiss, 2018-04-27:
Maybe I'm missing something, but I have this same error and just downloaded the master.zip.  Running the script I get the same error as the OP.  on the sensors I have 16: 0402 0.  I'm not sure where I would set this to f or m.  Any help would be appreciated.  Thank you!  My other senor values work as expected.

shaiss, 2018-04-28:
Thx for the prompt reply. I renamed it to Dist and set to m. No dice same
error.  all I see is "script syntax error bad argument #1 to 'writ

Thx for the help and let me know how I can get you more info on what's
going on.

teckel12, 2018-04-28:
@shaiss I don't believe it has anything to do with the `Dist` sensor.  This issue was something totally unrelated (other than a similar error message).

I would really need to know what version of Lua Telemetry you're running, if you're running the S.Port or F.Port protocol, and a telemetry log file from your transmitter would also be great (you can turn on telemetry logs and have every sensor value logged).  I have mine set to log something like 10 times a second, but you could change the frequency if you wish.  It's a good idea anyway as the information you get could be very valuable at some point.

Also, when do you get this error, as soon as you start?  Once something happens (like armed), etc.  You may also want to try the development version (v1.2.5) that you can get [here](https://github.com/iNavFlight/LuaTelemetry/tree/development/src).

Finally, you may want to open a new issue as this issue is totally unrelated.  It's a script error, but EVERY error is a script error.

teckel12, 2018-04-28:
@shaiss Actually, to keep things clear, I've opened a new ticket for your problem.  The link to the issue is #103

closed 2018-04-28 by teckel12