teckel12/LuaTelemetry #54: Display does not recover from failsafe

Allerhopp, 2017-11-20:
After a failsafe is finished, e.g. by moving the sticks, the display does not show the actual flightmode. Temp1 did change to 10005, but the display keeps flashing failsafe.

btw. perfect script, thx!

teckel12, 2017-11-20:
I also saw your similar INAV issue.  I'll be looking into this.  Are you using v1.2.0 of LuaTelemetry?

teckel12, 2017-11-20:
@Allerhopp The way I understand it:
* Tmp1=2215 which means you were in WAYPOINT flight mode.
* You then went into FAILSAFE where Tmp1 went to 52215.
* After returning from FAILSAFE, Tmp1 was 10005 (ACRO / FLAPERON mode).

In the LuaTelemetry script:
* Tmp1=12215 LuaTelemetry shows WAYPOINT flight mode
* Tmp1=52215 LuaTelemetry shows FAILSAFE and beeps (so far so good)
* Tmp1=10005 LuaTelemetry shows ACRO mode

If INAV is reporting Tmp1 as 10005, it would be correct that ACRO mode was displayed.  It may be that INAV is not setting the flight mode correctly after FAILSAFE, but LuaTelemetry doesn't know that.  It sees Tmp1 as 10005 and reports that as ACRO mode.

I ran a test.  It does seem the INAV doesn't recover from FAILSAFE mode.  I'm using a quad so I couldn't test FLAPERON.  But this is what I got:
* Tmp1=325 (HORIZON, ALTHOLD, HEADHOLD)
* Tmp1=40122 when I turned off transmitter, which is FAILSAFE, HEADHOLD, HORIZON
* Tmp1=40122 when I turn the transmitter back on.  It stays in FAILSAFE mode.

Basically, it could be that INAV is not doing something correct for FAILSAFE mode.  But, LuaTelemetry is just reporting what INAV is telling it, and it seems like it's doing that part correct.  Maybe INAV should come out of FAILSAFE mode, but LuaTelemetry doesn't know that and can only report what INAV is telling it.

I do question if you're running INAV 1.8.0 and LuaTelemetry 1.2.0 as it seems you experience something different that I do (I stay in FAILSAFE mode and don't return).  Also, you have FLAPERON enabled which I also question.

More information would be appreciated.

Allerhopp, 2017-11-21:
I am using iNav 1.8.0 and LuaTelemetry 1.2.0.
The iNav issue is a completely different one, but it came up in the same test. LuaTelemetry did not show "Acro" in my case, but kept flashing "Failsafe"  when i was back in Acro from Failsafe finishing the failsafe procedure with the sticks.

To finish the failsafe/RTH procedure in iNav i have to wiggle the sticks, at least for "failsafe_stick_threshold = 50" in my case. But LuaTelemetry did not show the change, kept flashing "Failsafe".

teckel12, 2017-11-21:
It keeps showing FAILSAFE because INAV is still reporting FAILSAFE mode. The 4 in the ten thousand digit is FAILSAFE, and as long as INAV is reporting FAILSAFE LuaTelemetry will display it.

To confirm...  Are you flying fixed wing and in FLAPERON and WAYPOINT mode for this test?

Allerhopp, 2017-11-21:
You did not exit FAILSAFE mode in your example, but i did. I copied the log files and OpenTX config for this copter into the iNav issue.

I ccopied the text messages from iNav code, because i think they have been changed recently and the Wiki still shows the old values. No clue, where i can find the messages for copters actually. Due to my config in OpenTX, i select flightmodes with "FLIGHT MODES", waypoint is higher prorized than anything else but RTH. So the one and only defined flightmode by me in this case was WAYPOINT. But i do not think, this matters. iNav must not cancel all modes and fall back to ACRO.

Allerhopp, 2017-11-21:
https://www.rcgroups.com/forums/showpost.php?p=38670012&postcount=643
Some more flights with changed setup did not show the issue again. Seems i fooled myself, sorry for the trouble.

teckel12, 2017-11-21:
The docs show the correct Tmp1 values (as I wrote the new INAV telemetry code and also updated the docs to match).

I was doing my test on the bench and I could not exit from failsafe mode. But I also manually entered the Tmp1 flight mode values you posted and LuaTelemetry went from WAYPOINT to FAILSAFE to ACRO correctly. FAILSAFE will only display if the ten thousands position's 3rd bit is true (value of 4). You said that position was a 1, which would translate to FLAPERON mode (and your using a quad). Are you sure the Tmp1 value wasn't just "5' as it should be in v1.8.0 of INAV?  A value of 10005 means you're running pre 1.8.0.

Again, I wrote the code and the docs, so I'm very sure of this.

Allerhopp, 2017-11-21:
I do not want to critizise, sometimes written things sound much harder than intended. But i am sure Temp1 was 10005, it is still in the logfile and i am sure to use 1.8.0, latest release.

[image]

But i am a iNav newbee and i do not understand the difference between FLAPERON (plane) and ACRO (quad). Is there a difference in the telemetry?

I am not sure anymore, that FAILSAFE was blinking after FAILSAFE exit. As i wrote, it is possible that i fooled myself. If you want, i can go back to the first configuration and test again.

teckel12, 2017-11-21:
I didn't think you were being critical, and don't get the wrong idea, I'm not upset in any way. I really just want to get to the bottom of this.

FLAPERON is a flight mode if you're flying a fixed wing and you have flaps controlled by a servo. FLAPERON means your flaps are on. If you're  flying a quad, FLAPERON does nothing as you don't have flaps to control.

I'll do an actual flight today to test this out. I noticed a few things that seemed odd but it could have been because I was bench testing without GPS.

teckel12, 2017-11-21:
I did a real world FAILSAFE test today and everything was as expected. I started in 3D hold mode (because it was windy and I didn't want it to drift) and turned off my controller. After a couple seconds it initiated RTH mode. I turned my controller back on and LuaTelemetry reported it was in FAILSAFE mode. I moved the sticks and it exited FAILSAFE and returned to 3D hold (which it reported in LuaTelemetry). 

I'm closing this issue, but if you experience a problem please don't be afraid to open another issue.

closed 2017-11-21 by teckel12

Allerhopp, 2017-11-23:
Seems clear now, what happened. When i ended failsafe and started RTH manually, the quad suffered another failsafe exactly over my head (TX was still in rangetestmode). That´s why i saw FFAILSAFE blinking. Also i was stumbling, because RTH also beeps and flashes and vibrates. In my pov flashing is OK, but beeping and vibration is not necessary in RTH. I think, i am able to change this for me in the source.

teckel12, 2017-11-24:
@Allerhopp I've been adding some configuration options and extended it in 1.2.1 to add any number of them.  In 1.2.1 I've already settled on having a voice alert on/off and a voice status notifications on/off.  So, if you don't want it chatty you can turn off all voice status notifications and only hear the more important alerts.

As for beep/vibration I was planning on having that also be a configuration option.  Maybe turning vibration and beep on/off independently.  The thought of the beep/vibrate was to give a warning for critical modes.  Some are a bit questionable, like RTH and even HEADFREE.  I figured RTH should initiate a beep/vibration because you're technically no longer in control (very similar to a FAILSAFE).  I believe also on my DJI Mavic Pro RTH gives a beep/vibration feedback.

I'll verify this with the DJI Mavic Pro.  If you could suggest config options that you would like to see added, I can try to incorporate them in 1.2.1 which should greatly expand your configuration options.   I opened a new issue #60 which you can add suggestions to.

Allerhopp, 2017-12-02:
Another observation: It seems sound and vibration are stored in a cache, eventually together with the rangetest warning beeps in my case. If the user stops failsafe and the FC returns to a standard flightmode, OpenTX continues to play the warnings until the cache is empty.

teckel12, 2017-12-02:
@Allerhopp There are a few seconds of cached beeps/vibration.  I don't believe i can resolve it as it's an OpenTX thing.  But, I'll look into it further.

teckel12, 2017-12-02:
@Allerhopp The loop runs many times a second, so it's not caching all of the beeps/vibrations or it would continue for a very long time.  It seems that it caches a couple automatically, and that's just how OpenTX works (probably so multiple instructions can happen at the same time).  Basically, it seems this is just how OpenTX functions at the current time.  Probably more of an OpenTX "problem", if it would even be called that as it's probably by design.

Allerhopp, 2017-12-05:
1.2.1 with disabled warnings feels much better for me. One thing, i do not understand: When telemetry gets lost, there are two messages: "telemetry lost" OK, and "engines disarmed", although T1 does not change its value. If telemetry recovers, again "telemetry recovered" and "engines armed". You never see a status change in the log. Is there a chance to intercept the arming messages? A newcomer will be scared because of that ;-)

teckel12, 2017-12-07:
I'll open this as a new issue to look into.