teckel12/LuaTelemetry #55: Not compatible with Horus transmitters

teckel12, 2017-11-20:
Make script compatible with Horus transmitters (I believe it will run as a widget?)
* [ ] Write replacement for lcd.getLastPos() to work on Taranis and Horus
* [ ] Fix screen layout issues on Horus
* [ ] Update documentation on how to use with Horus

teckel12, 2018-07-18:
Now that 1.4.0 adds multiple views, extend that to accommodate Horus with it's own custom view.

fmartinezo, 2018-11-12:
I understand little programming in c ++ and lua, but I'm about to get a 10 horus and I'm interested in the topic

teckel12, 2018-11-12:
@fmartinezo Stock, a Horus doesn't support Lua. I believe you need to switch to OpenTX to "kinda" work with Lua. I also believe you need to manually start each Lua script each flight, not automatic like with Taranis.

fmartinezo, 2018-11-12:
Yes, It will be a mandatory request to run lua on horus equal as you need some requirements in order to run your lua in taranis

fmartinezo, 2018-11-12:
I have been thinking about telemetry for the Horus and I suppose that these work in a similar way to the taranis, at least with openTX.
In the taranis we have 4 screens to show telemetry data, and even by special functions we can make taranis say its value, but there is a problem with the sensors of TMP1 and TMP2 that are formed by multiple values and that except for lua I do not see how to split them.

Reading reading, I have found the theme of the widgets but there I am lost.

It would not be possible to make a specific lua for the horus, which at the moment only represents values, without graphics, and that is able to speak as the current lua does? I say it for the codes that you mention of ERASE and FORCED

Also I've seen this that can be interesting as a starting point, since it's open source https://github.com/yaapu/FrskyTelemetryScript it is for mavlink, but I guess it would be reprogrammable for inav

teckel12, 2018-11-14:
@fmartinezo Here you go, this is as far as I could take it: https://github.com/iNavFlight/LuaTelemetry/archive/horus.zip

I can only run it as a one-time script, I can't figure out how to use the damn transmitter to do it as a widget.  Maybe because I'm trying to do it with the simulator, but nothing seems to work the way any tutorial works.

In any case, while it "works" the display is (as expected) totally wrong.  But, I can't get further than this because it doesn't seem to see any of my telemetry sensors.  Nothing works.

This is my frustration with this transmitter.  It's half-baked, and maybe half of that in the simulator.  Someone needs to take the above branch and give instructions as to how to actually get it working.  For example, what exact button strokes are required to make a widget?  How the heck do you get the telemetry sensors working in the script?

Only once someone with a Horus transmitter can answer these questions and be willing to help will this go further.  Any further requests will result in being forwarded to this message.  If you want it, you're going to need to do some work on your end first as I'm done messing with the terrible Horus platform without someone with the hardware willing to help as outlined above.

\<rant/>

teckel12, 2018-11-16:
I figured out the telemetry. On Horus I was running as a one-time script, which requires me to call the background task (which is where the telemetry is gathered).

Anyway, that's sorted out so now I just need to make a Horus view and make the config screen formatted better on the Horus.

I'd still like to look into running Lua Telemetry from a widget, so any help in that area would be appreciated.

fmartinezo, 2018-11-16:
hi Tim
I tried to execute the lua that you sent me from companion but I can not get it to work decently.
I have internally changed the value of the "path" so that it takes the files from the "widgets" directory but I can not get it out. I have run it from companion looking for it in the sd, but this is a single execution and what comes out is quite deficient.
There is a page https://github.com/yaapu/FrskyTelemetryScript/wiki/How-to-run-the-Yaapu-script-on-X10-and-X12-as-a-Widget that explains how to put it to run as a widget and therefore be permanent and load while the model is turned on, but I have not achieved anything at the moment because at the point where it says that it puts the name of the lua to me does not appear iNav. All my tests are from companion since at the moment I do not have the transmitter. However, I'm still looking

teckel12, 2018-11-17:
It's only running as a one-time script currently, and by running its not doing anything other than showing RSSI, as the view hasn't been created. It can't use a view from Taranis as the screen and font sizes make it unusable, it has to be totally rebuilt.

I know how to start a widget, the question I have is how to build the script to work as a widget. But this may require someone knowledgeable with Horus widgets and Lua Scripts. Not that I can't eventually figure it out, just doesn't make much sense at first glance.

teckel12, 2018-11-18:
Attitude view is looking good now, a few more things to add and I can move on to the map view.  After that I'll work on displaying the rest of the data.

fmartinezo, 2018-11-18:
fantastic, those are good news

teckel12, 2018-11-19:
Sometime tomorrow I'll port a few more features to the Horus view then post a pic of what it's looking like. Better to address any issues early.

teckel12, 2018-11-19:
[image]

Only the title and glass cockpit display are finished, but it gives you an idea of where I'm heading.  The green area will be the radar view, between the glass cockpit and radar will be the variometer, and the blue area will be the flight data.  Haven't got to the layout of the flight data, but it will probably be similar to the X9D's radar view.

fmartinezo, 2018-11-19:
It looks very nice. Very very very good job

teckel12, 2018-11-23:
@fmartinezo Latest progress...
[image]

fmartinezo, 2018-11-23:
@teckel12 Wow, it looks better than spected. Several questions...
Battery is per cell or is total battery?
What are 44.1 and 374?
Were are GPS coodenates?

In a few days my horus will come home. If you wish, I will notify you so that you can provide me with the code to do tests.

teckel12, 2018-11-23:
@fmartinezo Battery can be displayed per cell or total (config option). The 44.1/374 are speed/altitude, which is a common glass pilot display. Speed and altitude will also be displayed in the blank area at the bottom/center (just haven't programmed that yet). Same is true with the GPS data including coordinates, that will be on the bottom right.

See the pilot and radar view pictures for the X9D to get an idea of what it will look like and what the values mean.  There's also a Wiki which describes the views which will make everything on the Horus make sense.

teckel12, 2018-11-25:
For the most part, the Horus view is complete.  There needs to be some polish but the data is all there.  I need to make the config menu format correctly on Horus and then I'd like some beta testers.  Config menu shouldn't take too long, so maybe sometime this week it will be ready for testing.

[image]

[image]

fmartinezo, 2018-11-27:
I have already received my brand new horus x10s so when you have ready the telemetry I can start the tests.
By the way is it going to be a permanent widget or one-time script?

teckel12, 2018-11-27:
@fmartinezo The plan is both a widget and one-time script. Currently, it's only a one-time script.

I've added a GPS signal strength indicator and finishing up larger Horus-specific icons. Next, I'll rework the config menu, which shouldn't be too challenging. On-schedule to test later this week.

fmartinezo, 2018-11-27:
Great. I'm waiting for it. Ready to check

fmartinezo, 2018-11-28:
ok, I'll start with the tests and I'll tell you .....
What luck with the horus and frsky, I also want to send me another ..... :)
At the moment I've found something silly but it's going to make you change the code. There is a sound whose wav file has more than six letters which is the maximum allowed in horus for the names is the "hedhlda.wav"

teckel12, 2018-11-28:
@fmartinezo I believe the file name limits have to do with using the sound files in other parts of the transmitter.  In Lua scripts, there's not these limitations.  The Q7X also has a 6 character limit for audio files, but this audio file works fine on the Q7X in Lua Telemetry.  Also, testing in the Horus simulator plays the audio file just fine.

You'd need to confirm it actually doesn't work on the transmitter, as everything I've experienced says that the 6 character limit doesn't pertain to Lua scripts.

fmartinezo, 2018-11-28:
@teckel12 Hi. I started with the tests and I had to reset the horus twice. I tell you the first one was when loading telemetry for the first time, since internally it generates a "luac". the second, for the same reason when I tried to show the menu. After this and already generated the corresponding luac, it works without blockages. I attached the two screens in scoop in my horus
[image]
[image]

fmartinezo, 2018-11-28:
The tests I just did this afternoon / night in my house, so he has not fixed satellites. Tomorrow if I can I will go out to the street to do some more test.
congratulations for your work

teckel12, 2018-11-28:
@fmartinezo When you say you had to reset the Horus twice, what did you need to do?  Was it that you had to start the script again or the the entire Horus crashed or need to be rebooted?  Also, do you know what build options you have selected for the OpenTX firmware you're running?  These are the check boxes you select in settings in the OpenTX Companion when downloading the OpenTX firmware.

I made a few mostly cosmetic changes to the `horus` branch.  The two main changes you'll see is a larger font size in the config menu and the white box between the attitude and map is gone (when the variometer is turned off).  Use the same process to upgrade to the latest build (the steps 1-4 shown above).

I noted you're using the Spanish language translations.  There's a few new labels on the Horus display that are longer than on Taranis.  I believe the following could be translated and incorporated into the language files which would be useful for non-English speakers:

```
Fuel
Battery
Altitude
Distance
Current
```

A Spanish translation for the above would be helpful.

Finally, there's a testing-only number in the title between the model name and the TX battery gauge.  It shows the frames per second that the display is updating.  When not armed, you can long-press the enter button which will reset the script as well as the FPS number.  In the above screen shots, it shows you're running at about 5 frames per second.  I'd rather see more like 10 or higher, but there's a lot going on so it's kind of expected that it's not very high.  Also, telemetry doesn't get updated very frequently anyway, so it's not too much of a concern.  However, moving forward I would like to try and optimize the code a bit to make it faster.  So keep an eye on the FPS to see if we're going in the correct direction.

fmartinezo, 2018-11-28:
The tanslation will be
Fuel = no translation because it is hard to explain. This is the estimated percent remaining for battery
Battery = Batería
Altitude = Altitud
Distance = Distancia
Current = Consumo

fmartinezo, 2018-11-28:
What I meant by restarting the horus is that it was blocked and did not recognize any system or rtl key. I had to press the reset button to turn off the horus.
The version that comes down from opentx firmware was in Spanish, luac and luac marked
[image]
[image]

fmartinezo, 2018-11-28:
Clarifying more about the restart I had to insert a pin on the emergency off button that has the horus next to the slot of the sd card as it did not accept to disconnect with the switch and did not respond to system buttons, etc.

fmartinezo, 2018-11-28:
For what you say, I understand that you have made changes in the software and that I have to download it again. Is it like that?

fmartinezo, 2018-11-28:
I think I have misunderstood you in relation to the translation. Do you need literals or wav files?

teckel12, 2018-11-29:
@fmartinezo Yes, just download and install it again. It includes some cosmetic changes and it's possible it won't lock up again when you run it the first time. This is the kind of problem having my own transmitter will allow me to diagnose easier.

As for the Spanish translations, that's perfect, it's for the literal screen view.

I also played around with running the script as a widget. It's not really designed to have a single code base for Taranis telemetry script and a Horus widget. But, I believe I can make things work. I got the screen to display without any errors, so I feel confident it's possible.

fmartinezo, 2018-11-29:
@teckel12 Hi Tim. I just downloaded and installed the latest version. It has not been blocked the first time I have executed it and that I have started from scratch. I deleted everything and installed everything again.
I managed to block it and I tell you how. You turn on the transmitter only, you execute the script and it crashes. Then you turn on the receiver but it is still blocked. It gives the impression that when not receiving telemetry values it is blocked.
I have seen the fps in 5.2 but I have also seen a value around 7. It was in a screen change to see the options menu and return to the telemetry screen.
I'm going out on the street to set the gps and see how it goes, in addition to listening to the voices.
I'll tell you later

fmartinezo, 2018-11-29:
I just went up the street and the tests have been satisfactory. The sounds are correct and the script goes well. There is only the litmus test (definitive test) that will be in flight since it is forbidden to fly in the city. Here you have two photos. Do not miss out on the border because I have deleted the gps coordinates
[image]
[image]

teckel12, 2018-11-29:
@fmartinezo A few more updates to the Horus branch today.  Nothing should look different, but I changed how the screen is drawn and I'd be interested to see if the frames per second count is better, worse, or indifferent.

fmartinezo, 2018-11-29:
OK tell me when I can download it.
Turning the translation into Spanish for the word "Fuel" could be put "Batería restante". If "Batería restante" is too long, just put "Batería" and also match the voice that says "battery xx%".
For "Battery" put "Tensión" that's what we call voltage here.

fmartinezo, 2018-11-29:
Altitude, Distance and Current can be as I indicated

teckel12, 2018-11-29:
@fmartinezo The update process will always be the same.  As soon as I push changes to the `horus` branch, you can download them to your transmitter.  The process is:

1. Download [LuaTelemetry-horus.zip](https://github.com/iNavFlight/LuaTelemetry/archive/horus.zip)
1. Open the `LuaTelemetry-horus.zip` file
1. Open the `LuaTelemetry-horus` folder and navigate to the `dist` folder
1. Copy the contents of the `dist` folder (both the `iNav.lua` file and `iNav` folder with contents) to the transmitter's SD card's `\SCRIPTS\TELEMETRY\` folder

fmartinezo, 2018-11-29:
@teckel12 Is it available now?

fmartinezo, 2018-11-30:
@teckel12 Hi, I just downloaded the script and checked it. The fps have improved. They have gone from being around 5 (4.9, 5, 5.1) to being around 7 (6.9, 7, 7.1).
If you want something else, let me know and tomorrow I will resume the tests, now it's a bit late for me and I'm going to rest.

teckel12, 2018-11-30:
@fmartinezo Thanks. The latest release you tried uses images for background and foreground instead of drawing the entire screen.  I figured that would speed things up. 

I got the widget script to seemingly work but It was unstable. I'm not sure if it's just the simulator or not. I'm getting the X10S in a couple days so I'll probably wait to test on an actual transmitter before releasing.

At this point the build you're running should be ready to fly. Fly and let me know what happens.

fmartinezo, 2018-11-30:
Hello Tim, this weekend if there are no problems with the weather I will try the script and I'll tell you how it goes.
I'm signed up in two inav telegram groups. What do you think if I mention that this development exists in case someone else wants to be a tester?

teckel12, 2018-12-01:
@fmartinezo Testing is open to anyone, with the knowledge the script is untested in the field. There's safeguards in OpenTX for a script using too many CPU cycles or crashing. And the core of the script is well-tested on Taranis. Only the display code is new for Horus.

With that said, there's some setup to do before using Lua Telemetry, which takes a bit of effort.

But, go ahead and let others know, with the above slight warning.

fmartinezo, 2018-12-01:
Fantastic news I look forward to the widget and as always ready to do the checkup.
I have already told to the people that the script exists and that it would be good for us to do tests, logically under their own responsibility. I have sent them to this page so they can see how it goes, what has been done and how to download it and put it to work

fmartinezo, 2018-12-09:
@teckel12 This has been sent by a partner who has tried it. I have not been given the case, since I deactivate the script before dismounting, but when I have occasion I try it and I confirm it. He says:
 
Well, on Saturday I tried it once and it worked well in flight. I landed and when disarming I get on the screen "emergency mode".

The radio only  responded to the control sticks, the switches did not respond ...

teckel12, 2018-12-09:
@fmartinezo Not sure what emergency mode is. Sounds like it could be crash and the transmitter enters a crippled mode. The error is defiantly not directly from Lua Telemetry as it has no error like that.

I'll look at OpenTX for that error and obviously try to duplicate.

fmartinezo, 2018-12-10:
Have you received your horus?

ZinhoRacing, 2018-12-10:
Hi,
I had Emergency mode on the Horus if trying to write text on lcd outside of the lcd area in Lua script.
There is an issu in Open TX for that. If it can help.

@teckel12, I installed LuaTelem script on my horus (with crossifire, I know not working yet). The view is great ! :)
I had a quick read of your source code, I found that file data.lua is handling the reading of telem values. I just wanted to try to make the attitude works to see how it behaves (I thought of changing only the names for pitch, yaw and roll as these channels are available in telemetry in OpenTX over CRSF).
But unfortunatly Companion reporting a nill error when I try to compile ... I used Horus Branch code.
If you have a clue on the problem I would be happy to know, if not, please don't waste your time on it, I would prefer that you can progress on Crossfire integration for sure. :)

teckel12, 2018-12-10:
> Have you received your horus?

Yes, I received my Horus X10S.  I flashed OpenTX and installed Lua Telemetry, but haven't yet imported models to do real testing.  In the next couple days I'll have a fully-functional Horus test.

teckel12, 2018-12-10:
@ZinhoRacing Lua Telemetry shouldn't be writing text outside the LCD area, I'm fairly careful about that.  But, there could always be an unforeseen condition that could cause it.  I'll look into it.

For the attitude, it only needs pitch and roll, and Lua Telemetry already looks for two sensors `Ptch` and `Roll` which I believe is what Crossfire already names them.  But, I believe Crossfire outputs these values as radians while INAV outputs them as decimal (multiplied by 10 so it's just an integer).  So, you should be able to just do the following in `iNav.lua` at about lines 128-129 and it *should* work:

```
			data.pitch = math.deg(getValue(data.pitch_id)) * 10
			data.roll = math.deg(getValue(data.roll_id)) * 10
```

BTW, `data.pitchRoll` is set to true if the sensors `Ptch` and `Roll` exist.

effysh, 2018-12-10:
wonderful job guys where the updated script can be downloaded for testing?
is this is the one: https://github.com/iNavFlight/LuaTelemetry/archive/horus.zip ?

ZinhoRacing, 2018-12-10:
@teckel12 it is working (for native crsf telemetry data), look nice.

[image]

I like the large pilot view.

teckel12, 2018-12-11:
@ZinhoRacing Very cool! I'm going to work on my Horus/Crossfire development tomorrow. Unless I get distracted by my EMAX Tiny Hawk BnF I just got TODAY :wink:

ZinhoRacing, 2018-12-11:
Wowww I want one ! :smiley:
Is it Inav compatible ????  It's a F4 matek F411 look like.

Crazy little thing !! Enjoy.

effysh, 2018-12-11:
> @effysh Yes, that's the link to download the latest build. Think of it as a nightly build, as it changes nearly every day. Following are the instructions to get the latest latest Horus build. You can do this every day to get the latest.
> 
> 1. Download [LuaTelemetry-horus.zip](https://github.com/iNavFlight/LuaTelemetry/archive/horus.zip)
> 2. Open the `LuaTelemetry-horus.zip` file
> 3. Open the `LuaTelemetry-horus` folder and navigate to the `dist` folder
> 4. Copy the contents of the `dist` folder (both the `iNav.lua` file and `iNav` folder with contents) to the transmitter's SD card's `\SCRIPTS\TELEMETRY\` folder

Hi 
i just tested the script and i get an error when i press to execute the inav lua  file 
all of sensors  are working on my horus 
[image]
[image]
[image]
[image]

teckel12, 2018-12-11:
@effysh Not sure why it's not working for you. What version of OpenTX are you running? Even if you don't have any sensors it will still run.

effysh, 2018-12-11:
Ok i just figure it out
I copied the folder dist in to telemetry folder instead of its content (inav folder and inav.lua) now its working 100%
Thanks again for this great script

teckel12, 2018-12-12:
@effysh Glad you figured it out.  Would different wording for the install instructions make more sense?

fmartinezo, 2018-12-12:
@teckel12 Hi Tim new spanish voices released. Less metalic and less robotic. Files are on spanish issue (closed)

teckel12, 2018-12-13:
Working on increasing speed and reducing memory.  Also playing with making it a widget.  While I can get the screen to render as a widget, nothing works *yet*.  It has to do with how the script was set up, it just wasn't designed with a widget in mind.  A few more days of tinkering and I hope to have a widget breakthrough.

[25 more replies]

closed 2018-12-29 by teckel12