teckel12/LuaTelemetry #60: Add config options for controlling warning beeps and vibration feedback

teckel12, 2017-11-24:
* [ ] All Vibration - on/off
* [ ] All Beeper - on/off
* [ ] Battery Beep/Vibrate - on/off
* [ ] RSSI Beep/Vibrate- on/off
* [ ] RTH Beep/Vibrate- on/off
* [ ] Headfree Beep/Vibrate- on/off

Allerhopp, 2017-11-24:
I would prefer a solution analog OpenTX RSSI warnings:
As default one beep and one vibration, if the status worsens. This is useful for Battery and Failsafe only imo. RSSI is already covered by OpenTX.
Additional alarms can then be made configurable.
My own experience, when failsafe happened with the recent version: two problems occured simultaneously: control link broken, OK, but why is my TX continously beeping and shaking? ;)

teckel12, 2017-11-24:
4 of 6 added in #62

teckel12, 2017-11-24:
@Allerhopp LuaTelemetry's feedback is designed to enhance what the transmitter does by default. A warning would be just a single feedback. Critical information provides feedback every 5? seconds.

For example, if RSSI is below the critical level that you've set in your transmitter LuaTelemetry will keep warning you of this every 5? seconds. Same is true with critical battery level. Similar as well is max altitude (except every 10? seconds because it also speaks the altitude).

In all cases, when there's something critical, there will be flashing inverted text to go along with it. So, when RTH was flashing, that would be the indication as to why it's beeping/vibrating.

Pull request #62 (untested) adds new config features to better control this. It currently addresses turning RTH and HF from critical to just information. Also, you can totally turn off beeping and vibration feedback independently.  I still need to think about battery and RSSI warnings as how those should be controlled.

teckel12, 2017-11-24:
RSSI, battery and altitude feedback/warning options added to config menu in #62

teckel12, 2017-12-02:
Included in v1.2.1

closed 2017-12-02 by teckel12