teckel12/LuaTelemetry #69: LUAC option in the docs

Allerhopp, 2017-12-02:
"Make sure you enable the luac build option in your OpenTX firmware"
Hopefully i am not completely wrong, but from my experience, this is not neccessary.
If users are happy with the compiled version, there is no need for the luac option, even the lua option is not needed. It is only needed for mixer scripts. In my pov this statement should be changed in: "If you want to make changes in the source, you will need OpenTX with luac build option"

New users can try LUATelemetry without any lua option - it lowers the hurdle for this excellent script in my pov.

teckel12, 2017-12-02:
I believe I tried it without the luac option and it didn't work.  But, I'll try it again and adjust the docs as needed.  Thanks!

teckel12, 2017-12-02:
@Allerhopp Tested without `luac` and indeed LuaTelemetry still works.  Not sure what I was testing previously where I got confused.  Merged into master.

closed 2017-12-02 by teckel12