teckel12/LuaTelemetry #71: Feature Request: GPS Model Finder

SPlissken, 2017-12-03:
I was thinking that this excellent script might benefit from a last known GPS location feature.
This would mean that in the event of a crash and battery ejection the last know coordinates could be reviewed.
Maybe sample every few seconds and allow review of last 5 samples in case of bad data transmitted at the point of impact?
I've had this ability before with the use of FRS_Logger on Android and a BT module connected to the TX module. It gave me a lot of confidence when flying long range somewhere new with my TBS Disco.

Freezing the last known coordinates can work but I've seen these corrupted and become unreliable before as the result of a crash, so the ability to log and review via a friendly UI would be really helpful, I think.

teckel12, 2017-12-04:
Sounds like a good idea. Any thoughts on how one would review this?

SPlissken, 2017-12-04:
I was thinking that it could be at the bottom of the config menu and open a details page to show the GPS coordinates list. The font would need to be large enough that people could easily transcribe into their phone/tablet.
Do you think this might be a memory hog?

Allerhopp, 2017-12-04:
Recently i have done a lot of flights in rangetestmode until failsafe happened. OpenTX keeps showing reliable the last valid GPS position when the telemetry link breaks (usually before the control link). The user can see it also, when he switches to the telemetry page. What do you think about a routine, where LuaTelemetry shows the last GPS position in huge letters when the telemetry is broken? There is not much else to show for the script in that case. When telemetry recovers, it could switch back to the default screen.
Imo the SPort error handling should prevent bad data from coming through. Due to user error (wrong altitude in waypoint) i lost the copter once and found it again within 5m of the last logged GPS position.

SPlissken, 2017-12-04:
I think that works well for a sudden battery ejection or telemetry loss but what about this use case:

Craft goes down with power and telemetry, battery stays connected and GPS is lost either physically or number of visible sats is reduced dramatically, this could result in incorrect coordinates being sent or default position being sent which will overwrite the good data in OpenTX.

@Allerhopp I have done what you describe before and it has worked however I have also had the above happen to me too and not had the correct data displayed.

I think that just a single last known good position is all that is needed but leaving LUA to determine what is good and bad might be a pain so allowing a human to eyeball a few pairs of coordinates might make it simpler to implement.

teckel12, 2017-12-04:
@Allerhopp If telemetry is lost, I like to see the distance and using LuaTelemetry's orientation information to see the direction.  Without using GPS coordinates, I can basically walk in the correct direction to where the downed craft is located.  If there's a connection, I can also turn on beeping and even without GPS it can be found.  So only showing GPS doesn't seem ideal.

Currently, the config menu can only be accessed if the craft is unarmed.  I did that as a precaution, but I've been rethinking this.  Adding a "Review GPS Coors" option on the config menu sounds logical, as long as it could always be accessed (even if armed or it thinks it's armed).  I'll look into this, it shouldn't use much memory and I'm not too worried about memory now that I'm distributing a pre-compiled script which was the big memory pinch (compiling on the transmitter).

teckel12, 2017-12-09:
Resolved with #73

teckel12, 2017-12-09:
Merged into  v1.2.2

closed 2017-12-09 by teckel12

teckel12, 2017-12-09:
@SPlissken A little insight into the new GPS log feature.  When LuaTelemetry detects valid GPS data it saves it every second in a 5 slot FIFO.  The last option on the config menu is `GPS`.  You can move down to this option, hit `Enter` and scroll through the last 5 coordinates.  If you're flying at the time, you'll see the info change as it's updating the latest coordinates every second.

It's a nice little fall-back just in case something happens.  5 seconds should be plenty as the last crash I had at 190 feet (motor screws vibrated loose) was all of about 2 seconds till landing on the roof of a church.  The last 5 seconds of GPS coordinates should good for any condition I can think of.

BTW, here's a video of retrieving the crashed drone with my Mavic: https://www.youtube.com/watch?v=xA166oGto5U&

SPlissken, 2017-12-09:
@teckel12 Nice one, I'll have a play with it next week.
Great rescue with the Mavic, and I thought my 160 4" was a small iNav build!

teckel12, 2017-12-10:
I have a 75mm INAV build too. Only GPS, no barometer or mag though. Underpowered for sure (1.5" props). Will probably convert to indoor only and run BF.

SPlissken, 2017-12-10:
I'm quite pleased with my 160, had a bit of trouble isolating the on-board compass though:

80mph on 3s  (more to come: waiting on a motor upgrade to give me >10:1 thrust to weight)
RTL
Waypoints and Follow Me (Bluetooth LE)
AUW 240g
12 min hover time

I also have a dodecacopter based on 6045s that could do with knowing its way home...

Maybe there should be a competition for the silliest auto-navigating vehicle?

teckel12, 2017-12-10:
My 130 is a hoot to fly with 4s graphene batteries. Well, it was till the motor screws (with locktite) came loose after I installed rubber motor dampers to reduce jello. Needless to say that wasn't successful.

Installing GPS on even a silly craft is a good idea. I only ever hand one model decide to fly back to China, but I'd like to avoid that anyway.

SPlissken, 2017-12-10:
Funny that, my fly-aways have always headed due south like migrating birds to the Gulf of Guinea (lat 0, long 0).

I was really disappointed in the 4s graphene 500mah batteries I tried: massive voltage sag. It could have been down to the overpropped 1306 motors with 3 blade 4045BNs though.

I know that it's not fashionable but I find that 3s LiHVs work best for the 160 with BX1306s, they're cheap too now because of their bad reputation for longevity.