teckel12/LuaTelemetry #76: Home  reset information

Allerhopp, 2017-12-10:
Maybe a minor enhancement:
Due to my setup: MSB and telemetry on UART1, telemetry is only active, when armed. And arming is only possible with a satlock. Therefore i never hear the message home reset. I could change this, but for safety reasons i would like to keep the setup. I am used to make a manual home reset before take off. But this manual home reset is not covered by the standard announcements. Is there a chance to announce a successful manual home reset?

teckel12, 2017-12-10:
@Allerhopp With INAV, once there's a GPS lock and before being armed home reset happens constantly, not just once. Lua Telemetry doesn't announce home reset at this time, as this information isn't passed via telemetry. Instead, LuaTelemetry announces that "GPS is good, ready to arm" (I assume you don't hear that). It doesn't say home reset, because it happens constantly from this point to being armed. 

Once armed, if you trigger home reset LuaTelemetry will announce that home position was reset.  You should be hearing this I'd you have a channel that you use to reset the home position. This will only work if your running INAV v1.8+ as home reset wasn't being sent till that release. 

I hope I've understood the problem correctly. Let me know if I'm missing something.

Allerhopp, 2017-12-10:
I did not know, that iNav updates home position after the first GPS lock. No need for a manual home reset then. But indeed i do not hear a announcement, when a manual home reset is triggered. But i will recheck this again.

teckel12, 2017-12-12:
@Allerhopp You won't hear the home reset announcement unless you're armed as that's how INAV works.

To test: Arm, launch, move to new location and **while still armed** do a home reset. You'll hear the home reset announcement (as long as you're running INAV 1.8).

If you then fly and do a RTH it will return to the location where you did the last home reset.

Allerhopp, 2017-12-12:
Thx, i will test it, when weather and job permits and come back the next days.

Allerhopp, 2018-02-06:
Can someone confirm, that the home reset announcement works? I followed the instructions, home reset actually works (GUI and real life), but i never get the audio message. iNav 1.8, script and audio files V1.2.3. What am i overlooking?

teckel12, 2018-02-06:
@Allerhopp Detail what you're doing to simulate the home reset?  It works for me.

teckel12, 2018-02-12:
@Allerhopp It's possible that the `homrst.wav` WAV file doesn't exist in the iNAV subdirectory.  Maybe you upgraded just the lua script but not the wav files?  Please see this link for install instructions and verify you have the wav file required:

https://github.com/iNavFlight/LuaTelemetry#inav-lua-telemetry-screen-setup

Allerhopp, 2018-02-12:
As i wrote, i can play homrst.wav in the OpenTX file browser. But it is indeed not very imortant. Must be something specific with my setup. Thanks for your time, this script is perfect for me anyway.
[image]

teckel12, 2018-02-13:
I'd rather leave this issue open and look more into this.

teckel12, 2018-02-13:
@Allerhopp Looking into this further...  You mentioned:

> activate home reset (no message, but Tmp1 first digit changes from 1 to 3 during home reset)

That's not the expected result with INAV 1.8.0.  With INAV 1.8.0, a home reset will change the `Tmp2` value, not `Tmp1`.  Also, it should change from a number like `3912` which means `GPS_FIX`, `GPS_FIX_HOME` & 12 satellites to `7912`.  It adds `4000` to `Tmp2` when there's a home reset.

If `Tmp1` is changing from starting with a `1` to starting with a `3` when you do a home reset, it means you're not running the release INAV 1.8.0, which is required for LuaTelemetry.  You may be running a development version of 1.8.0 or a 1.8-RC1, which did have the home reset setup that way for a short period of time (10 days from October 16th to 26th, where it was then changed to `Tmp2` which was more appropriate, and allowed for other flight modes to be added.

While LuaTelemetry will work with INAV 1.7.3+ other features have been added in later versions.  Specifically for the home reset, see the last line of [Tips & Notes](https://github.com/iNavFlight/LuaTelemetry#tips--notes)

> INAV v1.8+ is required for `Home reset` voice notification

I can absolutely confirm that you're not running the 1.8.0 release of INAV.  Looking at the source code revisions and the dates the file changed, you must be running either 1.8-RC1 or a development build before the official 1.8.0 release, which is needed for the home reset to function.

Allerhopp, 2018-02-14:
Impressive investigation! This explains everything. A CLI dump shows:
# INAV/SPRACINGF3 1.8.0 Oct 24 2017 / 06:57:49 (0f1975ce)
I was sure, this is the official release from the numbers, but..... 
Thanks again!

closed 2018-02-14 by Allerhopp