teckel12/LuaTelemetry #80: Break-up readme further by creating wiki?

teckel12, 2017-12-12:
Once release history is pulled out of readme, consider keeping readme more concise by off-loading  detailed info to wiki (linked from readme).

pkerney, 2017-12-12:
I was looking for a good place to ask this question without creating an "issue" and this looks like a good place.
I haven't had much experience with LuaTelemetry, maybe about 5 flights with fixed wing, and I still am trying to understand the display symbols on a X9D+ despite looking at the description in the readme.
If someone could spend a bit more time describing the display and usage in the Screen Description that would be great especially with the X9D+.

teckel12, 2017-12-13:
@pkerney I assume you mean the 3 orientation indicators across the center?  They're the same as with the Q X7 except with the additional screen width all 3 can be displayed at the same time.  From left to right:

1. Compass-based model direction.  If it's pointing to `N`, you're flying North.
2. Shows the direction of the model based on launch/pilot position and orientation.  The small `+` is your model and the dotted line starts from your launch position.  So, if your model launched facing South and your model is behind you to the North, the indicator would show the `+` below the launch point (behind you as the top of the display is your launch orientation.  Still launching facing South, if your model was South West of the launch location the `+` would be to the upper right.  Basically, if your standing and facing the launch orientation, you can look at this graphic and point in the direction of the dotted line towards the `+` and that's where your model is.  If you're within 25 feet of launch, this graphic is hidden.
3. The final graphic is the launch/pilot-based orientation view.  Again like in 2 above, it's based on the models's launch orientation. This shows the direction your model is pointing based upon the launch orientation.  Still launching facing South, if your model is facing South, the arrow is up.  If you point the model to the North, the arrow will be facing down.

Basically, graphic 1 is just a simple orientation.  Graphic 2 & 3 are based on the direction you launched from.

How do I use them?  I honestly don't use the first graphic, but I know some people like to think of the direction the model is facing in that way.  Graphic 3 I use a lot (actually, graphic 2 & 3 are displayed at the same time on the Q X7).  If I'm flying FPV and my battery is low and I'm disorientated or the sun is in the direction of the model I'll do a 3D HOLD and then rotate so graphic 3 is pointing up.  This way, I know left is left, right is right, forward/back are all the same.  I can then use graphic 2 to know where to look for the model (with a little help from the altitude and distance).  If I can't spot the model, I can basically fly back towards the launch position as graphic 2 tells me which direction to move (if the `+` is to the upper left, I move my right stick down and right).  Once the model is away from the sun or I can find it, I can fly it home.

I've also used graphic 2 to find a downed model.  I just walked in the direction of the dotted line towards the `+` for the number of feet the distance measurement states.  My stride is about 3 feet so I can count off my steps and find the model with ease (when I know I'm close, I turn on the beeper).

Let me know if this helps and if there's a better way to explain it to understand it better.

teckel12, 2018-07-18:
@pkerney BTW, I've been adding a new view designed for fixed wing and similar to a pilots view. It will be included in v1.4.0

closed 2018-08-21 by teckel12