teckel12/LuaTelemetry #85: iNav with Dragonlink V3

marcdornan, 2018-01-08:
Apologies of this is not directly related to an issue (yet!) but I do not know another way to contact you though gitHub.

I want to use this with Dragonlink V3 which obviously does not support S Port telemetry natively. Do you think I can use Mavlink and then convert to S Port telemetry on the ground side with a Teensy 32 (I have done this successfully many times with Ardupilot to get Taranis telemetry). Can you think of any showstopper in your LUA if I use Mavlink iNav telemetry, transmit it through transparent serial on DL v3 and then use a Teensy to convert to S Port telemetry. I am wondering if Flight Modes will work, for example.

teckel12, 2018-01-09:
@marcdornan As long as you are creating SmartPort telemetry using the same sensors and the same information it will work.  For example, with flight modes you would need to duplicate the flight mode bit logic (see `FSSP_DATAID_T1` and `FSSP_DATAID_T2` in https://github.com/iNavFlight/inav/blob/development/src/main/telemetry/smartport.c).

With that said, I'm not familiar with Dragonlink nor Mavlink to know what snags you may have.  SmartPort telemetry in INAV is passing a LOT of information and LuaTelemetry uses most of it.  So, all the flight mode stuff (for example) would need to be available and duplicated in order to work (or have full functionality).

marcdornan, 2018-01-09:
> main/telemetry/smartport.c).

Well Dragonlink can transmit transparent serial so that should in theory be
ok.

I will try iNav Mavlink to SPort conversion and see what the LUA script
does and see if we need to change iNav Mavlink or the conversion process to
support this. This is all very similar to what Ardupilot users have been
doing for years to get LUA scripted telemetry on a Taranis.

closed 2018-04-18 by teckel12