teckel12/LuaTelemetry #86: Crossfire Support

reptil1990, 2018-02-02:
Hey,

When do you think is Crossfire implemented into the Lua and iNav?
It gained big popularity and with the new wing of TBS, i think a lot of iNav builds are coming!

Thank you!

teckel12, 2018-02-02:
See #36 

Basically, Crossfire doesn't send enough telemetry information for LuaTelemetry to function.  Specifically, flight modes and flight mode modifiers are no where near complete.

From the looks of it, Crossfire firmware would need to be enhanced before INAV could pass this information.

So, it's not that LuaTelemetry needs to be updated to work, it's that Crossfire needs to be updated to support more features.  Then, INAV would need to be updated to support this enhancement.  Only after these were done, could LuaTelemetry use this information.

Anyway, probably best to ask Crossfire when they would be updating their firmware to support all the telemetry features that are supported in SmartPort, INAV, and LuaTelemetry.  I don't have any Crossfire hardware no need it, so it's up to others with the experience and need to address this with Crossfire.  

Feel free to take this up with Crossfire and get back to me.

KnuckleUpFPV, 2018-02-12:
This was working with my crossfire on 2.2.0. It gave me all the info the qx7 normally displays. When I updated opentx to 2.2.1 this stopped working with crossfire. I was hoping there was a fix. Any idea as to why it worked on 2.2 with crossfire telemetry on, but now it doesn't? Crossfire can also do mavlink, but im not sure if that helps in any way. I have all the sensors setup on my telemetry screen, but the inav lua is so much nicer. I also like the call outs from the lua. Thank you in advance. I love this lua. Great work!

teckel12, 2018-02-12:
@KnuckleUpFPV Sounds like a problem with 2.2.1.  I haven't tried it yet.  There's a max baud rate addition in 2.2.1 that's specific to Crossfire and the QX7.  Maybe that has something to do with it?  I'd look there first.

KnuckleUpFPV, 2018-02-12:
I'm not sure man. Now that I think about it, it was working on 2.2.1 with the baudrate at 115200. Then I did the inverter mod and now it isn't working. I forgot all about adding that inverter. It's the second gen qx7. Pins on bottom slow inverter. Only one that went crazy was capacity. It would read crazy high then low. I didn't get the lua from github though. It was in an rcgroup post. Said it was made to work with the qx7. It looks identical to the one here on github. Think its possible someone modded your lua? I couldn't find the rcgroup post again. This github is really all I find when I search for it.

In order for this to work, what should I ask trappy to add? 

I also upgraded my crossfire firmware after doing the inverter mod. Did it all in one day to get it done. I know thats a lot of variables. If there is anything I can do I am happy to help.

teckel12, 2018-02-12:
@KnuckleUpFPV I don't even know where to start as from everything I see, there's no way Crossfire would work.  I don't have Crossfire so I'm at a total loss here.  But, if you select Crossfire telemetry in INAV, I can't see how it would work as it's totally different than SmartPort telemetry.

LuaTelemetry was always working with the QX7.  It was actually designed only for the QX7 but later updated to support other transmitters.  There would be no reason to modify LuaTelemetry to work with QX7 so it doesn't make sense that someone would do that.  Unless, you were using an old version of LuaTelemetry before being included with the INAV project.  That repo would be here:

https://github.com/teckel12/Taranis-iNav-Lua

KnuckleUpFPV, 2018-02-12:
I never selected any telemetry in inav itself. Only turned telemetry on in crossfire menu on the module itself. Ch1 set to crsftx  and ch2  set to 
crsfrx.

Thanks for your time. I will switch over to lua pilot. 

Everything that the lua displays I have on telemetry screens, all back from the crossfire. GPS coordinates, heading, voltage, altitude, speed, distance from home, and a few others I can't think of off the top of my head right now. Takes 3 screens to display them. Your lua just made it nice and compact so they were all on one screen.

Again I truly appreciate your time. Maybe someday it will work again. Thank you.

teckel12, 2018-02-16:
@KnuckleUpFPV As long as all the telemetry sensor names are the same as what's expected with SmartPort it should work.  What I'm guessing is that if Crossfire worked before and it doesn't work now, it's because Crossfire no longer follows telemetry sensor convention as detailed here:

https://github.com/iNavFlight/inav/blob/development/docs/Telemetry.md#available-smartport-sport-sensors

This changed with INAV 1.8.0 and then I updated LuaTelemetry to match what INAV is not sending. 

Not sure how the crossfire works as I don't have crossfire to know.  But do you have any web links to a repo or webpage for crossfire where we could let them know they need to change something to be compatible with INAV 1.8.0+

KnuckleUpFPV, 2018-02-16:
I'll post this on the crossfire page to see if we can get some help

KnuckleUpFPV, 2018-02-16:
Some guy said the info is in the crossfire manual. I googled the manual and looked through it. All I can see if the sensor names. I dont see any of the packet info. I dont know if the sensor names in the manual help you or not. I appreciate your time.

teckel12, 2018-02-16:
@KnuckleUpFPV Looking at the [Crossfire manual](https://www.team-blacksheep.com/tbs-crossfire-manual.pdf) on page 39, it shows the telemetry sensors.  Of those, only Curr, GPS, and Alt match up with SmartPort telemetry which is used in LuaTelemetry.  This is why I've said that Crossfire won't work with LuaTelemetry nor could it have ever.  Basically, it doesn't send the required telemetry that is vital to LuaTelemetry.  Mainly, Tmp1/Tmp2 telemetry communicates with LuaTelemetry the flight mode, which controls everything else.

If you think that somehow OpenTX v2.2.0 really worked, I would suggest going back to that version and then we could look into it.  But, I would highly doubt that LuaTelemetry ever worked with Crossfire as it simply doesn't send the required information.  Unless something is different than what the manual states.

KnuckleUpFPV, 2018-02-16:
If its a naming thing maybe it's because it was an openlrsng profile before becoming a crossfire profile. And the sensors were already in the list. I'll try renaming sensors again or use another openlrsng profile and change it to crossfire to see if it starts working again

rhys100, 2018-03-09:
I loaded the lua on today and got a Script panic, not enough memory. Is this because im using crossfire? Does this only work with frsky smart port?

KnuckleUpFPV, 2018-03-09:
It's a memory issue with the transmitter. It's an opentx thing. Restart your tx and open inav lua first. It should load. If you use any other lua or open the telemetry screens first it causes the panic error or syntax error.

teckel12, 2018-04-05:
Yeah, panic errors like this can happen if too much is loaded on the transmitter. Make sure your transmitter has luac turned on and remove features you don't need.

closed 2018-04-18 by teckel12