teckel12/LuaTelemetry #88: Script panic on Taranis X9D

leocopter, 2018-03-12:
## Expected behavior

## Actual behavior

## Steps to reproduce behavior

## Screen shot or other information on behavior

leocopter, 2018-03-12:
Hi. I am using OpenTx 2.2.1 (with the corresponding SD card files) on a Taranis X9D. I get a "Script panic not enough memory" when I try to run the script. However, everything works well on my Taranis X7.

teckel12, 2018-04-05:
A script panic is an out of memory error. LuaTelemetry is pre-compiled as long as you install it as instructed. Look at the features you have turned on with the transmitter. Also, scripts and other things you have installed could be taking up all your transmitter memory.

teckel12, 2018-04-12:
@leocopter To further help, here's the section on the proper method of installing this lua script so it's pre-compiled and won't use as much memory (and void the script panic error).

https://github.com/iNavFlight/LuaTelemetry#inav-lua-telemetry-screen-setup

closed 2018-04-12 by teckel12