teckel12/LuaTelemetry #89: FrSky FPort Protocol Support?

SPlissken, 2018-03-18:
@teckel12 just noticed that FPort is due to hit production in 1.9.1.

Any chance your script will work with FPort?

The GPS feature has been super helpful, can't thank you enough for that!

teckel12, 2018-04-05:
It could work if FPort sends the required info. I would hope that they extended SmartPort telemetry as it's the most feature-rich, but I haven't looked into it yet. I'll look at the code and see how close we are.

teckel12, 2018-04-12:
@SPlissken It looks like FPort should work perfectly with this LuaTelemetry script.  It should work without any modification.  I'm not currently running FPort, but I'll test it out.  I'm using both X4R-SB and XSR-M receivers, so it seems I should be good with hardware.  My F4 FCs have hardware inverters, and my receivers are going through inverters too, so I'm not 100% sure I can just flash and go.  The documentation isn't 100% clear if I must bypass the inverters or not.

In any case, it should work, but I haven't tested it yet (as it's just getting warm enough to fly again).  Any insight you have on how to use the X4R-SB and XSR-M without modification on F4 FCs with hardware inverters would be appreciated.

SPlissken, 2018-04-12:
@teckel12 Good news!

I haven't had the chance to test it out yet either as I rely on LuaTelemetry quite a lot but looking at this it seems quite straightforward:
https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol

> a free hardware port on the flight controller (F3 or better) that is capable of running SmartPort with your receiver (i.e. must be able to run inverted bidirectional, or your receiver needs to have the 'uninverted SmartPort' hack applied).

and 

>  (Effectively, this connection uses the same pins on both sides that would be used to connect SmartPort if a non-FPort firmware was used.)

SPlissken, 2018-04-25:
@teckel12 I've done some preliminary testing and I can confirm that all seems to be OK. The only thing which is slower/problematic after the FPort switch is reading PID information via betaflight-tx-lua-scripts but that is nothing to do with this and is slow on BF too.

wolfpackt99, 2018-04-27:
seeing issues on xsr Hacked on UART 6 of Omnibus F4 Pro V3. Discovered new sensors but seems to be missing some required by the script?
Taranis X9D.

SPlissken, 2018-04-27:
@wolfpackt99 Have you tried to delete sensors first and then discover sensors with it fully powered and armed?

SPlissken, 2018-04-27:
@teckel12  I would just leave everything connected as you have it for S.Port and play with serialrx_inverted to see if it works, from what I've read it might work and save you some rewiring. I'm on an F3 so I didn't have to change anything apart from the firmwares. Sounds like you enjoy rescuing your quads too much, maybe you are subconsciously sabotaging them...

wolfpackt99, 2018-04-27:
so.. this morning i deleted all sensors and re-discovered. a few new ones came in. I have 17 sensors now.
changed the name of 0420 to Dist. Getting attempt to perform arithmetic on field 'he

i can't see past the 'he

teckel12, 2018-04-27:
Maybe your Taranis log files would be helpful.  I assume you're using an QX7 transmitter based on the shortened error.

Also, are you doing the install as stated in the instructions (which installs the pre-compiled lua script instead of the source script)?

teckel12, 2018-04-28:
@wolfpackt99 Just to be sure you're using the latest, I released v1.2.5 today.  While this probably won't correct any issues, I just want to be sure you're using the latest version for testing.

SPlissken, 2018-04-30:
I'm receiving the following sensors from 1.9.1 F.Port on opentx-x7 2.2.0:

AccX
AccY
AccZ
Tmp1
Tmp2
A4
VFAS
CURR
Alt
Fuel
VSpd
Hdg
RSSI
RxBt
F000
GSpd
GPS

teckel12, 2018-04-30:
Three issues:

- These sensors from S.Port are missing: `Dist` & `GAlt`.
- `CURR` is in all caps while with S.Port it's mixed case: `Curr`, but maybe a typo.
- Sensor `F000` is not part of S.Port telemetry, maybe that's something else?

teckel12, 2018-04-30:
Looking at the code, there's no good reason why `Dist` & `GAlt` would be missing, and it seems that `CURR` was a typo.

I would suggest deleting the `CURR` sensor if it's all caps, get a GPS lock, then search for sensors again which will re-create the `Curr` sensor (and hopefully mixed case) and it *SHOULD* create the `Dist` & `GAlt` sensors.

And insight into the `F000` sensor if you have any.  Maybe deleting that one too and see what happens when you search for sensors.  If it gets recreated the same, the two numbers in the `ID` field should identify what it is.

Thanks!

SPlissken, 2018-04-30:
CURR was a typo. It took me a few goes of delete/discover before 0420 showed which I think happened when I activated POSHOLD during one of the tests. (Off-topic but two-way MSP over F.Port seems a little temperamental compared with S.Port).
F000 is still there and now there is 0810 too. 

A4
VFAS
Curr
Alt
Fuel
Hdg
AccX
AccY
AccZ
Tmp1
F000 (ID F000 27)
VSpd
RSSI
RxBt
Tmp2
GSpd
GAlt
GPS
0810
0420

SPlissken, 2018-04-30:
I think F000 is something to do with bi-directional communication for the R-XSR and possibly other RXs: I've certainly seen it before.
Any ideas on 0810?

teckel12, 2018-05-01:
Rename `0420` to `Dist` or just run 1.2.5 of Lua Telemetry and everything should work. I'll look into the other sensors but they could just have something to so with your RX. I get an extra A2 sensor for voltage for example.

SPlissken, 2018-05-01:
Yes, I saw the change in the code for 0420.

A2 used to be quite handy but it is a bit redundant now with accurate flight controller telemetry. If you use it keep the input under 3.3v, at least on a D4R-II.

teckel12, 2018-05-02:
@SPlissken I wire up the A2 with a voltage divider because why not?  I think of it as a backup and to verify the INAV telemetry battery voltage output.  I've considered building it into Lua Telemetry, but I question if anyone even does it anymore with the "complexity" of adding a single voltage divider resistor.

SPlissken, 2018-05-02:
I just use a standard mini potentiometer for A2 which can be calibrated, is light, costs pennies and can be bought at lots of places. In my experience FBVS-01 isn't very accurate and is quite bulky for what it is.
Until teckel12 fixed it, the voltage code in frsky.c wasn't working so A2 was pretty much obligatory if you wanted to remotely monitor voltage, accurately at least.

teckel12, 2018-05-02:
I'm fairly sure I just threw two resistors at it to make a voltage divider and keep in in range, then did the calibration on the transmitter.  Yeah, I change `Ratio` and `Offset` on the transmitter for the `A2` sensor for calibration.

Another "trick" for voltage is to turn on the `Filter` option for the battery telemetry sensors.  This evens out all the spikes.

SPlissken, 2018-05-02:
I prefer a screwdriver to opentx!

teckel12, 2018-05-13:
Added in #108

teckel12, 2018-05-14:
Merged into [1.3.0](https://github.com/iNavFlight/LuaTelemetry/releases/tag/1.3.0)

closed 2018-05-14 by teckel12