teckel12/LuaTelemetry #95: Add support for FrSky D-series receivers

teckel12, 2018-04-24:
See: [Modify FrSky D-series telemetry for compatibility with Lua script](https://github.com/iNavFlight/inav/pull/3116) from @alexeystn

Only README changes should be required for LuaTelemetry:
* [ ] Note that D-series receivers are supported
* [ ] Add list of common supported D-series receivers
* [ ] Note that on D-series receivers the `07` sensor name is changed to `Dist` (like how the `420` is changed with X-series receivers

SPlissken, 2018-05-02:
Happy to test, I've got the hardware ready to go.

teckel12, 2018-05-09:
@SPlissken I can make a build for you with the changes.  What target do you need?

teckel12, 2018-05-10:
@SPlissken The [PR for the D-series telemetry](https://github.com/iNavFlight/inav/pull/3116) hasn't been merged to development yet, so it's not part of 2.0.0.  Attached is the D-series PR for SPRACINGF3
[inav_1.9.1_SPRACINGF3.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/1991651/inav_1.9.1_SPRACINGF3.hex.zip)

SPlissken, 2018-05-11:
@teckel12 All looks good. I don't have a current sensor on the build used for testing but the rest of the data came through fine and displayed correctly on an X7. I didn't validate any of the telemetry data displayed in the script but the voltage seemed accurate and the data looked meaningful.

teckel12, 2018-05-11:
@SPlissken 👍

teckel12, 2018-05-13:
Added in #108

teckel12, 2018-05-14:
Merged into [1.3.0](https://github.com/iNavFlight/LuaTelemetry/releases/tag/1.3.0)

closed 2018-05-14 by teckel12