teckel12/LuaTelemetry #102: Incorrect designation in the menu when specifying the maximum height. ### f instead of ### m.

Pacman111178, 2018-04-26:
## Expected behavior

## Actual behavior

## Steps to reproduce behavior

## Screen shot or other information on behavior

Pacman111178, 2018-04-26:
Incorrect designation in the menu when specifying the maximum height. ### f instead of ### m..
I just tried with my X9d + synonymous with a log file the voice output .. but unfortunately did not say anything where he was at 130m ...

Pacman111178, 2018-04-26:
Sorry my Problem when i actvate the Baro alt it is ok ... with only the Gps Alt it not work !

Pacman111178, 2018-04-27:
> @Pacman111178<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FPacman111178&data=02%7C01%7C%7C80873ef58204485e981108d5ab813c30%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636603493903591415&sdata=MabTtYFerCbqPjiYJXTNrDCTHzmJHwx4ConTzyz5t9I%3D&reserved=0> Lua Telemetry uses the unit that you specify on your transmitter for the telemetry sensor. If it's showing V, it's because you've set it to V. To set it to meters m you just need to change it. Follow these instructions from your transmitter (QX7 was used for these instructions):
>
>   1.  From the main screen short press Menu
>   2.  Press the Page till you get to the TELEMETRY page
>   3.  Scroll down to the Alt sensor and press Enter
>   4.  Scroll down to the Unit option and press Enter
>   5.  Scroll to the m unit for meters and press Enter
>   6.  Hit Exit 4 times to return to the main screen
>   7.  Turn transmitter off and turn back on
>   8.  Long press Page to return to the Lua Telemetry screen
>
> You may also want to verify the units for Dist, Alt, GAlt & Gspd as well to make sure you've set them correctly. They default to correct units, but they could have been changed at some point. See this transmitter setup<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FiNavFlight%2FLuaTelemetry%23from-transmitter&data=02%7C01%7C%7C80873ef58204485e981108d5ab813c30%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636603493903591415&sdata=7l5dbRaJE7vqjoBGrOvmd0IuXHfkbnwY3i99smiVxaU%3D&reserved=0> for more information.
>
> [... cut ...]
> From the INAV source code:
>
> if (sensors(SENSOR_BARO)) {
>     smartPortSendPackage(id, getEstimatedActualPosition(Z));
> }
>
> What this is saying is to only send the Z position (axis) which is the altitude, if you have a barometer sensor. So, not showing the altitude is correct if you don't have a barometer.

Hello Tim .
Thank you for your prompt reply . The Baro is mostly not used on the plane because he made a lot of bugs at that time, is also in the Inav Wiki, would be nice if the Lua would fall back on the Gps sensor in case of missing Baro sensor. I have nowhere set the unit V, delete your alt sensor and you have the same. Find your Lua perfect, keep it up!

teckel12, 2018-04-27:
@Pacman111178 Version 1.2.5 of Lua Telemetry supports a fall-back to using the GPS for altitude if a barometer isn't available.  Also, it fixes the `V` unit when a sensor isn't found.

Pre-release version 1.2.5 is available here: https://github.com/iNavFlight/LuaTelemetry/tree/development/TELEMETRY

teckel12, 2018-04-28:
Resolved with PR #106 v1.2.5 now available for [download](https://github.com/iNavFlight/LuaTelemetry/releases/latest)

closed 2018-04-28 by teckel12