teckel12/LuaTelemetry #103: script syntax error bad argument #1 to 'writ

teckel12, 2018-04-28:
Issue initiated by @shaiss

All I see is "script syntax error bad argument #1 to 'writ

teckel12, 2018-04-28:
Based on the error that you provided, it greatly isolates what the problem could be.  It seems that it's trying to write to the `config.dat` file but it can't.  My guess is that's because you didn't copy over the `iNav` folder that's included in the `TELEMETRY` folder.

Both the `iNav.lua` and the `iNav` folder need to be moved to the TELEMETRY folder on your transmitter's SD card.  See the [INAV Lua Telemetry Screen Setup](https://github.com/iNavFlight/LuaTelemetry#inav-lua-telemetry-screen-setup) for more details (specifically step 2).  I'll clarify step 2 to state that **BOTH** the `iNav.lua` file and `iNav` folder must be copied.  I'll also look into creating an error if this folder doesn't exist (or at least not causing the script to crash).

Let me know what happens.

teckel12, 2018-04-28:
I can confirm that if the `iNav` folder doesn't exist the script will indeed throw this error.  This will be addressed in v1.2.5

teckel12, 2018-04-28:
@shaiss I've created a more graceful error message if the `iNav` folder doesn't exist.  You can download this development version [here](https://github.com/iNavFlight/LuaTelemetry/tree/development/TELEMETRY).

But, you'll also need to move the `iNav` folder to your transmitter's SD card as well to get past the more graceful error message.  The `iNav` folder contains the sound files that Lua Telemetry uses as well as where it creates a config file used for your custom configuration.  The error happened because it previously assumed the `iNav` folder existed and tried to create your config file the first time you ran Lua Telemetry.  With 1.2.5 it now first checks to see if it can create the file before writing to it.  If it can't create the config file, it throws a graceful error telling you what's up.

teckel12, 2018-04-28:
Resolved with PR #106 v1.2.5 now available for [download](https://github.com/iNavFlight/LuaTelemetry/releases/latest)

closed 2018-04-28 by teckel12

shaiss, 2018-04-29:
I'll give it a try tomorrow and report back!  Thx for addressing!

teckel12, 2018-04-30:
@shaiss If you could, just update the `iNav.lua` script first, which should give you a more specific message (and not crash).  The upgrade to 1.2.5 won't really fix the problem, but it at least will tell you what's wrong.  Then, move over the `iNav` folder from to the `TELEMETRY` folder, which is the real solution to the issue.  But, any feedback would be appreciated.

shaiss, 2018-04-30:
will do that over the next day or two and get back to you!