teckel12/LuaTelemetry #109: Not enought memory Taranis X9D + 1.2.5 (OpenTX 2.2.1)

David-VG, 2018-05-03:
Hi,

I know this is OpenTX related but just to let you know:

I get this message with OpenTX 2.2.1 with the last 1.2.5. Not a problem with 1.2.4 or earlier scripts:

"Script panic
not enougth memory"

There are a similar issue here:
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/97

I dont get the same error at the simulator, just with the real TX.

teckel12, 2018-05-03:
@David-VG How are you installing the script?  Are you moving the file from the `src` folder or `TELEMETRY` as noted in the instructions.

Basically, the file in `src` isn't pre-compiled, and it will take more memory to compile.  The one you should be using is the one in the `TELEMETRY` folder, which is pre-compiled and therefore uses less memory.

Out of memory isn't an issue with Lua Telemetry, it's either that you didn't install the correct script or you have too much running on your transmitter (there's very limited memory).  Newer versions will use slightly more memory as well.

Anyway, please follow the [INAV Lua Telemetry Screen Setup](https://github.com/iNavFlight/LuaTelemetry#inav-lua-telemetry-screen-setup) instructions.  Also, if you could review the final item in the [Tips](https://github.com/iNavFlight/LuaTelemetry#tips) to be sure that answers your question.

David-VG, 2018-05-03:
Hi @teckel12 ,

Yeah, i tried both methods (usually used the pre-compiled one but today upgrading to 1.2.5 and noticing the memory issue also tried the lua ones). When I use Inav i only use the Lua Telemetry precompiled scripts. 

While using inav, long time ago i was using the old (and smaller) betaflight vtx sript but i dont use it as a Telemetry Screen now, just Lua Telemetry as a Screen.

Really there is an issue related to OpenTX memory and now there is less memory avaliable for the X9 model. With some speciffic firmwares some scripts works (happens also with betaflight-tx-lua-scripts).

I will try those binary firmwares + the luac and will give some feedback.

David-VG, 2018-05-03:
Well, trying opentx-x9d+-noheli-lua-luac-internalppm-en-2.2.1.bin (as I commented at the betaflight issue) the script works again. :/

teckel12, 2018-05-03:
@David-VG You could also remove the `luac` to save more space.

closed 2018-05-03 by teckel12