teckel12/LuaTelemetry #112: Fuel Percentage

mrduncan626, 2018-05-21:
If I'm interpreting the Fuel Percentage from the information I see in this script description correctly, this value is based on the battery capacity provided in iNav and setting all 3 battery capacity parameters to "0" will disable this function.  However, the Fuel Percentage value in the Lua displays is "0" which results in a warning on the radio when the aircraft is armed.  However, if I just leave the radio and iNav aircraft on for about 5 minutes, the Fuel Percentage will continue to increase and after this value has increased above 20%, the radio warning goes away when the aircraft is armed.  

If I set the battery capacity in iNav to the capacity of the battery I'm using and set the critical and warning parameters, the script still indicates a Fuel Percentage of "0" but continues to increase over time.

What can I do to totally disable the parameters which control the Fuel Percent reading?

teckel12, 2018-05-21:
@mrduncan626 If you don't want the Fuel gauge, remove the `Curr` telemetry sensor.

Do you not have a current sensor?  In previous releases of INAV, it wouldn't output the current telemetry if you didn't have a current sensor.  I believe in the latest release of INAV this changed so the current sensor could output the battery voltage.  I'll have to look into this.

If you have a current sensor, you do need to set a few things up so INAV output's the battery percentage.  If you have a current sensor, I can walk you through that as you'll want the fuel reading instead of the inaccurate battery voltage.

mrduncan626, 2018-05-21:
Yes, the Omnibus F4 Pro V3 has a built-in current sensor.  Any help in setting this up would be great but I'm really not looking for a functional Fuel Percentage reading/output.  But, if Fuel Percentage is required for the LUA to operate, that's ok also.  However, problem is not an inaccurate battery voltage reading but rather an inaccurate fuel percentage reading.  FYI, I did setup the battery capacity parameters but this had no effect on the Fuel Percentage issue as described above.

teckel12, 2018-05-21:
@martinbudden Okay, so you don't want to remove the fuel gauge?  I read your question "What can I do to totally disable the parameters which control the Fuel Percent reading?" as you wanted to remove the fuel gauge.  You're also saying you're setting values to zero to disable the fuel gauge.  Just confused if you want to remove it or make it work.  To remove it, remove the `Curr` telemetry sensor from the transmitter.

Anyway, if you want the fuel gauge to work correctly, you need to do some setup.  First, you need to be sure your current sensor is calibrated correctly.  This in itself can be a challenge.  There's some YouTube videos on how to do this if it hasn't already been calibrated.

Once the current sensor is calibrated, you need to set a couple CLI settings.  `battery_capacity` needs to be set to the mAh you wish to draw from your battery.  Next (with INAV 1.9.0+) the `battery_capacity_unit` needs to be set to `PERCENT`.  If you were using LuaTelemetry with previous versions of INAV, you turned on the `smartport_fuel_percent` setting.  This parameter was refactored in INAV 1.9.0 to allow for more flexibility by using the `battery_capacity_unit` setting instead.  

Also, the critical and warning values are for the OSD only.  That information doesn't get passed down to the transmitter.

This should be spelled out in the readme section [In INAV Configurator](https://github.com/iNavFlight/LuaTelemetry#in-inav-configurator).  If there's a problem with these instructions or they could be worded in such a way to make more sense, please let me know.

mrduncan626, 2018-05-21:
OK thanks.  Not looking to remove the Fuel Percentage.  I'll configure it as you are suggesting and read through the link you have recommended.

teckel12, 2018-05-21:
@mrduncan626 Great to hear!

closed 2018-05-21 by teckel12