teckel12/LuaTelemetry #114: 18650 "Cell Low" and "Cell Critical" allow lower than 3.1v

davidngrc, 2018-05-25:
now I can only set to 3.2v for "Cell Low", and 3.1v for "Cell Critical".
but some 18650 like  sanyo ncr18650ga 3500mah, can go like 2.5v
SONY VTC6 can even go like 2.0v

teckel12, 2018-05-25:
@davidngrc I was trying to save people from theirself. Also to clarify, critical shouldn't be the lowest the battery can go.  At the critical level you still need to return and land.  So even though Titan batteries can dip to 2.5v, you should really never be flying at that level, you should have already landed.  And a critical warning once on the ground isn't going to do a lot of good.

There's also the issue with knowing the number of battery cells if the voltage drops this low.  Cell count isn't sent from INAV, so the calculation is done in LuaTelemetry.

I have some ideas about how to do this, but I still feel allowing a user to set a minimum voltage of 2.0 could cause problems.  Maybe 2.6v at a minimum (once I solve the cell count issue).

Thanks for the suggestion!

teckel12, 2018-05-26:
@davidngrc I believe I sorted the cell count issue out and can lower the minimum voltage warning to 2.6v.  Really don't feel confident about letting people set it to 2.0v.  I'd need more evidence as everything I see even with FW flights for 4 hours is that they don't let SONY VTC6 go to 2.0v.  Seems a critical level of 2.6v is reasonable unless you can find links to support lower levels.

teckel12, 2018-05-26:
To do:

- [ ] Make sure `data.cells` is recalculated when telemetry or models are reset.