teckel12/LuaTelemetry #115: Add support for Taranis X-Lite transmitter

teckel12, 2018-05-26:
- [ ] `lcd.drawLine` 6th argument support
- [ ] Allow up/down arrow keys to work to change view or high/low values
- [ ] Button replacement for menu button on QX7 and X9D

teckel12, 2018-05-27:
Just waiting on the release of OpenTX 2.2.2 and the X-Lite transmitter...

ReganGTX, 2018-08-25:
Not sure if this is the right place to ask this but I've followed the steps on this and I can't get the lua telemetry script to show up, I'm running OpenTX 2.2.2RC5 on the X-Lite and Inav lua telemetry v1.4.0, I added both file and folder to my radios \SCRIPTS\TELEMETRY directory and added the script to my display screen and nothing, tried deleting sensors and rediscovering them after GPS 3D fixed, to no avail.
Any help would be much appreciated as this is my last step to get everything set up.
My setup:
Matek F411-Wing on Inav 2.0.0 latest 
Frsky X-lite OpenTX 2.2.2RC5/ R9 Lite module with latest Frsky firmware/ lua telemetry v1.4.0
Frsky R9 Slim+ with Fport firmware