teckel12/LuaTelemetry #116: [feature request] Support for vtx smartaudio

matoy, 2018-05-27:
Hi,

That would be great if there would be an option to change the VTX settings (band, channel, and mostly power) from the options menus for smartaudio VTX devices.

I know it's possible with BF's lua script here:
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/X9/vtx.lua
and it works very well.
i guess many people like to fly with 500/800mw power but when not flying (for example, when waiting for GPS to get satellites), it's better to reduce power to 25mw so VTX doesn't get too hot.

i hope you'll consider this request positively :)
thank you anyway, great lua script.

cheers.

NB: loading BF lua script on inav firmware works for changing VTX settings page but the problem is that you can't load BOTH inav lua script in one screen and bf lua script in another screen because there's not enough memory on the RC (taranis x9d+).

teckel12, 2018-05-28:
If they're both compiled, they may both work.

Also, adding the functionality of another script would probably result in the same amount of memory usage, so if there really isn't enough memory to run both, there's probably not enough to have both in the same script.

Finally, I don't have a VTX that changes frequency nor would I have a need for this (I always use the same frequency all the time for my fleet of drones). So, there's no ability or reason for me to even attempt this.  It would need to come from another developer.

closed 2018-05-28 by teckel12

teckel12, 2018-05-29:
@matoy Looked into this a bit more.

Probably what should happen is that someone should create a  fork of the betaflight lua script which strips out all the PID setting stuff and makes the script only for setting the VTX settings.  This shouldn't be too difficult and I could see it being very useful once the PIDs are dialed in.

Also, this is really outside of the scope of the Lua Telemetry script.  It doesn't really fit into what Lua Telemetry is designed to do, display telemetry and flight information from INAV.

One more thing...  There's a now merged pull request for OpenTX to increase the amount of available memory for the X9D transmitter.  This  should be part of v2.2.2.  

Oh, and another...  Make sure you're installing the Betaflight script correctly!  They provide compiled files (like Lua Telemetry) which will take much less memory.  But, if you're using the files from the `src` folder you may be using a lot more memory as a result.

Best of luck!

laurentopia, 2019-04-01:
I ran into out of memory issues as well and also thought that combining vtx control would use less memory. I'm thinking this might be the case if each lua screen is its own interpreter or VM.

If that's not the case, maybe a few info on how to reduce memory would help.
This is in openTX 2.2.3

Also you speak of compiled lua, are they the extension .luac?

teckel12, 2019-04-01:
@laurentopia You won't be able to use both Lua Telemetry and adjust PIDs or VTx from the controller at the same time. You can do one, then switch to the other, but there's no enough memory on the controller to do both.

You can, however, change the VTx from OSD and Lua Telemetry from the transmitter. That would be the best method at the current time.

I'd like to add VTx support in Lua Telemetry and there's an open issue on it. But no promises as there's limited documentation on doing it.

laurentopia, 2019-04-01:
Got it. How do Lua displays allocate memory on open TX? I couldn't find any doc, just a few examples.

teckel12, 2019-04-02:
@laurentopia When you say Lua displays what are you referring to? Do you mean the OpenTX pages where each page can have a lua script or telemetry? 

If you're referring to the views withing Lua Telemetry, they are all done custom by Lua Telemetry so there wouldn't be any OpenTX docs. Each view is loaded and run. When you change views, the old view memory allocation is cleared and the new view loaded and run. The config menu is also another view. It's done this way as it's the only method of having multiple views within the limited memory OpenTX provides to Lua scripts.

laurentopia, 2019-04-05:
Very nice. Makes it easier to debug things also. Can a telemetry lua script such as inav view or VTX also have lua scripts running in the background?
I'm just wondering, if memory gets deallocated when switching display views, how I can get 200b of freemem in that other case. #268

teckel12, 2019-04-05:
@laurentopia I believe you're confusing telemetry **pages** and the display **views** in Lua Telemetry. They are not at all the same.

Within Lua Telemetry, I manage the memory used for each view and clear the memory each time you change a view. No two views ever run at the same time. 

As far as telemetry pages in OpenTX, all telemetry pages are running at all times. Each page (lua script) can have background and foreground tasks. For example, when Lua Telemetry is in the background it will still give audible and vibration feedback. When in the foreground, it then creates the display. Each script can control this.

So, if you try to have BF Lua Script on one page and Lua Telemetry on another, both are running at the same time. There's no memory that is cleared between page changes because both are always running using memory.