teckel12/LuaTelemetry #120: Head wind or Tail wind display using GPS speed and Air speed values if possible.

ninja-zx11, 2018-06-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

teckel12, 2018-06-21:
@ninja-zx11 I believe wind speed is being worked on in INAV.  Once this is sorted out, this data could be sent as telemetry information.  I don't fly fixed wing so wind speed isn't of much value.  I also don't have a pitot sensor to work on the code if it was done in the transmitter.

If you have a pitot sensor, I would appreciate telemetry logs.  Turn logging on for all telemetry sensors and set your transmitter to log telemetry.  This is a good idea to do anyway if you're not already doing it.  In any case, if you can send me some telemetry logs I could use it for a simulation and write wind speed coding in the transmitter.

ninja-zx11, 2018-06-21:
I don't have fixed wing assembled for INAV yet but will be setting up soon with matek f405 wing F4 board , GPS and pitot tube sensor.Then i will generate the logs you mentioned.
Thanks for providing LUA telemetry on Qx7 Tx!!