teckel12/LuaTelemetry #122: Error message

ThomasH3, 2018-06-24:
I'm getting an error message when I try to load the script on taranis x-lite
Syntax Error .... not enough memory

teckel12, 2018-06-24:
Did you load the compiled Lua file as per the instructions? You can't use the script in the src folder, as that's not compiled and the transmitter doesn't have enough memory to compile it.

The other possibility is that you're trying to use other Lua scripts on the same model. While you can do that, there's not much memory to work with on the transmitter. For example, if you're trying to use the PID / vTX Lua script as well as Lua Telemetry, you'll probably run out of memory.

ThomasH3, 2018-06-25:
you are right .... I had second script running
I'm now getting an "bad argument" Error for line #6

teckel12, 2018-06-25:
@ThomasH3 I'm guessing the error is `bad argument #6` not line 6.  It sounds like you have older firmware in your X-Lite.  I know there was issues with older firmware on the X-Lite specifically with this error (they forgot to add argument #6 for a few commands).  I'm fairly sure this fix was included in 2.2.2 RC1, but I'm VERY sure it's in [2.2.2 RC2](https://www.open-tx.org/downloads.html). 

Here's the issue: https://github.com/opentx/opentx/issues/5909
And here's the fix: https://github.com/opentx/opentx/commit/535e41d43e54248629af6f4f79307e81bd7075f3

According to this it was included in 2.2.2 RC1, but I'd use 2.2.2 RC2 as there's many more fixes in RC2.

Confirm that this works, thanks!

ThomasH3, 2018-06-25:
perfect ... thank you ..... I will test it on the weekend

teckel12, 2018-06-30:
@ThomasH3 v1.3.2 moves the config menu into another script which reduces the amount of memory used.  This may help if you're low on memory.

ThomasH3, 2018-07-01:
Hello ... can you change the way "FUEL" is displayed. Maybe adding a selection on config menu. I've tested the script on Cleanflight and Betaflight. I can receive mah used rather than decreasing percentage.
On my X9E I can easily  read the telemetry info. Due to small screen (and font) on the x-lite  the output seems  a little overloaded. Maybe removing the "Heading" and some of the "GPS" details could give some space to increase the gauge size.

 I find the script very useful. …. a really great tool !!

teckel12, 2018-07-02:
First off Lua Telemetry may work somewhat on Cleanflight and Betaflight, most of the core operation will be lost because INAV outputs a LOT more flight information and it's been custom designed to work with Lua Telemetry. While we could improve Cleanflight and Betaflight to work perfectly with Lua Telemetry, I feel it's out of scope for those software packages as it's more geared towards FPV racing than GPS flying.

Second, the fuel was specifically designed to show a decreasing percentage for simple and accurate information to when your battery is low. Everything in Lua Telemetry is designed around this percentage of fuel remaining. While INAV can output mAh it can't be used with Lua Telemetry as it needs to know the percentage from the FC. In other words, that wouldn't work with much of the central logic of Lua Telemetry.

Third, I use the X7 transmitter and I don't find that anything is too small to read. I'd actually like a smaller font option. The heading and the GPS are some of the most useful screen information for me. The fuel, battery, and RSSI I don't really need to see as it gives audio feedback for every 10% fuel and when anything is low.

Honestly, with your wanting to use Cleanflight, Betaflight, don't care about GPS nor heading, it sounds like maybe you should just use mAh telemetry and setup a custom telemetry screen that will output exactly what you want. No need for Lua Telemetry or any Lua script at all for that matter. Just display telemetry info on a screen and get exactly what you want, and a larger font as well.

closed 2018-08-21 by teckel12