teckel12/LuaTelemetry #128: No GPS data if HDOP >2

Gregmaan, 2018-07-03:
There should be an option to get always gps data and not only if HDOP <2.
For rescue actions with bad HDOP every info is helpful.

teckel12, 2018-07-03:
@Gregmaan You still get GPS data if HOP is higher than 2.  Actually, Lua Telemetry v1.3.1 only knows that the HDOP is higher than 10 or less than 10.  Unreleased Lua Telemetry v1.3.2 gets the HDOP in 0.5 step increments from 5.0 to 1.0, but still doesn't restrict GPS data under a certain HDOP.  Basically, there's no option to adjust in Lua Telemetry.

I believe what you're referring to going on in INAV.  INAV doesn't get a GPS fix till it decides it has a lock and can set the home position.  Not sure exactly what the logic is in INAV, but it's not a HDOP of 2.0 as I got a GPS fix today from INAV and my HDOP was 4.0.  Anyway, Lua Telemetry waits till there's a GPS fix from INAV to show GPS data.  This serves a useful purpose, because if flying and GPS fix is somehow lost, Lua Telemetry will show the last good GPS fix.  If it would always show whatever GPS data it's getting, it could show VERY bad GPS data, for example, you would always crash on Null Island.

Anyway, if you could explain in a little more detail as to what was going on and happening, maybe I could provide a more clear explanation.

Gregmaan, 2018-07-04:
Hmm thats strange, I copy my comments from #3495:
@teckel12 When I connect the FC to the configurator I can see HDOP values. If the HDOP is >2 I get a warning on my RC ..Bad GPS and the Lat/lon is blinking(Or before first arm I see no GPS data). If the HDOP is <2 I got the message GPS good and I can see GPS lat/lon.
The configurator shows 3D lock in all cases
So for me it looks like there is a filter on HDOP ...

teckel12, 2018-07-04:
Nope, LuaTelemetry 1.3.1 thinks GPS is good when INAV sends it information that it has a GPS lock, home lock, and at least 6 satellites. The next release of LuaTelemetry will change this to only look at the GPS fix, basically giving even more control to INAV to determine if there's a GPS lock or not.

If there's logic of GPS lock based on HDOP, it's in INAV. But, I know it's not a HDOP of 2.0 as I get a GPS fix at 4.0, if not higher (that's just the highest HDOP I've seen).

Gregmaan, 2018-07-04:
So id did further tests...
Today I did not get any GPS Sensor data to the RC with HDOP<2.
Looking into the telemetry screen only a few sensors (Three) getting data (blinking stars in screen 12/13 X9D Plus)
Deleted all sensors and discovered them new. Now it looks fine.
Will do further tests tomorrow.
Under which conditions I do not get SPort data?
Physically the connection looks fine.
Is there a problem with the data transmission over softserial (default wiring Matek F405 wing)?
It is no Lua problem....

teckel12, 2018-07-04:
@Gregmaan If the sensors are not blinking, it means that the transmitter is not getting telemetry from INAV.  For GPS data, INAV doesn't sent any GPS sensor data till there's a GPS fix (you can't even discover sensors till you have a GPS fix).  It can't know coordinates till there's a fix so that makes sense.

Lua Telemetry also waits till INAV has reported that it has a home lock.  But, that happens almost immediately after fix, so it's just there as an extra level of security.  I'm going to look into this, but I've never seen the home lock lost, even if the GPS signal was lost.

Basically, it sounds like you could be having another problem, but it doesn't sound like it's Lua Telemetry doing it.

Gregmaan, 2018-07-06:
@teckel12 I did some further tests but nothing seems to solve the strange telemetry warnings.
I did not had the lost telemetry sensors again. But the warnings "GPS good" and "GPS bad" make me always nervous in flight. Allthough I can´t rescue my plane in agricultial areas, without GPS position (Beeper 10m range, and LOS is less than 5m).
I chatted over telegram yesterday....Michel gave the hint...reset your GPS to defaults via u-center..
But that does not cure the problem.
Are there any more details I can send, to get it solved?

teckel12, 2018-07-06:
I've never had "GPS good" and "GPS bad" while flying, not once.  With that said, v1.3.1 of Lua Telemetry does require at least an HDOP of about 10 and 6 satellites.  But, it also requires INAV to be reporting that it has a GPS fix.  If your HDOP is going over 10, you have under 6 satellites, or INAV is losing a GPS fix, you will get the warning (but like I said, I've never got that while flying).

Are you getting any I2C errors in the configurator?  It sounds more like a problem with a weak GPS signal or pull-up resistors needed.

With Lua Telemetry v1.3.2, it won't report "GPS bad" unless INAV loses a GPS fix (which comes from the GPS itself).  The minimum satellites and HDOP of 10 or less was removed.  To be honest, the HDOP of 10 or less is meaningless, which is why there was rework on this in v1.3.2.  From my logs, I always had a HDOP of 10 or less when there was a GPS fix.  So really, the HDOP was meaningless.  In v1.3.2 it has meaning.

If you want to run Lua Telemetry v1.3.2 and INAV 2.0.0 RC2 I can compile an INAV for you and get you instructions.  I'll just need to know what target your FC is.

Gregmaan, 2018-07-06:
Maybe testing 1.3.2 would be a good option to get this sorted out.
My target is MATEKF405SE (Wing).
I will provide telemetry logs from my radio. Maybe this can explain my problem..
Thanks!

Gregmaan, 2018-07-07:
...without backround fishing in the dark ....
On the Matek F405 Wing (SE) SBUS is connected to RX2 and Smartport is connected via Softserial to TX2.
There is a case with LTM telemetry commit: 7c301e67d4aa16d5ef2992e49cc7134a990eb7b4
Maybe this is my problem?

Gregmaan, 2018-07-07:
Another idea is the GPS configuration..
enable GPSPASSTHROUGH, go to U-Center, The message block PVT is not enabled.
I thought in passthrough i can see what GPS Autoconfigure changed?
But to follow my idea..disable autoconfigure and setup the gps manually I did not find any information to do this.
Attached the log with jumping "GPS good/GPS Bad" messages from the RC.
[Theer.xlsx](https://github.com/iNavFlight/LuaTelemetry/files/2172651/Theer.xlsx)

teckel12, 2018-07-08:
@Gregmaan You'll need to copy over two files to your transmitter to run Lua Telemetry v1.3.2:

1. [iNav.lua](https://github.com/iNavFlight/LuaTelemetry/raw/development/dist/iNav.lua) - To your `\SCRIPTS\TELEMETRY\` folder which replaces the current `iNav.lua` file
1. [config.luac](https://github.com/iNavFlight/LuaTelemetry/raw/development/dist/iNav/config.luac) - To your `\SCRIPTS\TELEMETRY\iNav\` folder, this is a new file

For assistance with v1.3.2, here's the [v1.3.2 README](https://github.com/iNavFlight/LuaTelemetry/tree/development)

And here's INAV 2.0.0 RC2 with the Lua Telemetry changes that will be part of the release 2.0.0:

[inav_2.0.0_RC2_HDOP_MATEKF405SE.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/2173291/inav_2.0.0_RC2_HDOP_MATEKF405SE.hex.zip)

teckel12, 2018-07-08:
@Gregmaan BTW, what GPS are you using?

Gregmaan, 2018-07-08:
Thanks for the files. I will test it today. Iuse the Beitan BN-220 GPS. I think it´s relativly common. SBAS and Galileo disabled.

teckel12, 2018-07-08:
I'm sure the GPS chip is a clone, and there's also good clones, like the Beitan BN-880 and bad ones. It's very possible that your GPS doesn't report a GPS lock till under 2.0 HDOP. I've ditched all my GPS receivers and only fly with the Beitan BN-880 after a few failed GPS receivers I've tried (even lost a model as a result).

Gregmaan, 2018-07-09:
I had stored some values in the flash of the module, so that my known values are "pre set" (MSG/PVT every 5th cycle).  But that also made no difference( I also checked that the values are really stored).
With the 1.3.2 I flew 3 packs (about 1 hour): Best thing no more cycling GPS messages! Thank you!
Some things that I found:
- High HDOP values are displayed as 99 (Configurator shows realistic HDOP. Maybe this is the problem of my GPS!)
- There is a huge delay, if the HDOP > warning value. Not so fine if you fly a copter or a Mission and should be prepared to go out of GPS based modes.
- Acoustic ouput seems to be not stable: Sometimes I got no GPS good info and no ready to fly message, Sometimes I got GPS good but no Ready to fly message. I did not get a battery information every 10% (90,60,40,30)
- It would be great if  the battery percent warnings are not based on the voltage if a current sensor is configured.(Or maybe my it works only based on Wh?).
I f you´re interested in more beta testers I would like to do.
Sorry for my poor english.

Gregmaan, 2018-07-09:
Fine, I understand some things better now.Thanks!
But the fuel thing...
I have a current sensor configured. If I set the max cell voltage higher then 4.2 maybe 4.4 I got 70% fuel with a full battery before takeoff. And the message output is not every 10% step.
You need not to answer ... I will look at the things more detailed.
Hopefully LUA 1.32 is coming with the next RC!
Thanks!

teckel12, 2018-07-09:
@Gregmaan INAV has new "full battery" detection.  If you set the max cell voltage too high, INAV believes your battery wasn't full when you connected it.  Therefore, the fuel percentages will start lower as a result and will be based only on the voltage.  I believe you should set the values to the default, which I believe are as follows:

```
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
```

INAV does this in case you accidentally install a half-used or not fully charged battery.  If it would rely on the mAh used, you may run into a situation where you run out of battery unexpectedly.  It's a good idea, and works well as long as you have the above values set correctly.

I'll look into the battery warnings.  It could be that it's been programmed to not report fuel levels if the battery wasn't full.  As then only the voltage is really looked at.  But, I'll need to check as honestly I've never looked at what happens with a battery that wasn't full (or INAV thought wasn't full).

Gregmaan, 2018-07-10:
Thanks, but it´s still not fitting with the consumed fuel:
I am running
set bat_cells = 4
set vbat_cell_detect_voltage = 425 (default 420)
set battery_capacity = 2100
set battery_capacity_warning = 420
set battery_capacity_critical = 315
In case of a full battery maybe the treshold seems to be too small: 
OSD shows 4.18, and after less than 50% consumed (OSD)  the telemetry says less says 30%.
For me this looks like battery warnings are still based on voltage in case of only 0,02 V offset to full.
Or the consumed mAh are only calculated between full and warning?

teckel12, 2018-07-10:
@Gregmaan what do you have vbat_max_cell_voltage set to? If it's too high, INAV will assume the battery isn't full and then fuel will be based only on voltage.

Lua Telemetry doesn't calculate the fuel percentage at all. It comes directly from INAV. On OSD you can display the battery percent and that will match what is shown in Lua Telemetry. If the battery percent isn't correct, there's a problem with the INAV config as that's where all the calculations are done.

Gregmaan, 2018-07-10:
@teckel12 Sorry I only copied a few lines of my diff and forgot if its default ........ It is default 420
OK I understand, no lua problem, THANKS!

Gregmaan, 2018-07-10:
@teckel12 I can understand the lua battery percent warnings now: The remaining capacity is calculated from full to critical. That explained the offset of about 20%. Not perfect, but setting it to a real critical value is OK for me.
Today I had the sensor lost problem again. Only rebooting the FC cures it. As you told me, I will come up with more detailed data for further analysis next time it happened.

teckel12, 2018-07-10:
@Gregmaan I've been trying to figure out how INAV calculates the battery level myself as it doesn't work the same in 2.0.0 as it did in 1.9.1.

Previously in INAV 1.9.1 the battery capacity I would set to about 90% of the full battery capacity.  Then, in Lua Telemetry I would set the low warning to 30% and the critical to 20%.  So, even if I would drain the battery to 0% fuel, it would really be at 10% of the full battery capacity.

With INAV 2.0.0, I guess I need to set the battery capacity to the full capacity and critical to 10% to simulate the same as what I did in INAV 1.9.1.  This makes it a little difficult to get the model to give a low current warning the same time that Lua Telemetry does, but with a little math it should work.

I'll test it out tonight.

Gregmaan, 2018-07-11:
@teckel12 It would be nice if Lua telemetry could use the low and critical amounts and the full capacitiy entered in INAV. So they should alert at the same time. But I think thats not new for you...

teckel12, 2018-07-11:
It's not realistically possible to pass values like this over SmartPort. I've added a new section to the readme "Suggested Battery Settings"

Gregmaan, 2018-07-12:
@teckel12 That sounds fine for me.
Mabe it is possible to get the RC3 for the MATEKF405SE and LUA 1.32?

teckel12, 2018-07-12:
Because of changes that can happen, the plan was to release Lua Telemetry 1.3.2 with the the release of INAV 2.0.0.   The changes that Lua Telemetry needs in INAV for HDOP are going to be part of 2.0.0 RC3 as it's already merged.  So you can flash INAV 2.0.0 RC3 and install the development branch of Lua Telemetry (which is basically the same process as installing a release version). 

Maybe I'll make a WiKi on doing Lua Telemetry upgrades and installing the development branch.

Gregmaan, 2018-07-12:
Thanks!, I´d looked at the dev branch, and I know how to update.

closed 2018-08-21 by teckel12