teckel12/LuaTelemetry #130: Half-full battery installed, what happens to fuel gauge and battery percentage warnings?

teckel12, 2018-07-09:


shellixyz, 2018-07-10:
If the battery is not full when plugged in the voltage is used. vbat_max_cell_voltage = 100%, vbat_min_cell_voltage = 0%. Same as with 1.9.1.

teckel12, 2018-07-10:
@shellixyz I plugged in my bench battery that was about 3.9v per cell and I got a fuel value of 100%. With 1.9.1 I'm fairly sure it reported the voltage as the percent in this case. But with 2.0.0 RC2 I got 100% fuel.

I'll try it again this afternoon as maybe I missed something.

teckel12, 2018-07-10:
@shellixyz I did a test using the same half-dead bench battery on two quads both running the same INAV compile (same target as well).  One reported the fuel at 43% while the other reported the fuel at 100% as I previously experienced.

But, I think I know what's going on.  The one that reports 100% flies with a 3S battery while the test battery is 4S.  So, it probably looks at the bench battery (which is about 15 volts) and thinks it's a 5v/cell 3S battery instead of a 3.75v/cell 4S battery.  Because it thinks it's 5v/cell, it's obviously 100% full.

I assume if I set `set bat_cells = 0` I'd get 43% fuel instead of 100% as INAV would do the calculation for me.  Or, I could also probably setup battery_profile 2 with the bench battery specs and set INAV to automatically select the correct battery profile.

teckel12, 2018-07-11:
Confirmed that it's due to the `bat_cells` setting.  Set this to `0` and the lower fuel percentage is used.

closed 2018-07-11 by teckel12