teckel12/LuaTelemetry #134: Please add voice notifications for altitude similar to % battery remaining.

ninja-zx11, 2018-07-13:
Is it possible to have voice notifications for altitude (eg every 5m or 10m) as well like battery % remaining.

teckel12, 2018-07-13:
I assume you fly fixed wing. As I don't, I don't know what's important and what isn't.

Would vario beeping be more informative? Would voice altitude reporting be everytime the altitude changed by 5, 10 feet or whatever, or would it report the altitude every several seconds?

I like the idea, just trying to understand what would work best.

ninja-zx11, 2018-07-13:
I fly both but i don't have OSD (i have connex prosight) on my quad and thought it will be useful for someone not having osd. and yes vario beeping would be useful for fixed wing pilots too!!
I think if we have an option of selecting voice altitude reporting everytime the altitude changes by 5,10,15 etc,then pilots can change it according to their taste.eg having an option of reporting @ 5,10,15 and 20 etc.Thanks

teckel12, 2018-07-13:
Should be fairly easy to implement. I'll look into it tomorrow. With all that's being added in the next release, I may change the version to 1.4.0.

teckel12, 2018-07-14:
Added help wanted label for testing of the vario beeper and voice notifications. I'm going to look for scripts that do vario beeping to see how they're implemented and compare to the one I wrote in a vacuum (as I don't fly fixed wing).

See above for installing the development branch (it's super simple moving 2 files to the transmitter).

Any feedback would be greatly appreciated!

ninja-zx11, 2018-07-14:
That's awesome Tim!! Just copied new files to my QX7 and will be testing soon.Set my notifications to 10m.Will report back soon.

teckel12, 2018-07-15:
@It looks like RC3 and RC4 introduce a few new problems.  One of which directly effects Lua Telemetry.  The Tmp1 sensor is mangled in RC3 and RC4.  This is the primary sensor used for flight modes.

Looking into it now...

teckel12, 2018-07-15:
False alarm.  Works fine if you set the correct receiver protocol!

teckel12, 2018-07-15:
See at #136 Variometer beeping and graph are all wrong, and need rework.

teckel12, 2018-07-15:
I late added the 50 m/ft setting. I'll verify it works.

teckel12, 2018-07-16:
Fixed the issue when setting it to 50m. Just upgrade to the latest development branch like before.

ninja-zx11, 2018-07-16:
Thanks.I will upgrade.

Gregmaan, 2018-07-27:
@teckel12 There is still a minor issue with the voice output of the height:
- Configured voice output every 20m
- Fly at e.g. 100m ->voice 100m
- Go down to 99m -> voice 80m
..and so on.

teckel12, 2018-07-27:
@Gregmaan Good catch!  Development branch updated which should take care of this.

closed 2018-08-21 by teckel12