teckel12/LuaTelemetry #137: No GPS FIX

fmartinezo, 2018-07-15:
**Describe the bug**
Since I have installed on a plate matek f405-ctr version 2.0.0.rc3 and activating the lua telemetry in my taranis the message does not GPS FIX appears when I have about 18 satellites and ready to fly.
My lua telemetry installed was v1.3.1 and works fine on this plate whith inav 1.9.1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

fmartinezo, 2018-07-15:
Thanks so much. It will be checked tomorrow

fmartinezo, 2018-07-16:
One question more. Is this development compatible with inav 1.9.1?

fmartinezo, 2018-07-16:
Both development files copied to it's folders as you say in your install development page. Checked now and the problem persist.

teckel12, 2018-07-16:
There's no longer a "**NO GPS FIX**" message with the latest development branch.  So, if you're still getting that, you're not running the development branch.  Please verify you're copying the two files to the correct location and you're restarted your transmitter.

fmartinezo, 2018-07-16:
@teckel12I have returned to check absolutely all things. I have deleted and re-downloaded the development branch files. I have deleted the sensors. I have copied each file back into its corresponding folder. I have rediscovered sensors and they appear to me 18. I have made the changes that you recommend for 0420 and those corresponding to meters, km, etc. I have restarted the transmitter. The lua inav is selected as page 1 and I do not select it every time
Everything remains the same, the message "no gps fix" continues appearing with 11 satellites. I do not know what else to do, it's driving me crazy.
Transmitter taranis x9dplus
Receptor r-xsr
Plate Matek f405-ctr
inav 2.0.0-rc3 release.
inav 2.0.0-rc3 release.

fmartinezo, 2018-07-16:
@teckel12 Hi, I tried another quad that has inav1.9.1, omnibus f4proV3 controller and x8r receiver and the branch development works correctly. I see the GPS coordinates perfectly in its three selection options instead of the "no gps fix" message This time i have 9 to 11 sats. GPS is a aliexpress black gps apm compatible instead of the beitian 880 which is what I have in inav 2.0.0rc3

teckel12, 2018-07-16:
@fmartinezo When you start Lua Telemetry, it shows the version (you can cycle power to the transmitter and go to Lua Telemetry which will show the version for a few seconds).  Since the "NO GPS FIX" is no longer in v1.3.2, I'm fairly sure you're running the latest release version 1.3.1.

To be clear, to install the development build, you must use the instructions on this page: https://github.com/iNavFlight/LuaTelemetry/wiki/Install-Development-Build  That includes using the exact links provided (the `iNav.lua` and `config.luac` links on the above page).  There's no need to delete anything else or make any other changes.  Just copy those two files and when you start Lua Telemetry the next time, it should say v1.3.2.  Others have used this procedure to install the development build and it has worked for them so I also know the procedure is correct.

fmartinezo, 2018-07-16:
Thank you very much for your last indications, that is precisely what was happening. I do not know what happened, but I downloaded the two development branch files correctly and installed them according to the instructions. I deleted the files that I had on the card and I downloaded again the development branch files and install them according to the instructions and now finally version 1.3.2 appears, when before the 1.3.1 appeared.
Tomorrow morning I will try on inav 2.0.0rc3 and inav 1.9.1 and tell you.
Thank you very much for your time and dedication

teckel12, 2018-07-17:
@fmartinezo 1.3.2 is designed for 2.0 only, which is why it's not released yet (till 2.0 final is released). It should work with 1.9.1, but there could be a few issues (GPS HDOP will show higher than it really is and there could be issues where position hold or 3d hold are displayed incorrectly).

Let me know how v1.3.2 works for you.

fmartinezo, 2018-07-17:
Hello, I tried this morning and the message no longer appears in 2.0.0rc3, I have also tested with 1.9.1 and apparently it goes well too. I have not been able to fly since where I live is a restricted area of flight.
For me it is very important to be able to see the GPS coordinates for the case of drone loss.
The theme is closed for me.
Thank you very much

closed 2018-07-17 by fmartinezo