teckel12/LuaTelemetry #141: show pitch in small number in pilotview

vavisimo, 2018-07-19:
Hi;
Im using  in flight pitch alightment on my plane , and the pilot view is helping a lot , but i would also prefer having it shown in numbers if im 5deg up or 10deg up or 3deg down as those small numbers are how to see on the thin line :)

**Describe the solution you'd like**
for me it would be ok to add the number where place alows it :)

Keep the good work Tim!

teckel12, 2018-07-20:
@vavisimo Glad your using and liking pilot view. I'm still working on enhancing it and a better indication of pitch is being worked on. The attitude specifically has a lot of enhancements planned. Working on doing it accurately as well as efficient as possible.

Obviously, there's other things missing, like fuel/battery/RSSI, etc. But those are fairly easy and it's just a matter of positioning, so I'm holding out till the heavy lifting is further along.

Keep an eye out for development merges, you'll probably find new features.

BTW, what transmitter do you use?

vavisimo, 2018-07-20:
Taranis X9D Plus

vavisimo, 2018-07-26:
Perfect, tried it and i love it!

closed 2018-07-26 by vavisimo