teckel12/LuaTelemetry #142: Add Cruise Mode

teckel12, 2018-07-20:
Newly added cruise mode is not delivered via frsky telemetry. Look into adding this in INAV and adding support in Lua Telemetry.

davidngrc, 2018-07-20:
I think yes.

in iNav source  \src\main\telemetry\smartport.c
you can see there is a function 
            case FSSP_DATAID_T1         :
                {
                    smartPortSendPackage(id, frskyGetFlightMode());
                    *clearToSend = false;
                    break;
                }

and this function exists in   \src\main\telemetry\frsky.c

uint16_t frskyGetFlightMode(void) 
{
    uint16_t tmpi = 0;

    // ones column
    if (!isArmingDisabled())
        tmpi += 1;
    else
        tmpi += 2;
    if (ARMING_FLAG(ARMED))
        tmpi += 4;
    
    // tens column
    if (FLIGHT_MODE(ANGLE_MODE))
        tmpi += 10;
    if (FLIGHT_MODE(HORIZON_MODE))
        tmpi += 20;
    if (FLIGHT_MODE(MANUAL_MODE))
        tmpi += 40;
    
    // hundreds column
    if (FLIGHT_MODE(HEADING_MODE))
        tmpi += 100;
    if (FLIGHT_MODE(NAV_ALTHOLD_MODE))
        tmpi += 200;
    if (FLIGHT_MODE(NAV_POSHOLD_MODE))
        tmpi += 400;
    
    // thousands column
    if (FLIGHT_MODE(NAV_RTH_MODE))
        tmpi += 1000;
    if (FLIGHT_MODE(NAV_WP_MODE))
        tmpi += 2000;
    if (FLIGHT_MODE(HEADFREE_MODE))
        tmpi += 4000;
    
    // ten thousands column
    if (FLIGHT_MODE(FLAPERON))
        tmpi += 10000;
    if (FLIGHT_MODE(FAILSAFE_MODE))
        tmpi += 40000;
    else if (FLIGHT_MODE(AUTO_TUNE)) // intentionally reverse order and 'else-if' to prevent 16-bit overflow
        tmpi += 20000;

    return tmpi;
}

and the flight mode is defined in \src\main\fc\runtime_config.h
as you can see   **NAV_CRUISE_MODE** , just change the above tmpi accordingly will work (I guest)
typedef enum {
    ANGLE_MODE      = (1 << 0),
    HORIZON_MODE    = (1 << 1),
    HEADING_MODE    = (1 << 2),
    NAV_ALTHOLD_MODE= (1 << 3), // old BARO
    NAV_RTH_MODE    = (1 << 4), // old GPS_HOME
    NAV_POSHOLD_MODE= (1 << 5), // old GPS_HOLD
    HEADFREE_MODE   = (1 << 6),
    NAV_LAUNCH_MODE = (1 << 7),
    MANUAL_MODE     = (1 << 8),
    FAILSAFE_MODE   = (1 << 9),
    AUTO_TUNE       = (1 << 10), // old G-Tune
    NAV_WP_MODE     = (1 << 11),
    **NAV_CRUISE_MODE** = (1 << 12),
    FLAPERON        = (1 << 13),
    TURN_ASSISTANT  = (1 << 14),
} flightModeFlags_e;

teckel12, 2018-07-22:
@davidngrc I know how the code works, as I wrote some of it. The question is not if the mode exists or how to write the code, but how the mode could be added to the existing telemetry sensors. Tmp1 is out of space, so that's not an option. The formatting in your post is unreadable to know what you're talking about also.

If you have code to submit, pleas format it as code so it can be read.

teckel12, 2018-07-22:
NAV_CRUISE_MODE

teckel12, 2018-07-24:
PR to add Cruise Mode to INAV FrSky/S.Port telemetry which can then be added to Lua Telemetry

https://github.com/iNavFlight/inav/pull/3647

teckel12, 2018-07-25:
Cruise mode added to LuaTelemetry development branch, requires above linked INAV PR, which should be included in the next release.

closed 2018-08-21 by teckel12