teckel12/LuaTelemetry #143: dev build causes syntax error in taranis

P-I-Engineer, 2018-07-21:
**Describe the bug**
x9d on 2.2.1

cannot load inav lua.

teckel12, 2018-07-22:
@P-I-Engineer Yes, documentation hasn't been updated with the changes over the last couple days. You now also need the menu.luac and view.luac from the development branch.

I'll update the documentation later today. If you know how to get around Github, you can go to the development branch and to the `dist` folder. Those are the files that you need for the latest development build.

teckel12, 2018-07-22:
@P-I-Engineer I had a few minutes so I updated the [Upgrade to Development Build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) instructions so it works for v1.4.0.  New luac files were created to **greatly** reduce used memory.  Sorry about the lack of timely documentation.  Enjoy and please give feedback on the pilot view if you're using it.

P-I-Engineer, 2018-07-22:
getting same nill value error. 

i've fully cleaned out all the inav stuff from the telemetry folder and re added 1.3.1, then added the 1.4 stuff.

teckel12, 2018-07-22:
@P-I-Engineer I forgot a file, `data.luac`, that would cause a problem! Wiki updated to include the file. Try again, sorry.

P-I-Engineer, 2018-07-22:
it loads! haha, thanks.

i'll try it out later tonight if the wind dies down

teckel12, 2018-07-22:
Windy here too, maybe in my area? Toledo, Ohio for me. Probably fly this evening as well when the wind calms down.

P-I-Engineer, 2018-07-22:
outside seattle, thanks for making this, it's really cool

P-I-Engineer, 2018-07-23:
I got satellites to test out the lua script. The new pilot view is cool and I like the decimal hdop. The pilot doesn't draw well on my taranis, but I have a non plus rev A, so my screen may not be able to handle the contrast.

1 suggestion, can you lower the low battery and critical battery cell count. I'm flying on a li-ion pack and will run my voltage lower than 3.5.

Thanks

teckel12, 2018-07-23:
@P-I-Engineer The pilot view is changing almost daily. For example, the ground is filled solid as of this morning. This may change how readable the view is as well on your transmitter.

I have the X7 transmitter, and only have the configurator to see what the X9D screen looks like.

closed 2018-07-26 by teckel12