teckel12/LuaTelemetry #152: What to do with home direction when overhead

teckel12, 2018-07-31:
When within 20ish feet, home direction currently disappears. Maybe better to display directly ahead instead of disappearing?

teckel12, 2018-07-31:
Home direction is always displayed.  Not perfect, but better.  Not sure what a perfect solution would be, so this will have to do.

ninja-zx11, 2018-08-01:
How about it changes alternately between  icon (looks like home) and home direction icon when it reaches home?Just a thought.

teckel12, 2018-08-04:
@ninja-zx11 I only have a home icon along the orientation at the bottom to show the direction of home. I felt this made the most sense to show the absolute amount that one would need to turn to go towards home.

The issue is when your heading towards home and then you're right on top of home (say within 20' or 6 meters). Probably not as important for fixed wing as quickly you'd be beyond the 20' and home would be behind you. But, for a quad you could be hovering overhead.

How I have it working now is that the home icon jumps to the center top, but I'm not overly happy with that either. Maybe make the home icon inverted or inside an inverted box? Some kind of subtle indication without it jumping to the top.

Ill try the inverted icon as an option.

teckel12, 2018-08-06:
It's currently a flashing home icon on the center/bottom of the attitude view (just above the compass heading).  That seems to flow well and not give an indication that you're going the wrong direction (when it was disappearing).  I guess I'm happy with it.

closed 2018-08-21 by teckel12