teckel12/LuaTelemetry #154: Get yaw, roll, pitch from INAV

teckel12, 2018-08-03:
Create switch in INAV to send yaw, roll, pitch instead of accelerometer data. This will provide more accurate position information and provide missing yaw which will assist with heading vector indication and wind direction.

davidngrc, 2018-08-03:
awesome...

small note, 
want to try the development Commits on Aug 2, 2018 build, but after putting the file in x9d, and try launch it, 
show script syntax error    ?:0    attempt to call a nil value.
I have already compile the iNave dev Commits on Aug 3, 2018 build 
@_@

davidngrc, 2018-08-04:
@teckel12 I did delete the old file(folder), and upload the new.

davidngrc, 2018-08-04:
@teckel12 I just pull my X9D sd card out, and delete the files and folders, then upload the new file and folder in X:\LuaTelemetry-development\dist\ to   SD CARD \SCRIPTS\TELEMETRY\

teckel12, 2018-08-04:
@davidngrc I'll make sure all the compiled files are up to date.

I develop on 3 different machines. And because OpenTX Companion does the compiling (and only does a compile if something changed) it's not impossible for a file to not get compiled when it should. I've also been working on a different branch the last couple days, which makes it more likely that the dev branch was left in this state.

In any case, I'll make sure they're all up to date.

davidngrc, 2018-08-04:
cant wai😄

teckel12, 2018-08-04:
@davidngrc Think it's sorted out. The iNav.lua recompiled, but not until I made a change to the source and changed it back. Who knows what the OpenTX Companion was thinking. I wish there was a command line way to compile OpenTX Lua scripts.

teckel12, 2018-08-04:
@davidngrc If you're using the pilot view and could get a screen shot of the display that would be awesome. It doesn't need to be special, just looking to see how visible the attitude display is on the transmitter as I only have the QX7.

I'm considering making the ground lighter, to insure the rest of the data is more readable. But I'm only seeing it in the OpenTX Companion so I'm not sure how well it's represented.

davidngrc, 2018-08-04:
just download your latest dev build "Commits on Aug 4, 2018
Why didn't you get compiled?"
I could not even run the first script, 
btw, I am on X9DP OpenTx 2.2.1

davidngrc, 2018-08-04:
when I go to "file manager" (long press menu), and execute the script there, it show error
```
script syntax error
iNav.lua
truncated precompiled chunk
```

teckel12, 2018-08-04:
@davidngrc Have you used the release Lua Telemetry successfully (not the development branch)?

davidngrc, 2018-08-04:
@teckel12 just download the latest master version "Latest commit 528d896  on Jun 30"
I long press the "page button" to launch, error
` syntax error ?:0 attempt to call a nil value.`
but if I go to the "file manager" (long press menu) and execute the inav.lua, this time it can run, showing the version is 1.3.1

I then download the release build on "released this on May 30"
I long press the "page button" to launch, error
` syntax error ?:0 attempt to call a nil value.`
but if I go to the "file manager" (long press menu) and execute the inav.lua, this time it can run, showing the version is 1.3.1

what happen???
I have 18408 bytes free memory, is this related?

teckel12, 2018-08-04:
It sounds like you're not following the install instructions. Remove iNav.lua and the iNav folder from the TELEMETRY folder. Then, follow the install instructions for the latest release (1.3.1). There's no commits or anything that you should be using. Just follow the RELASE instructions and it will work.

davidngrc, 2018-08-04:
yes, all the above, and still 
I long press the "page button" to launch, error
`syntax error ?:0 attempt to call a nil value.`
but if I go to the "file manager" (long press menu) and execute the inav.lua, this time it can run, showing the version is 1.3.1

teckel12, 2018-08-04:
@davidngrc One more thing I just thought of. Be sure you've deleted the config.dat file in the iNav folder. Who knows what is in that file at this point. It's possible there's data in there that v1.3.1 is not expecting.

That file would have been deleted when you did a clean install, but I'm not certain that was done either.

teckel12, 2018-08-06:
@davidngrc I worked on the development branch a lot yesterday and this morning I wiped everything from my transmitter and downloaded the latest development branch to my transmitter.  Everything started up just fine.  So I'm sure the development branch is working correctly if you want to try it again.

davidngrc, 2018-08-06:
@teckel12 I think I might mess up my x9d software/setting, so I just upgrade to OpenTx 2.2.2, 
and download the latest development branch, it work beautifully, awesome. 👍

closed 2018-08-21 by teckel12