teckel12/LuaTelemetry #156: Smart Audio VTx control support.

zector555, 2018-08-04:
Like to control VTx settings via smartaudio on INAV Lua.  Cannot run inav and bf lua at same time on taranis to adjust video, get memory error.

teckel12, 2018-08-04:
The BF Lua is bloated, uses too much memory. Taranis doesn't have much memory available to work with. Not the fault of Lua Telemetry. You could try the development build of Lua Telemetry as it uses less memory. But also possible that you'll just need to choose which script to add.

What really needs to happen is that someone needs to create a fork of BF Lua that only does the audio VTx control. But, I can't be the one as I can't test it as I don't have the need to change VTx frequencies.

Another possible option is to use INAVs OSD to change the VTx frequency. I believe this can be done (again, I can't even test this).

teckel12, 2018-08-04:
I just thought of something else, if your not using OpenTX firmware with the Lua and Luac options checked you may be using extra ram to compile BF Lua each time.

Make sure you're using OpenTX firmware with the lua and luac options checked. Remove Lua Telemetry from your telemetry screens. Run BF Lua on its own (be sure to enter the VTx panel). Shut down your transmitter. Then turn it back on and set the development version of Lua Telemetry as your second telemetry screen.

zector555, 2018-08-04:
I can run the BF or INAV lua separate, they both work for me.  I cant have bf lua and inav lua setup at the same time on my taranis.  PID control would be nice to have, but i think users after setup would have more use for vtx control.  I would be very nice if inav lua could control vtx all in on lua script.

teckel12, 2018-08-04:
@zector555 See my comments on using the dev release of Lua Telemetry which uses less memory. Also, the comment about making sure your have the lua and luac boxes checked for your OpenTX firmware, which will compile BF Lua and make it use less memory.  With those changes, it may work right now.

VTx control inside of Lua Telemetry is really out of scope. Probably the best solution is for a stripped-down BF Lua that only does the VTx control and not all the PIDs and layout stuff for different transmitters that's not required.

The other HUGE problem with adding VTx control inside of Lua Telemetry is that I don't have the need for VTx control (I always use the same frequency), nor do I even have a free UART, so there's no way I could test it. In other words, someone else would need to write the code. To date, I've been the only developer of Lua Telemetry, so someone wound need to step forward and write the code.

Basically, it can't happen as it is today. So following my previous suggestions are the only current options.

zector555, 2018-08-04:
@teckel12 Thanks for the info, and thank you for your time with lua telemetry.  Its very nice and handy, i can live without vtx control, just switch scripts on my taranis when i need to.  I did check / install the lua luac when i installed opentx.

teckel12, 2018-08-04:
@zector555 Try the development build of Lua Telemetry, it uses much less memory.

teckel12, 2018-08-10:
@zector555 Started a fork of the BF Lua script that would only do VTx control. Still work to do, but it's a start:

https://github.com/teckel12/Taranis-VTx

davidngrc, 2018-08-19:
@teckel12 just test https://github.com/teckel12/Taranis-VTx     master branch @ Aug 11, 2018 
with
OpenTx 2.2.2 (X9D Plus)
iNav 2.0.0RC6
vtx e708TM3 (TBS SmartAudio)
SP Racing F4 EVO clone (on tx2 pad)

it work, so cool.
by the way, should add some remark, like "long press menu to save"

teckel12, 2018-08-19:
@davidngrc That's great! I haven't made any code changes other than remove code and consolidate things. The strange interface (like long press  menu to save) is how the script normally works. I wanted to work on reducing the memory or even making it as a module that could be accessed directly by Lua Telemetry before making any changes (like centering on larger displays or making the controls more logical).

teckel12, 2018-08-22:
[Taranis VTx](https://github.com/teckel12/Taranis-VTx/releases/latest) released, enjoy!

closed 2018-08-22 by teckel12

access3424, 2018-11-06:
hi, i'm trying all your suggestions on the taranis qx7 and i still run out of memory. Is there anything else I can try?

teckel12, 2018-11-07:
@access3424 It's a memory thing. The more models you have, the more scripts running, they all add up. There's just not much memory to work with and Lua Telemetry and thr VTx script both use a lot of memory. Not much else I can do.

The source for my VTx fork is available to anyone who'd like to try and reduce the memory further. I don't do an kind of adjustable VTx (I always use the same frequency) so there's little incentive for me to work on this further. But, someone who needs this can work off what I've started.

Rclay11541, 2018-11-13:
Hi Tim, This is AWESOME! My VTX is right at the FCC limit so if i'm working on the drone in the field it's really great to be able to turn the power down. 

Any chance this can work with TBS Smart Audio? I tried it and I'm getting "No Telemetry" flashing on the bottom of the script. 

Thanks!

teckel12, 2018-11-13:
@Rclay11541 No idea as I don't use any of this technology. This is deigned as a starting point for those with this hardware to further reduce the memory footprint and enhance it further.

kbilicic, 2019-02-18:
Hey guys.
I haven't found any nice usable lua script besides BF which uses a lot of memory and because of that works only sometimes :)
I decided to do some work and create a framework which will aloow anyone create their own lua screens all in one script 

Check it out here, it's still work in progress, keep an eye on it if that is something what you need.
I'll post some videos in next few weeks
kb-lua-framework