teckel12/LuaTelemetry #160: On Development branch -> setting menu -> max altitude, long press + or - will change the value continuously.

davidngrc, 2018-08-06:
currently, press + key, then add one meter, if I want to add 1000 meter to the "max altitude", I need to press + key 1000 times...

is it possible use tick right stick to control this value change? ( like in OSD menu )

the same goes to "cell low" and "cell critical", "fuel low", "fuel critical"

teckel12, 2018-08-06:
@davidngrc Not if you have a QX7, you just spin the dial.  For feet it steps in 10 feet increments.  I didn't do this for meters because it's set to a default of an odd number of meters that equals 400 feet.

I don't see a good way to accomplish this.  Willing to see a PR that addresses this from someone with a X9D who thinks this is important.

davidngrc, 2018-08-06:
I really wish openTx run javascript, LUA script is alien for me.

davidngrc, 2018-08-06:
I got an idea, how about add one more option
altitude value    x1,     x10,     x100 option?

teckel12, 2018-08-06:
@davidngrc It wasn't as hard to implement the X9D & X-Lite hold as I had feared.  The latest development branch allows you to hold the +/- buttons which will repeat quickly, just like it works in the other menus on your transmitter.

davidngrc, 2018-08-07:
possible added to "cell low", cell critical"?
and increase the speed in "max altitude", like +10 meter or feet per 0.1 second?

teckel12, 2018-08-07:
@davidngrc I don't follow you.  The long-press works with everything in the config menu, not just the altitude.  Even scrolling through menu options works with long-press.

I don't control the speed of the long-press repeat, it's automatic and built into OpenTX (which is why it was easy to implement).  I could change the step from 1m to something larger, but then you couldn't select an exact height (like 122m for 400 feet).  Making it variable would be too complex and unneeded for a value that's probably set once.

Since this is something you probably set once, probably the way it is as good as it gets without losing functionality or greatly increasing complexity (which uses more memory, which is very limited).  Also, I would think that most people fly under 400 feet or 122m (the default).  Those that really fly to 1000m probably would just turn off the altitude alert, so the actual height isn't used.  But, if you really want to set a 5000m altitude ceiling, consider the 5 minutes it takes you to scroll to 5000m your punishment for flying way too high.

davidngrc, 2018-08-07:
Indeed, the long press work on all options, super nice.
I saw some guy flew 1500m (cloud surfing) last week, like you said, I don't think they care about the altitude alert either.
LOL.

davidngrc, 2018-08-08:
👍

closed 2018-08-08 by davidngrc