teckel12/LuaTelemetry #161: On development branch -> menu -> "fuel low", "fuel critical" should allow 1% increment.

davidngrc, 2018-08-06:
now only allow 5% increment for now, it would be better if we got 1% control.

teckel12, 2018-08-06:
@davidngrc This isn't specific to the development build.  The limitation is based on how it decides to report a low voltage.  Everything is setup to work in 5% steps so that's how the configuration is also setup.  Also on the lines of #160, smaller steps would require a longer time to change settings.

davidngrc, 2018-08-06:
how about add function : long press, then value continuously change?
in OpenTX 2.2 Lua Reference Guide https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/key_events.html
it mention something like

`The key event mechanism

Each time a key is pressed, held and released a number of events get generated. Here is a typical flow:
when a key is pressed a FIRST event is generated
**if the key continues to be pressed, then after a while a LONG event is generated
if the key continues to be pressed, then a REPEAT events are being generated**
when the key is released a BREAK event is generated
`

teckel12, 2018-08-06:
@davidngrc Long-press wasn't implemented as I have a QX7 which uses a dial control instead of plus/minus buttons.  I didn't even know there was a long-press for the plus/minus buttons till just now.

It may require a significant modification to change, so I'm going to shelf this issue as it's functional as-is unless you wanted to submit a PR.  It's also a non-issue with this particular issue as the battery percentage is in 5% steps which doesn't require many button presses anyway.

davidngrc, 2018-08-06:
I really wish OpenTx run javascript, LUA script is alien for me.
Just submit a feature request in OpenTx, hope someone will add it.

teckel12, 2018-08-06:
The long-press issue for the X9D is resolved in the development branch.  But the 5% steps for the battery is something that should probably stay this way.  While it could probably be done, I don't see much value in it as it's just for notification and 5% steps are fairly granular already.

davidngrc, 2018-08-07:
some people may like to have more control 😂, so 1% is better than 5%.

teckel12, 2018-08-08:
@davidngrc Sorted out issues with changing from 5% to 1% steps. Now gives battery percentage notifications every percent between warning and critical, but maybe that's better anyway. Tested a lot on the bench, but I feel confident it's good to go.

davidngrc, 2018-08-08:
Will help you test this on weekend.

closed 2018-08-19 by davidngrc

teckel12, 2018-08-19:
@davidngrc I've tested this a bunch in the field and I haven't had any issues.