teckel12/LuaTelemetry #162: Script Syntax Error

Fast-Guy, 2018-08-11:
Script Syntax Error  ?:0 attempt to call a nil value

I get this error 27sec after starting the Lua Script. Doesn't matter if running the radio by itself or with the model powered. Happens very close to 27sec every time. 
Frsky QX7 R9m R9
Opentx 2.2.2 
Lua Telemetry v1.4.0
Inav 2.0 RC4
Matek f405 Wing

**Additional context**
No other scripts running and the script runs perfectly for the 27 seconds before the error. Happened the same with Opentx 2.2.1 and all RC versions of Inav 2.0

teckel12, 2018-08-11:
@Fast-Guy This error typically means that one of the Lua files in the development branch is out of date with the others. But, it typically would happen immediately (or when something was selected, like the config menu). 

What doesn't make sense is the 27 seconds thing. Does something change at that point? Like does your model change to ready to arm? Or so you select something like a switch or the menu?

Since Lua Telemetry is pre-compiled, there's no useful error messages. The error message your getting is about the only message you'd ever see, and it doesn't tell me anything unfortunately.

Fast-Guy, 2018-08-12:
Thanks for the assistance. I do not get a config.dat file with v1.4.0. Only config.luac. I do get a config.dat file with v1.3.1 though. I did follow the instructions to run the development branch and use the files from the dist folder.

teckel12, 2018-08-12:
@Fast-Guy The config.dat file is created based on your config settings, it's not part of the distribution.

Basically, I'm trying to duplicate what you're running, as I load the files from the `dist` folder of the development branch and I have no issues. Something that could be different is your config.dat file. 

You could also just delete your config.dat file. It's possible there's a setting in there from some other release that's causing a problem.

In any case, you must be doing something different, as I run everything stock on the same transmitter with no problem. I've also never had a situation where something would fail after so long, nor has anyone reported anything like this. So you must have something unique on your end.

If you don't figure it out, maybe your transmitter config from the OpenTX Companion and the contents from your TELEMETRY folder. Then I could totally simulate your environment.

Fast-Guy, 2018-08-12:
After carefully reading what you said and being a long time computer user I decided to do a full format on my Micro SD card. I had done a quick format somewhere along the way. Reinstalled the Opentx card contents and Lua Telemetry v1.4.0 and now everything is working as it should. With all the Inav update  deletes and adds so something must have been left behind. I'll test some more and report back if there are any problems. Thanks for all the help and for creating such a useful Lua.

teckel12, 2018-08-12:
@Fast-Guy Glad to hear it's working. Crashing after exactly the same length of time each time you ran it even with the model turned off was a real head-scratcher. Sometimes things just don't work for unknown reasons. But at least your past it now.

closed 2018-08-12 by teckel12