teckel12/LuaTelemetry #169: Frsky X Lite and R9 slim+ with Fport

ReganGTX, 2018-08-25:
I'm having a issue, I've followed the steps on the Inav Lua Telemetry page and I can't get the lua telemetry script to show up, I'm running OpenTX 2.2.2RC5 on the X-Lite and Inav lua telemetry v1.4.0, I added both file and folder to my radios \SCRIPTS\TELEMETRY directory and added the script to my display screen and nothing, tried deleting sensors and rediscovering them after GPS 3D fixed, to no avail.
I'm assuming this has something to do with me using R9M lite module and the R9 slim+ with Fport?
I am getting full telemetry just no Lua telemetry.
My setup:
Matek F411-Wing on Inav 2.0.0 latest
Frsky X-lite OpenTX 2.2.2RC5/ R9 Lite module with latest Frsky firmware/ lua telemetry v1.4.0
Frsky R9 Slim+ with Fport firmware

I'd like to requests a future update for this to work with R9 receivers with Fport firmware.

Thank you,

ReganGTX, 2018-08-25:
Fixed my own issue, needed to update open TX 2.2.2
Opposed to 2.2.2 RC5

closed 2018-08-25 by ReganGTX

teckel12, 2018-08-25:
@ReganGTX Yes, they didn't make all the updates for the X-Lite till 2.2.2 final. The RC versions have limited X-Lite Lua support.

ReganGTX, 2018-08-26:
Thanks, Tim