teckel12/LuaTelemetry #170: Huge CPU usage

error414, 2018-08-25:
Hi,

I have problem wit last verision (cloned from github, used dist folder). After activate Inav lua script in "Display" page "Screen 1 Script iNav", taranis is slow as hell.

See video, I think that it is best description.  https://youtu.be/mPPkhgMAR0k

teckel12, 2018-08-25:
@error414 Has something to do with the release OpenTX 2.2.2  It wasn't doing this before.

teckel12, 2018-08-26:
@error414 See: https://github.com/opentx/opentx/issues/6127

teckel12, 2018-08-26:
@error414 Try Lua Telemetry v1.4.1 and report back.

zingaros, 2018-08-26:
Hello, 
User of the former version (1.3.1), I tried today the v1.4.1 on a Taranis X9D+ with OpenTX 2.2.2: pilot view is gorgeous (kudo!) but voice is most of the time impacted: its not any more Amber that is speaking but Dark Vador :-)

error414, 2018-08-26:
Ahh, sorry, I forgot write that I use opentx 2.2.2, it's nice you was able to find out from screen.
I have pulled last version and I can confirm, it WORKS fast. Tank you very much.

zingaros, 2018-08-26:
Error414: are you also experiencing, wile having the script activated, voice issues or I’m the only one (see my comment above)? Many thanks

error414, 2018-08-26:
https://youtu.be/7GikKAZwESk sound is still bad :(  

once happend to me that I got lua script error and then taranis said that he dosnt see SD card

teckel12, 2018-08-26:
It seems OpenTX 2.2.2 is **much** slower with SD card access. I didn't experience any of these problems when using a pre-release of 2.2.2 either. So somewhere in the 2.2.2 process something drastically changed.

The OpenTX team currently seems to be more interested in pointing blame than looking into the issue.  But I'm confident with enough evidence they'll look into it.

The solution for the short term may be to install a pre-release version of 2.2.2. I'm going to try a few versions I happened to download which may also help to isolate the problem with OpenTX.

teckel12, 2018-08-27:
@error414 Try using OpenTX firmware without the `luac` box checked. It seems that option drastically slows down OpenTX when there's SD card access. That change along with Lua Telemetry v1.4.1 should result in **MUCH** faster results. I get no "Darth Vader" voice alerts either with `luac` turned off.

Let me know how that works. Thanks.

teckel12, 2018-08-27:
@zingaros Try updating the OpenTX firmware to not include the `luac` option, that seems to be causing the fantastic slowdown and resulting voice problems. Also, I believe this issue is made much worse if you're using an extremely slow SD card as I don't experience it nearly as bad as your experiencing. Just something to keep in mind.

teckel12, 2018-08-27:
@error414 Your SD card must have very slow access.  The slower the SD card access, the more pronounced any slowdown will be.  There's a few things I'm doing with Lua Telemetry to try and address this issue.  I feel confident I can get the Darth Vader voice straightened up in a day or two.

The OpenTX team has been very unsupportive in my attempts to isolate and resolve the issue.  With that said, they did finally acknowledged there's some room for improvement and a change may very well happen.  After sorting out what I can change with Lua Telemetry, I plan on looking into the OpenTX code to address the issue directly.  But, OpenTX releases happen VERY infrequently, so this isn't really a solution as that could easily be 6 months away even if the change was merged today.

In any case, I'll continue to work on Lua Telemetry to resolve the Darth Vader voice issue.  I'll post back here with any updates.

error414, 2018-08-27:
hmm, it is possible, I use old 8GB SD card, it should be class 10 but who knows. I wanted to try samsung evo 64GB but it seems that taranis doesn't suport 64GB sd card. 
[image]

teckel12, 2018-08-28:
@error414 Try this pre-release build of 1.4.2: [LuaTelemetry1.4.2.zip](https://github.com/iNavFlight/LuaTelemetry/files/2326574/LuaTelemetry1.4.2.zip)

Just copy the contents to the `\SCRIPTS\TELEMETRY\` folder on your transmitter's SD card.  Let me know if the "Darth Vader" voice is gone as a result.  I feel very confident that it will sound much better with this pre-release.

The OpenTX team has also been working on cleaning up some code having to do with this.  It should be included in the 2.2.3 release of OpenTX.

zingaros, 2018-08-28:
hello
i followed your exchanges on Open TX: whaoo! Great job!
i tested the pre-release build of 1.4.2 : not working for me. it says "script syntax error". I tried several time with and without telemetry feed. Tested on Open TX 2.2.2 with the option no heli (luac is not enabled) on a taranis 9xd+.

error414, 2018-08-28:
@teckel12 1.4.2 works!!! without darth vader voice

teckel12, 2018-08-28:
@zingaros A script syntax error typically means that you have a mix-up in versions of Lua Telemetry files.  For example, if you just copy the `iNav.lua` file, you're almost guaranteed to have problems.  For the `LuaTelemetry1.4.2.zip` file linked above, make sure you copy the `iNav.lua` file as well as all of the contents from the `iNav` folder to your SD card's `\SCRIPTS\TELEMETRY\` folder.

FYI, the firmware options won't matter as far as running Lua Telemetry.  I would suggest `noheli` and `lua`, but neither are needed for Lua Telemetry to function.  Previously, I would have also suggested the `luac` option as well, but with 2.2.1/2.2.2 it's causing unnecessary file access.  So that option is up for debate till they release 2.2.3 which should include a fix.

teckel12, 2018-08-28:
f

teckel12, 2018-08-29:
History...

1. Issue reported with Lua Telemetry
2. Problem isolated to OpenTX's `luac` option
3. Fixed Lua Telemetry to get around OpenTX `luac` bug
4. Posted issue with OpenTX reporting `luac` option bug
5. OpenTX contributors attack because their code is bug free
6. Finger pointing that it's not a bug in OpenTX, but lazy programming with Lua Telemetry
7. Pull request to fix OpenTX even though it's bug free and there's no way it could be a bug in OpenTX
8. Lua Telemetry updated further to get around OpenTX `luac` bug till the fix is released
9. In denial and history revisionist OpenTX contributors continue to point blame even through there's a pull request that fixes the bug in OpenTX
10. Challenging OpenTX contributors that the above is what happened causes them to delete my message, as they can't argue with the truth

What a cheery group of individuals!  They seriously need to work on their customer support and triage skills.

teckel12, 2018-08-29:
@zingaros I'll create a new package and make sure everything is up to date.

teckel12, 2018-08-29:
@zingaros Try this: [LuaTelemetry-1.4.2b.zip](https://github.com/iNavFlight/LuaTelemetry/files/2334026/LuaTelemetry-1.4.2b.zip)

This assumes you already have 1.4.1 installed and these files go on top of your existing ones.  This was done with a new compile script in the hopes it would be more reliable.

teckel12, 2018-08-30:
@zingaros If it still crashes, try this version which includes debug info which may give some insight into the problem: [LuaTelemetry-1.4.2-debug.zip](https://github.com/iNavFlight/LuaTelemetry/files/2336186/LuaTelemetry-1.4.2-debug.zip)

This uses about 10k more memory, so it shouldn't be used long-term.  But, nice to be able to debug an isolated problem in a case like this.

zingaros, 2018-08-30:
Thanks
I will try tonight

zingaros, 2018-08-30:
@teckel12 Hello, 1.4.2b works perfectly 👍  so looks only it was a pb of packaging right?
Debug do script error but who cares? ;-) Thanks a lot for your continuous support!

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]
closed 2018-09-28 by teckel12