teckel12/LuaTelemetry #172: Use "getValue" directly instead of storing values in data variable

teckel12, 2018-08-28:
Per @schwabe in https://github.com/opentx/opentx/issues/6134
> I gave a very short look at your script. You can improve memory usage a lot if you don't store all telemetry to a global variable. Calling getValue whenever you need is perfectly fine.

teckel12, 2018-08-28:
Will update several sensors in this way and measure the memory used to determine what amount of memory could be saved if done globally.

teckel12, 2018-08-29:
@schwabe I changed 4 values, deleting them from the `data` table and instead calling `getValue` directly.  I didn't do any cherry picking, I just did the first 4 (well, I skipped `gpsAlt` because I'm not always just reading that value, sometimes it defaults to `0` if there's not a GPS fix).  In any case, here's the results:

Before change: **30,476** bytes free
After change: **30,420** bytes free

So it doesn't save any memory, it actually uses slightly more, and makes the code harder to read.  There must be more overhead calling the `getValue` function than the space used for the variable.

I was really hoping this save at least some memory, it was something I didn't try, and I've tried a LOT of things.  I'd love to hear more suggestions, so please don't let this make you feel discouraged.  Keep the suggestions coming!

schwabe, 2018-08-29:
getValue has no memory overhead, the only memory overhead you have from getValue is the space it uses in the script itself. Are you sure you removed also default value assignment etc? And remember the size of your lua script itself is also important. Shortening your data. to just d. probably saves you a number of bytes.

teckel12, 2018-08-29:
@schwabe Unfortunately yeah, I'm sure. I can provide a&b builds if you want to test, it only took a s few minutes to setup.

It's probably only saving the length of the variable and the space it uses, that's just not enough to be more than several getValue calls. I've experienced that lua (unlike other languages) doesn't always save memory when you make a repeating command into a function (sometimes it's better to repeat the same code than make a function). Lua is quirky with memory usage for sure. 

However, the other suggestion you made has worked out brilliantly. Which was to cache the last view loadScript and then purge it before loading a different view script. After a bunch of testing, I needed to both set the previous view to nil, do garbage collection, then doing the new loadScript. But, there was little overhead from doing this, and it allowed zero loads per UI loop but still providing the advantage of having multiple views (or a config view) with no memory overhead.

teckel12, 2018-08-29:
@schwabe Also, I've tried shortening variable names but it doesn't help that much but makes the source much harder to read. At one time I did exactly what you suggested. Renaming `data` just `d` but it only saved an insignificant few bytes. It just doesn't scale well as you'd drastically sacrifice code readability for maybe 100 bytes of memory savings in total. There's other ways to save 1000k or more without making readability compromises (for example, the several script loads at startup).

closed 2018-09-17 by teckel12