teckel12/LuaTelemetry #173: But v1.4.1 Pilot view “Battery Percentage" always show 0

davidngrc, 2018-09-01:
I am using opentx 2.2.2, iNav 2.0 FPORT

Pilot view “Battery Percentage" always show 0
I think this also exists in v1.4.0

when using classic view, the battery percentage bar is not 100%, like 80%  (cell voltage is 4.02v)
but in in x9d+ "telemetry" screen, the Fuel sensor shows 100%
iNav already using      smartport_fuel_unit = PERCENT

[image]

teckel12, 2018-09-01:
@davidngrc Set another telemetry screen to show the sensors fuel and battery voltage. Fuel in Lua Telemetry just comes direct from the sensor. Voltage is either the total or cell depending on what you select in the config.

My guess is that INAV is guessing the wrong number of cells, and calculating the fuel. Your min voltage of 2.8 may be throwing INAV off (as this is typically set much higher, like 3.5 volts).

Setting another telemetry screen with the fuel and battery telemetry is the way to go, that will show if the problem is with your INAV settings. Fuel being wrong really suggests that, as Lua Telemetry does no math to calculate fuel, it comes directly from INAV.

teckel12, 2018-09-11:
@davidngrc Any update on this?

davidngrc, 2018-09-13:
I have test this on a new matek f405 wing last night, from full 4.2v to 2.3v per cell (almost kill my 18650 battery) no problem found

Previously, I use FC: SP Racing F4 EVO, maybe tonight I will test this one again

teckel12, 2018-09-13:
@davidngrc The fuel % comes straight from INAV, so any problems with that must be with INAV.

The battery voltage could be a problem because your taking your voltage all the way down to 2.3v. The issue is that INAV doesn't sent the cell count and both INAV and Lua Telemetry need to guess the number of cells. This works well if you always start with a full battery. But, if the battery isn't full either INAV and/or Lua Telemetry may calculate the wrong number of cells. Neither have a fool-proof way of calculating this when you take your battery voltage that low.

If you have cell voltage reporting issues, I would suggest changing Lua Telemetry to show the total voltage instead of cell voltage, then there won't be an issue in Lua Telemetry. You also may need to mess with the new default battery settings in INAV to set your battery cell count, voltages, etc.

closed 2018-10-07 by stale[bot]