teckel12/LuaTelemetry #174: Skript Error in Pilot View V1.4

Pacman111178, 2018-09-02:
Hello . I flew my Lidl Glider with Inav 2.0.0 on an omnibus F3Nano with external Baro and suddenly there was a script error. The whole thing happened two times but only in the pilot's view!
[image]

teckel12, 2018-09-02:
@Pacman111178 try running the latest development build: https://github.com/iNavFlight/LuaTelemetry/files/2334026/LuaTelemetry-1.4.2b.zip

teckel12, 2018-09-04:
@spaelectronics What do you mean by "nothing"? To me, that means you can't even get it to start or you get a blank screen. The description of this issue is that it did work but then failed, with a particular error.

The debug doesn't show additional information except is there's a crash. Also, 1.4.0+ is really designed for thr release INAV 2.0.0, there could be issues with RC versions.

Also, why do you have multimodule turned on? Are you running SBUS? If you're sending new firmware, check just the noheli option (no lua or luac either). It will run faster and use less memory that way.

If it still doesn't work, I'll need your `config.dat` file from your iNav folder.

spaelectronics, 2018-09-04:
What I mean by nothing, is I had the same issue on all 3 versions. 

The error I get is the same syntax error as the original post.

I'm using the latest iNav version (2.0).

And I have multimodule turned on because I have a multimodule. I don't use it on this particular plane, but I use it to control my other Spektrum and Hubsan models.

In this case, I'm using a FrSky X8R receiver.

And what do you mean by: 
"If you're sending new firmware, check just the noheli option (no lua or luac either)"?

Isn't lua & luac needed to run lua scripts, such as this?

I haven't even been able to get this to work, not even once. From the very first time I installed it, I just keep getting this syntax error.

spaelectronics, 2018-09-04:
The very first time I load the script, I get this:
[image]

The second time, and so on, I get this:
[image]

I have no config.dat because I can't get it to work to edit any settings.
This happens both with the model turned on and off.

I tried formatting the SD Card, but that didn't seem to help.
Any ideas?

teckel12, 2018-09-04:
@spaelectronics This issue was that it worked for a while, and then there was a script error. If it doesn't work at all, that would be a different issue, which is why I wanted to clarify your "nothing" comment. The script error is just a general error, you'll always get that error for just about any issue in Lua Telemetry as it's pre-compiled.

You don't need to include lua or luac to use Lua Telemetry. As it's pre-compiled, neither are needed. I wouldn't suggest using the luac option at all as the transmitter typically isn't able to compile scripts anyway as it runs out of memory in the process. Also, there's a bug in OpenTX that the luac option causes unneeded and too much SD access which causes slow downs and possible robotic speech.

As to your problem, it sounds like there's a problem with installation as I haven't heard of this issue except for a miscondigured install. If you want me to duplicate it, I'll need a transmitter dump, your transmitter model, and your SD card contents. I can only duplicate OpenTX 2.2.2, so I'd suggest first trying 2.2.2 as that alone could solve the issue.

As far as being installed incorrectly, are you installing via the install instructions (using the release ZIP)? I'd delete the iNav folder and any iNav.lua or iNav.luac files in the TELEMETRY folder to be sure you don't have multiple versions installed.

Let me know if you find anything. Or the contents of your transmitter so I can duplicate.

spaelectronics, 2018-09-04:
I'm using OpenTX V2.2.2RC5.

Yes, I installed it correctly, using the release zip. 

I've already tried completely wiping the SD Card, so there was no duplicates.

I'll provide the contents of my transmitter later on today.

spaelectronics, 2018-09-05:
I'm using the Taranis X9D Plus.

I didn't know it was released?
I'll have to look into it.

spaelectronics, 2018-09-05:
Got it working!
All I did was upgrade OpenTX from 2.2.2RC5 to 2.2.2.

Thanks!

teckel12, 2018-09-05:
@spaelectronics :+1:

teckel12, 2018-09-06:
@Pacman111178 Sounds good, will release 1.4.2 final incorporating the changes.

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]
closed 2018-09-28 by teckel12