teckel12/LuaTelemetry #175: x9e Rotary Encoder doesnt work

Peschi90, 2018-09-10:
### Describe the bug

First of all REALY GOOD JOB with this script!!!

But at the moment i have a "little" problem.
The Telemetry start correctly but if i want to change the settings vie pressing "menu", then i cant Navigate with the Rotary Wheel from my Taranis x9e. I also cant use the Wheel to switch the telemetry view. 
The datas that are displayed are all correct.
In my OpenTX menus everything works normal.

I dont know excactly what for information you need to help me. If you need more i  will give it to you fast.

###Aditional context

Firmware OpenTX 9xe Version 2.2.2  10.08.2018

teckel12, 2018-09-11:
@Peschi90 Interesting catch! The 9xe must have slightly different event values for the rotary wheel compared to other Taranis transmitters. 

I'll look into this, should be an easy fix due to the way the control events are programmed. Expect a fix in v1.4.2 which I plan to release this Sunday.

Peschi90, 2018-09-11:
perfect! i will test it after release
Thanks for the fast reply

teckel12, 2018-09-16:
@Peschi90 Try the latest development build, it should now work on the X9E transmitter.  Instructions for upgrading here:

https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

Peschi90, 2018-09-16:
Thank you very much! with the development Build it worked for me!!

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]
closed 2018-09-28 by teckel12