teckel12/LuaTelemetry #177: Battery warnings always play

skorokithakis, 2018-09-14:
This is probably my fault again, but I can't figure out how, so here we are. When I arm the copter, I keep getting "battery critical: zero percent" audible warnings/vibrations/beeps. The problem is that the battery is over 95%, the UI shows that clearly, all the display bits show the right values, but my LuaTelemetry on my Taranis QX7 is freaking out.

I am running 1.4.1 with iNav 2.0.0, on default view. I have gone through the "battery settings" page and set things to all the recommended values (3.5/3.4) on both sides, but I get "battery zero percent" even though the gauge is showing 4.1 per cell. I am not using the Fuel display as I don't have a current sensor, but the fuel values from telemetry are a percentage that are around 95% as well (i.e. not 0).

What could be wrong?

teckel12, 2018-09-14:
@skorokithakis What have you set INAV and Lua Telemetry for the current value? PERCENT and the matching percent for Lua Telemetry?

It's possible that there's some kind of issue as you don't have a current sensor. I test this scenario, but it's possible there could be issues depending on different configurations.

Since you don't have a current sensor, I would remove the Fuel sensor in your transmitter to see how that works. That's how previous versions of INAV would have worked, but that changed with 2.0.0 so that's my guess with this issue.

Before you make changes, it would be great if you could export your transmitter configuration and your config.dat file from the iNAV folder on your transmitter. That way I could try to reproduce the problem to prevent the issue from happening.

teckel12, 2018-09-14:
@skorokithakis Perfect, and try to remove your Fuel sensor on your transmitter, that's usually how I test Lua Telemetry for those without a current sensors.

skorokithakis, 2018-09-14:
I just removed the Fuel sensor and tried again, no change: 4.09V per cell, "Battery critical: Zero percent" warnings.

teckel12, 2018-09-14:
@skorokithakis Won't be able to look into this till Sunday. But with this info I hope to be able to duplicate the issue. I'll let you know..

pkerney, 2018-09-14:
I'm getting this as well. Fresh battery, UI battery bar shows full, also voltage reads correctly but it is flashing and warnings going off. No current sensor. I still have a bit of investigation to do but it seems similar. I haven't spent much time using LuaTelemetry seriously yet.
Not sure if this is related but I am also getting div by zero lockups on LuaTelemetry and only way to recover is a TX power cycle which would be pretty scary if I was in the air and far out.

skorokithakis, 2018-09-14:
Wouldn't a lockup just cause the craft to return?

teckel12, 2018-09-15:
@skorokithakis If the script would crash your transmitter would still be functional. It would be like flying without Lua Telemetry is all.

There's a 1.4.2 pre-release in another issue that you should probably try as a few things have been resolved in that release. I can't work on Lua Telemetry till Sunday but 1.4.2 should be release then (hopefully resolving this issue as well).

teckel12, 2018-09-15:
@pkerney If you could also supply your transmitter dump and thr `config.dat` from your transmitter's iNAV folder that would be helpful in duplicating the issue and diagnosing it. Probably going forward I'm going to spell out exactly what files I need to diagnose problems like this, as at this point it seems only isolated issues are happening.

teckel12, 2018-09-16:
@skorokithakis Okay, back home.  Below are the instructions for the radio and model settings that I need to simulate (and hopefully reproduce your problem).  I believe the files you provided were to load back onto a transmitter instead of running in the OpenTX Companion.  In any case, the instructions below should give me a simulation of your transmitter:

1. Connect transmitter to PC using standard method ([trim in, power, connect USB](https://youtu.be/GAETmZsKqwY?t=2m13s))
1. From your transmitter's `\SCRIPTS\TELEMETRY\iNav` folder, copy the `config.dat` file to your PC
1. Run OpenTX Companion
1. From `Read/Write` menu, select `Read Models and Settings From Radio`
1. From `File` menu, select `Save As...` and save radio models and settings to OTX file
1. ZIP OTX file and the `config.dat` file and attach it to your issue
1. Make sure you specify the model name you're having the issue with

skorokithakis, 2018-09-16:
Oh sorry, I forgot to tell you. You need to rename the zip to OTX, as Github won't accept zip. The config.dat just contained the digits I pasted above.

Peschi90, 2018-09-16:
I have the same issue with the development build and no Current Sensor connectet and Configurated.

Peschi90, 2018-09-17:
after i set feuel from percent to mah inside luaTelemetry, i get no battery warning. 
But i dont know if warning works if the battery is realy low.

teckel12, 2018-09-17:
Try the latest development build, it should no longer report critical fuel levels when there's no current sensor. Instructions for upgrading here:

https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

skorokithakis, 2018-09-17:
I can confirm that I no longer get the spurious warnings in the latest dev build. I don't know if I get *any* warnings, as I haven't let the battery go that low yet, but the spurious ones are gone, thank you!

teckel12, 2018-09-17:
@skorokithakis As you don't have a current sensor, it will only give low battery warnings based on voltage, not fuel % (which is based on voltage anyway without a current sensor). Also, be sure you've turned on battery voltage warnings in the config menu.

I also noticed that you setup low battery warnings outside of Lua Telemetry for your iNAV model. These won't be needed after you're confident everything is working correctly. It could be confusing if both are setup.

teckel12, 2018-09-17:
@Peschi90 Try the latest development build linked in one of the above messages. The fuel setting in the config menu is now disabled if you don't have a current sensor (as you can't possibly show mAh used without one).

Let me know how it works.

teckel12, 2018-09-17:
@pkerney Try the latest development build. It should address this issue as well as clean up a few other issues.

If you still get any type of error (divide by zero or otherwise) please open another issue so different problems can be tracked separately.

skorokithakis, 2018-09-17:
@teckel12 Yes, voltage warnings are fine, thank you. I was planning to disable the old warnings once LuaTelemetry was working well, so it sounds like I'll be removing them soon, thanks.

closed 2018-09-28 by teckel12

skorokithakis, 2019-11-30:
@teckel12 I'm getting this again, and tried to file a bug only to find that I had the same problem a year ago. This time, I had a fuel sensor, but then removed it, and now continuously get the warnings. Is there something I can do to fix this?

skorokithakis, 2019-11-30:
Even after deleting the configs for this model, the warnings still always play. That's odd, because my other model (which never came with a fuel sensor) doesn't play the warnings.

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]