teckel12/LuaTelemetry #181: German Sound Files

Peschi90, 2018-09-18:
Hey, 
for me i made some German Soundfiles for the LuaTelemetry.
Maybe you want to implement the files into your Release?
My Idea is that german users can replace the english sounds with the german ones.

Only an Idea :)
[INav_Telemetry_DE_SoundFiles.zip](https://github.com/iNavFlight/LuaTelemetry/files/2392156/INav_Telemetry_DE_SoundFiles.zip)

teckel12, 2018-09-18:
@Peschi90 Multilingual support is welcome and has been on a list of things to do for some time.

Other than the sound files, is there on-screen text that would also be helpful for translation? As most words are fairly typical and many times shortened down to 3-4 characters, I'm not sure what would be appropriate for each language.

As for how I planned on adding this, I was thinking there would be a folder for each language where the sound files would be located. Also, a language file would be loaded for each language. The user would select the language and everything would change to that language.

It would be really helpful if you could go over the classic view, pilot view, and the config menu and translate what would be appropriate (keeping in mind the limited character space available). I could then setup the multilingual programming and hook up your sound files and text translations.

Thanks a lot! I've been wanting to add multilingual support for a long time.

Peschi90, 2018-09-18:
I'm not sure if it's necessary, but it's a nice feature, I think.
For me, it's important that my sound files have the same language as my global opentx sound files. For me, it sounds bad when telemetry is in a different voice than my openTX time or TX Battery.
The short words like "alt" or "spd" should not be translated becouse in German they have to be like "Höhe" or "Geschw". I think that is to long for the display and normally everyone in German who fly a Copter should know what alt or speed means :D 
but i think that it will be really nice if the settings menu has an translation.

I didn´t see all Pictures from the telemertry in my head at the moment. I will look later on the Taranis what would make sense to translate

teckel12, 2018-09-18:
@Peschi90 I can do some legwork on this too, by creating the config translation files making it easier and cleaner to translate.

teckel12, 2018-09-19:
@Peschi90 

### Here's the flight modes to review to see if translating would be appropriate:
! TELEM !
HORIZON
ANGLE
ACRO
NOT OK
READY
POS HOLD
WAYPONT
MANUAL
RTH
! FAIL !
! THROT !
CRUISE

### And here's the config menu items as well as those with English options:
Language = "English", "Deutsch"
Battery View = "Cell", "Total"
Battery Alerts = "Off", "Critical", "All"
Cell Low
Cell Critical
Fuel Unit = "Percent", "mAh", "mWh"
Fuel Low
Fuel Critical
Altitude Alert = "Off", "On"
Max Altitude
Timer = "Off", "Auto", "Timer1", "Timer2", "Timer3"
Tx Voltage = "Number", "Graph", "Both"
Rx Voltage = "Off", "On"
Variometer = "Off", "Graph", "Voice", "Both"
Vario Steps
Voice Alerts = "Off", "Critical", "All"
Feedback = "Off", "Haptic", "Beeper", "All"
RTH Feedback = "Off", "On"
HeadFree Fback = "Off", "On"
RSSI Feedback = "Off", "On"
Speed Sensor = "GPS", "Pitot"
View Mode = "Classic", "Pilot"
GPS HDOP View = "Graph", "Decimal"
GPS Warning
GPS Coords = "Decimal", "Deg/Min"
GPS

teckel12, 2018-09-19:
@Peschi90 The latest development build now includes your German language voice files.  It uses the transmitter's `Voice language` setting in radio setup.  So if it's set to `Deutsch` you should get the German voice files.  You may need to reboot the transmitter if it wasn't previously set.  Instructions for upgrading to the development build:

https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

Let me know how it works for you!

Peschi90, 2018-09-19:
**Flightmodes dont need so much translations**
! TELEM !
HORIZON
ANGLE
ACRO
**NICHT OK**
**BEREIT**
POS HOLD
**WEGPUNKT**
**MANUELL**
RTH
! FAIL !
**! GAS !**
CRUISE

The config menu needs a bit more. i hope the words are not to long. If they are to long, then i will shorten them a bit. Maybe you can say me what is the max character lenght for the Words.

Batterie Übersicht = "Zelle", "Total"
Batterie Warnung = "Aus", "Kritisch", "Alle"
Zelle niedrig
Zelle kritisch
Kapazität = "Percent", "mAh", "mWh"
Kapaz. niedrig
Kapaz. kritisch
Höhen Warnung = "Aus", "An"
Max Höhe
Timer = "Aus", "Auto", "Timer1", "Timer2", "Timer3"
Tx Spng. = "Nummer", "Graph", "Beide"
Rx Spng. = "Aus", "An"
Variometer = "Aus", "Graph", "Stimme", "Beides"
Vario Schritte
Sprach Alarme = "Aus", "Kritisch", "Alle"
Meldungen = "Aus", "Vibration", "Pieper", "Alle"
RTH Meldung = "Aus", "An"
HeadFree Meld. = "Aus", "An"
RSSI Meldung = "Aus", "An"
Geschw. Sensor = "GPS", "Pitot"
Ansicht = "Klassisch", "Pilot"
GPS HDOP  = "Graph", "Dezimal"
GPS Warnung
GPS Koordi. = "Dezimal", "Grad/Min"
GPS

The soundfiles i will test later when i am back home.

teckel12, 2018-09-19:
@Peschi90 Okay!  I've added your translations as well to the latest development branch.  A few things to note:

1. For the notification voice language, it uses the OpenTX Voice Language setting in the radio setup.
1. For the config menu language, it uses the Menu Language option for the radio firmware.  So, if your OpenTX menu language is German, the config menu will also be German.  If the config menu is still in English, your OpenTX menus are probably also in English.
1. Please review the [German language](https://github.com/iNavFlight/LuaTelemetry/blob/development/src/iNav/lang_de.lua) source, specifically:
   1. A couple of the translations are too long: `WEGPUNKT` and `Batterie Übersicht`.  There are notes which give the maximum character lengths.
   1. I made a couple changes to what you submitted as there was room: `GPS Koordinaten`, `Ansichtsmodus` and `Haptisch`.  I don't know German, so correct me if I'm wrong.
   1. Give it a once-over to make sure everything is correct.

Let me know how everything works and if I need to make any language changes.

Peschi90, 2018-09-20:
sorry i had no Time today i will test it tomorow and give you a feedback.

teckel12, 2018-09-20:
@Peschi90 In the meantime, if you could review https://github.com/iNavFlight/LuaTelemetry/blob/development/src/iNav/lang_de.lua for the German translation and shortening a couple field lengths that would be awesome.  I even had to add another config menu option today, so it's updated as of an hour ago.

Thanks again!

Peschi90, 2018-09-21:
i have checked it and shorten some words. I test the development version after work.

Peschi90, 2018-09-21:
I updated it a second time becouse i saw some problems with the Ü Ö and Ä inside the german Words.
at my taranis, changing the language works but only for soundfiles. i changed it to englisch and the sound files are english after an restart, the menu still is german.

How can i test the new prepared lang_de?

teckel12, 2018-09-21:
@Peschi90 It's correct that changing the voice language to German only changes the voice files (not the menus).  The option is for **voice** language, not the menus.  For the menus to be German, you need to have your transmitter flashed as German.  Basically, Lua Telemetry is following the same rules that you've already setup on your transmitter if you want a different language.

To change your transmitter to German, you need to send a flash file to your transmitter with German selected.  The Multilingual Wiki I created tells you how to do it:

https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support

There's easier ways I could have done this, but I figured why should I add a language option in Lua Telemetry if there's already language options in OpenTX.  And the way OpenTX is setup for language, the voice language is set inside the settings and the menu language is setup with flashing firmware.  Lua Telemetry respects both of these language selections.

Peschi90, 2018-09-22:
In think this is the best way! is the file inside the development build?

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]
closed 2018-09-28 by teckel12