teckel12/LuaTelemetry #182: Add "neutral throttle" prompt

skorokithakis, 2018-09-19:
I am not sure if this is possible, but there's this use case: I flip the switch to POSHOLD and my current throttle position becomes "holding". Raising the throttle makes the quad go up, lowering it makes it go down. Unfortunately, my throttle doesn't have a centering spring, so I can't really tell where the "holding" position is again without looking at the quad, and even that is not very accurate.

Maybe LuaTelemetry could detect POSHOLD and play a prompt like "neutral throttle" when throttle is back to neutral? That way I could always know when my throttle is at "hold".

teckel12, 2018-09-20:
@skorokithakis I like the idea. The only issue I can think of is that there's two settings in INAV for the throttle position for altitude hold. The other option in INAV is that the throttle center throttle position is fixed altitude.

So for this to work, there would need to be a configuration option that would allow the user to select this feature to be based on the last throttle position, center position, or turn the feature off.

Also, instead of a voice announcement, maybe a beep like the normal warning beeps but maybe higher pitched.

I'd also need to look at the throttle range where it would consider it an altitude hold. But I don't think that would be overly difficult.

And to clarify, this would work for both alt hold as well as pos hold.

Coding-wise it's fairly simple.

skorokithakis, 2018-09-20:
Certainly, that seems reasonable. By the way, is there a guide for what the beeps mean? I never understood why my radio is beeping at me...

teckel12, 2018-09-20:
@skorokithakis Beeps and vibration are warnings, and the reason should be indicated by whatever is flashing. There's no difference in any beep, it's the same frequency 2kHz every 3 seconds. It's not like 2 beeps means one thing and 3 beeps means something else. A beep (and/or vibration) are just an indication of some kind of warning (indicated by the flashing item on-screen).

I don't believe there's a situation where there would be a beep without flashing text to go along with it to let you know why.

You can turn off many (or all) of the beeps or vibrations. The config menu let's you be fairly specific with evactly what beeps or vibration you want (or don't want).

skorokithakis, 2018-09-20:
Ah okay, I might just not have noticed a flashing item because I was looking at the quad. I'll look at the display next time, thanks!

As an interface suggestion, a very short vibration (almost a click) when I've entered the neutral zone and a beep when I've left it would be an intuitive (at least to me) and easy-to-understand UI for this change.

teckel12, 2018-09-20:
@skorokithakis It may not work quite that way, as audio and hepatic feedback doesn't always happen at the time you want it to (if there's another sound or vibration recently).

I'll have to play with it a bit to see if it will work at all, before deciding how to best indicate neutral position.

teckel12, 2018-09-20:
@skorokithakis The latest development build now includes altitude hold throttle center position feedback.  It's a hybrid beep/vibrate (depending on your feedback config settings).  I've only tested it in the Companion but it seems to work.  The throttle center range will probably need to be tweaked, I just set it an arbitrary range of +/- 40 which seemed about right.

For the config menu, it's just a simple On/Off as come to find out INAV doesn't use the stick center but based on other settings so there's no way for the transmitter to know what INAV considers center throttle position if you have `nav_use_midthr_for_althold = on`.  Basically, for this to work, `nav_use_midthr_for_althold` needs to be set to `off`.

Here's the instructions for upgrading to the latest development build:

https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

Let me know how it works and what you think!

skorokithakis, 2018-09-20:
Fantastic, thank you very much! I will try as soon as possible (which is probably next weekend). Thanks again!

teckel12, 2018-09-21:
@skorokithakis I ran a battery through a quad with these changes and it worked well. I turned on pos hold at about 10 feet then went up and down a bit then used the feedback to find "center" and it was locked in. Then I flew up to 160 feet and did the same, stayed at the same altitude for a couple minutes. Then came down to 20 feet and did the same, worked great as well. Seems good, just some tweaking probably.

skorokithakis, 2018-09-21:
That's great news, thanks. I will try it too for a second data point and report back.

closed 2018-09-28 by teckel12

skorokithakis, 2018-09-29:
I just tried this and it is perfect, thank you very much. I love the high/low beep, although I think the vibration might be a *tad* too latent for the precision required. Maybe a third beep would be better instead of vibration when entering the dead zone? A few times I tried to center and got "vibration - beep" so I oscillated a few times to hit the dead zone. Not a huge problem, just a potential minor improvement.

skorokithakis, 2018-09-29:
Thinking about it more (and, again, this is predicated on the assumption that a beep is lower latency than a vibration), you could just play the high beep when entering the dead zone from above and the low beep when entering from below. If I am raising the throttle and hear a low beep, I know I'm in the dead zone, so there's no ambiguity there.

teckel12, 2018-09-29:
@skorokithakis The problem would be when you quickly go through the "hover zone" and it tries to play both tones it won't be able to, so you'll end up not getting the correct feedback.  You'll think you're in the zone when you are not.  I originally didn't have a vibration, just beeps at the edges, and it didn't work at all.

The way it is now, and seems to work (after a bit of pilot training to get used to it) is that the vibrate is the center and if you then go below the center it gives a low beep, and if you go above the center it's a slightly higher beep.

Now that I'm thinking about it, I can think of one possible improvement.  Instead of having the vibration be at the center of the zone, maybe it could be at either edge.  This would make hover zone seem a bit larger (as it's fairly tight as it is).  I also did try making the hover zone larger, which worked, until I drastically changed altitudes and then it wasn't quite as good.  So basically, the hover zone as it is now probably can't be increased (even though it sometimes seems like it could be).

skorokithakis, 2018-09-29:
Ah, okay, that makes sense. I thought the vibration was already at the edges, and I agree that that seems like an obvious improvement. Let me know if you want me to test the edge vibrations!

teckel12, 2018-09-30:
@skorokithakis After reviewing the code, it already gives heptic feedback when entering the hover zone and audio feedback when exiting. So there's nothing to change.

skorokithakis, 2018-09-30:
Fantastic, thanks for the feature!

PR #184 (2018-09-28): v1.4.2
* New radar view shows model in relationship to home position, can be displayed either as launch/pilot-based or compass-based orientation
* Added German language voice files, modes, and config menu (thanks @Peschi90). [Other language submissions encouraged!](https://github.com/iNavFlight/LuaTelemetry/wiki/Multilingual-Support)
* New feature gives feedback when altitude hold is activated and throttle position is in neutral hover position (vibrate when entering neutral hover position and beep when exiting)
* Cycle through views by short-pressing `Enter`
* Config menu now works with X9E tr [... cut]