teckel12/LuaTelemetry #192: Horus x10s

effysh, 2018-09-27:
Hi
wonder if there is any future support for the Horus x10s Tx with this project?

teckel12, 2018-09-27:
@effysh See #55. It's planned, but there doesn't seem to be much interest as I believe your the first Horus owner to request it. 

Horus would be implemented as a view just for Horus transmitters. This view would incorporate the multiple views available on Taranis.

The issues are theee-fold. First, I don't have a Horus transmitter so I don't completely understand how it would even work (I don't even know the procedure to setup a script on the Horus simulator). Secondly, as I don't have a Horus, testing can only be done in the simulator, and without  a good simulation environment I can't even begin to test. Third, and most importantly, Horus users need to step up and be willing to help, with creating a simulation build environment, method of setting up a script, testing code, or even better submitting pull requests with written/tested code.

As it is now, there's only a plan. What's needed is the Horus community to help, which is why I've flagged #55 as "help needed".

teckel12, 2018-09-30:
Duplicate #55

fchansa, 2018-10-21:
Request from a second Horus owner :-)

fmartinezo, 2018-11-12:
Request for a possible third horus owner :-)

teckel12, 2018-11-12:
@fmartinezo Request all you want, I can't write what I dont have nor understand. Issue #55 is tagged as help needed I believe.

teckel12, 2018-11-14:
Closed because it's a duplicate and Horus support will NEVER happen as long as no one is willing to help.

closed 2018-11-14 by teckel12