teckel12/LuaTelemetry #202: Several (3) battery profile types

fmartinezo, 2018-10-20:
Hi may be possible to create 2 or 3 battery profile tipes? basically for differents warnings for low voltaje.
Now i try to reduce from 3.5 (low) and 3.4 (critical) (li-po) to 3,2 and 3,1 for li-ion Then turn off my taranis and when it is turned on values was restored to 3,5 and 3,4.
I have several models on my taranis with both types of batterys and i think warning battery is very important, but for li-ion i am crazy because levels are not ok (percent and warning)

fmartinezo, 2018-10-22:
Okay, I understand the problem.
Now the problem is that if I change the values, now that I do it well the values are already conserved (I made the save wrong, which is why I lost them), but it does it for all the models, which forces me to oce I change the battery chemistry to enter the parameterization and modify it. Could not you save the parameterization by model or how well did you get it from inav?

teckel12, 2018-10-22:
@fmartinezo Got it! Yes, there's currently just one configuration for all models. Also, settings are also automatically saved as soon as you leave the config menu.

This does seem like something that could be enhanced in a few different ways. Basically, allowing a custom config for each model. For most, having the config mostly the same from model to model would be ideal, but allowing any config option to be changed for each model.

I'll put some thought into it on the best way to implement. It may just be that each model would start with the default values, that would the the easiest to implement.

Anyway, good idea! I believe there's something I can do with this.

fmartinezo, 2018-10-22:
@teckel12 Excellent, I will follow your progress at this point. An idea could be to give a name to the configuration file depending on the name assigned to the model in inav that for what I have seen you also have it in the lua. Pej config-f450.dat (for a frame f450) or config-z84.dat (wing wing Z84) and if it is not defined or does not exist, use the standard config.dat that could be renamed config-std.dat. I think this would occupy little memory and would only waste storage on the SD card

teckel12, 2018-11-11:
@fmartinezo I've added support for unique configuration per model: https://github.com/iNavFlight/LuaTelemetry/pull/199/commits/d45451ef156bfecb514440531dcb2832d614c4d9

This enhancement has been merged into the development build.  To use it, follow the [Upgrade to Development Build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) instructions.

fmartinezo, 2018-11-12:
Tested it and works fine. One configuration for model is ok. If i have 2 chemistry for the same model then create two models. No problem

fmartinezo, 2018-11-12:
Very very very good work. Thanks so much

teckel12, 2018-11-14:
Added in v1.4.3 release https://github.com/iNavFlight/LuaTelemetry/releases/tag/1.4.3

closed 2018-11-14 by teckel12